arendst/Tasmota #7693: Please Add a Option to Send ZbZNPReceived & ZbZCLReceived to MQTT

tcwj2008, 2020-02-07:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-02-07:
The problem is that enabling MQTT message for those message will swamp all users with raw data. These messages are only sent to logs with loglevel 3. One easy workaround would be to enable MqttLog at level 3. You will then receive `ZbZNPReceived` and `ZbZCLReceived` data.

blakadder, 2020-02-07:
tuyamcu has a setoption to enable sending extra data so everyone has a choice

s-hadinger, 2020-02-08:
Good point, I think I can reuse `SetOption66` for Zigbee as well. There is little chance that you have both Tuya MCU and Zigbee on same device and wanting different logging for both protocols.

PR #7699 (2020-02-08): `ZbZNPReceived`and `ZbZCLReceived`are published to MQTT when `SetOption66 1`
## Description:

`SetOption66 1` publishes `ZbZNPReceived` and `ZbZCLReceived` to MQTT instead of loglevel 3.

**Related issue (if applicable):** fixes #7693 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/aren [... cut]
closed 2020-02-08 by arendst