arendst/Tasmota #7697: Problem with LED-schemes

misll75, 2020-02-08:
### PROBLEM DESCRIPTION
Hello,

since v7.x.x there is a problem, switching to schemes with RGBW-LEDs.
I.e. when the white channel is on (ex. "color 000000FF"), and I want to
switch to color-rotating scheme with "scheme 2", the LED-Controlle switches
to white-channel again, instead of rotating the colors.

I have to switch to a certain color by myself before, to be able to use the scheme again.

Hope, I can help...
Andi

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [ v8.1.0.6] Tasmota binary firmware version number used: _____
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ OTA] Flashing tools used: _____
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
08:24:39 MQT: stat/sonoff_TVlicht/RESULT = {"NAME":"MagicHome RGBW","GPIO":[0,0,0,0,51,38,0,0,37,39,0,40,0],"FLAG":0,"BASE":34}
08:24:39 MQT: stat/sonoff_TVlicht/RESULT = {"Module":{"0":"MagicHome RGBW"}}
08:24:40 MQT: stat/sonoff_TVlicht/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"51":"IRrecv"},"GPIO5":{"38":"PWM2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"37":"PWM1"},"GPIO13":{"39":"PWM3"},"GPIO14":{"0":"None"},"GPIO15":{"40":"PWM4"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
08:25:17 MQT: stat/sonoff_TVlicht/STATUS = {"Status":{"Module":0,"FriendlyName":["TVlicht"],"Topic":"sonoff_TVlicht","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:39","StartupUTC":"2020-02-08T07:18:38","Sleep":15,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2020-02-08T08:07:40","SaveCount":751,"SaveAddress":"FA000"}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS2 = {"StatusFWR":{"Version":"8.1.0.6(004df12-tasmota)","BuildDateTime":"2020.02.07 18:04:40","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"337/699"}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"192.168.0.201","LogPort":514,"SSId":["Ponyhof",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008008","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS4 = {"StatusMEM":{"ProgramSize":570,"Free":432,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS5 = {"StatusNET":{"Hostname":"sonoff_TVlicht-0252","IPAddress":"192.168.0.226","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"212.33.36.155","Mac":"84:0D:8E:57:00:FC","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.201","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5700FC","MqttUser":"unicorner","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS7 = {"StatusTIM":{"UTC":"Sat Feb 08 07:25:17 2020","Local":"Sat Feb 08 08:25:17 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:10","Sunset":"17:58"}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS10 = {"StatusSNS":{"Time":"2020-02-08T08:25:17"}}
08:25:17 MQT: stat/sonoff_TVlicht/STATUS11 = {"StatusSTS":{"Time":"2020-02-08T08:25:17","Uptime":"0T00:06:39","UptimeSec":399,"Heap":24,"SleepMode":"Dynamic","Sleep":15,"LoadAvg":65,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Color":"000000FF","HSBColor":"0,0,0","White":100,"Channel":[0,0,0,100],"Scheme":0,"Fade":"ON","Speed":3,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Ponyhof","BSSId":"38:70:0C:F7:7D:B8","Channel":1,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The LED should rotate through the colors, when executing "scheme 2"

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-02-08:
As an additional feature it now allows you to change saturation during schemes. As you start a scheme with color 000000 saturation is 0 and you won't see any color change.

Set Saturation to 100 using either the slide or command ``HsbColor2 100`` and all should be well.

NB. Setting any basic color like Red, Green or Blue only has the same effect as you noticed.

arendst, 2020-02-08:
Duplicate of #7676

closed 2020-02-08 by ascillato2