arendst/Tasmota #7701: multiple devices setup

prabbit237, 2020-02-08:
**Have you looked for this feature in other issues and in the docs?**  

Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

I was setting up multiple smart plugs at one time (8 of them.) They are not all being put into service right away, so I wasn't setting them as "Living room lamp" and "coffee maker", etc. I was just getting them flashed, prepped and labeled with the MAC. But they all wind up with "tasmota" for the host name and friendly name, making it much harder to pick them out in the router's DHCP leases, etc. In the TasmoAdmin addon in HassIO, I get 8 plugs that look identical. 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

The MQTT Client ID defaults to a unique ID using "DVES_%06X." The topic/subtopic can also use a macro of %topic% and %prefix%. The fallback AP SSID even has a template. It'd be nice if the hostname and friendly name defaulted to  "Tasmota_%06X" or such (it'd be even nicer if the Hostname command would support such cause then a simple MQTT message could change all hosts at one time if desired but even just having the default use that pattern at initial setup would be really nice.)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

closed 2020-02-10 by arendst

prabbit237, 2020-02-10:
The topic wasn't really what I was looking to get set as a uniq entry. It's the hostname and friendly name that default to just "Tasmota." If I flash several devices and then power them all on, in my router, I see:

Tasmota     10.1.1.10       11:22:33:44:55:01
Tasmota     10.1.1.11       11:22:33:44:55:02
Tasmota     10.1.1.12       11:22:33:44:55:03
(etc)

In TasmoAdmin I see the same thing and in there it doesn't even show the MAC addresses. If I had them set for Wemo or Hue emulation, then Alexa would say "Discovered 3 devices" and they'd all be named "Tasmota." And mDNS won't work.

If they, instead, had Tasmota_%06 for the host name in the Wifi and/or the friendly name in the Configure Other then I can plug them all into a strip, scan in TasmoAdmin and then send a console command to set the template. Then I just turn them on one by one and see which one energized its relay, etc. Then I unplug that one, label it and then rinse-and-repeat till they're all ID'ed. Instead I've got to go into each one's webpage to see what its MAC is, change the hostname and friendly name, go back to TasmoAdmin and then change the name there to match.

Then once I get around to putting them in use, I can go back in one at a time and change the hostname, friendly name and topic to "coffee-pot", etc as needed but at that point, I know which one is which and on what IP and mDNS will work to load the configuration page directly.

ascillato, 2020-02-10:
> If I flash several devices and then power them all on, in my router, I see....

Please, erase your devices' flash and update to latest Tasmota. Tasmota now has as default hostname TASMOTA-XXXX where XXXX is last 4 digits of the MAC.

PR #7705 (2020-02-10): unique topic for every device by default
## Description:

**Related issue (if applicable):** fixes #7701

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).