arendst/Tasmota #7709: Support for Brightness, Colour Temperature on 2 PWM channels

DanielZimmer, 2020-02-10:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
I have some Verve Design ACL04HA lights, they run a TYWE2L module and I'm testing swapping this out for an ESP-12. They are Cool/Warm white lights and use 2 PWM Channels to control the lights but one channel is for brightness and the other channel is for colour temperature.
With the standard configuration they behave strange as Tasmota is expecting one channel to be warm white led and one channel cool white leds.

ie. PWM1 = 100%
PWM2 = 0%
is full bright, warm white

and PWM1 = 100%
PWM2 = 100%
is 100% full bright, cool white

This doesn't look to be a configuration option within Tasmota

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Add a configuration option to allow PWM1 Brightness and PWM2 colour temperature

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._
Alternatively this could be a configuration option under option 37

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-10:
You can control each pwm channels using channel command

s-hadinger, 2020-02-10:
Hint: `Module 48`

https://tasmota.github.io/docs/#/Lights?id=pwm-ct

closed 2020-02-10 by ascillato2