arendst/Tasmota #7714: Can't subscribe SENSOR topic from another device for displaying on OLED screen

xury77, 2020-02-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I cannot subscribe from another tasmota device using the subscribe command. Using self compiled binary with SUPPORT_MQTT_EVENT enabled in user_config_override.h.
As you see bellow requested topic is read correctly, but no event triggered after
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Wemos D1 with SSD1306 oled screen
- [ ] Tasmota binary firmware version number used: 8.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: platformio on visual studio code
- [ ] Flashing tools used: tasmotizer 1.1
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
22:42:14 MQT: stat/tasmota/RESULT = {"GPIO0":{"0":"Brak"},"GPIO1":{"0":"Brak"},"GPIO2":{"0":"Brak"},"GPIO3":{"0":"Brak"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"Brak"},"GPIO10":{"0":"Brak"},"GPIO12":{"17":"Przycisk1"},"GPIO13":{"4":"DS18x20"},"GPIO14":{"0":"Brak"},"GPIO15":{"0":"Brak"},"GPIO16":{"0":"Brak"}}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
stat/tasmota/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":369,"Rules":"ON Mqtt#Connected DO Subscribe Wschod, strych/tele/SENSOR, DS18B20-4.Temperature ENDON  ON Event#Wschod DO DisplayText Wschod: %value% C ENDON"}
22:45:30 SRC: Backlog
22:45:30 CMD: Grupa 0, Indeks 2, Komenda "RULE", Data ""
22:45:30 MQT: stat/tasmota/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
22:45:30 SRC: Backlog
22:45:30 CMD: Grupa 0, Indeks 3, Komenda "RULE", Data ""
22:45:30 MQT: stat/tasmota/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
  Rules output here:
Grupa 0, Indeks 1, Komenda "SENSOR", Data "{"Time":"2020-02-10T22:46:45","DS18B20-1":{"Id":"000002299591","Temperature":12.2},"DS18B20-2":{"Id":"000005FC188F","Temperature":4.4},"DS18B20-3":{"Id":"000005FC903C","Temperature":14.2},"DS18B20-4":{"Id":"021550986FFF","Temperature":3.7},"BME280":{"Temperature":12.7,"Humidity":48.3,"Pressure":974.8,"SeaPressure":997.1},"PressureUnit":"hPa","TempUnit":"C"}"

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:46:45 MQT: stat/tasmota/RESULT = {"Command":"Error"}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
trying to subscribe mqtt topic from another tasmotized device wit four ds18b20 sensors and one bme280sensor

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
To get ds18b20-4 temperature using subscribe command and next display value to OLED 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-11:
Your rules have some extra spaces between ON and ENDON. 

Anyway, why not using the standard precompiled version of Tasmota and making the following rules:

* In the Tasmota device that have the temperature sensor:

```yaml
Rule1 1 
Rule1 on tele-DS18B20-4#Temperature do publish cmnd/tasmota/EVENT TEMP=%value% endon
```

* In the Tasmota device that have the Display:

```yaml
Rule1 1 
Rule1 on EVENT#TEMP do DisplayText Wschod: %value% C ENDON
```

xury77, 2020-02-11:
Thank You **ascillato** 
Is working well I did not know that way.
So I can close this issue.

closed 2020-02-11 by xury77