arendst/Tasmota #7716: Display Time Zone, Current Date and Time, Time Zone Rules, and NTP server(s) in Information screen

apastuszak, 2020-02-11:
**Have you looked for this feature in other issues and in the docs?**  
Yes.
**Is your feature request related to a problem? Please describe.**  
I don't know what the time zone settings are for Tasmota

**Describe the solution you'd like**  
I'd like to see the information screen display the following:

- Current Time Zone
- DST rules
- Current Date and Time
- NTP server info
- If NTP is enabled or not.

**Describe alternatives you've considered**  
I looked at the documentation to see if there was a way to see this from the console and I did't see anything.

**Additional context**  
There are built in timers now.  I think having correct time and date set is important

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2020-02-11:
`Backlog Time;NTPServer1;NTPServer2;NTPServer3;Timezone;TimeSTD;TimeDST`

apastuszak, 2020-02-11:
That worked.  Still think it would be nice to have it in the GUI, as long as it doesn't use precious RAM needed for something else.

apastuszak, 2020-02-11:
I just used this command to set my time zones:

Backlog NTPSERVER2 us.pool.ntp.org; NTPSERVER3 0.us.pool.ntp.org; TIMEDST 0,2,3,1,2,-240; TIMESTD 0,1,11,1,2,-300

Works great.  But when I reboot the switch it goes back to the default NTP info.  Is there some way to get to survive a reboot?

apastuszak, 2020-02-11:
Applied it again and rebooted and that time it took.

I'm good with the command line to get this info.  I will close the issue.

closed 2020-02-11 by apastuszak