arendst/Tasmota #7719: Add support for WeMos D1 Mini motor shield I2C

janlinders, 2020-02-11:
**Have you looked for this feature in other issues and in the docs?**  Yes 

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
Would love to see that Tasmota can drive a DC motor via de Wemos D1 Mini Lite/Pro 
connected with Wemos D1 Mini motor shield on top. The footprint can be kept small and less wiring needed.
I2C is now on D1 and D2 of the Wemos D1 Mini.

I2C address can set to : 0x2D, 0x2E, 0x2F or 0x30 (default) on the motor shield
D1 = SCL
D2 = SDA
RST=RST

I would love to help in testing/debugging and extend the Tasmota functionality.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

mike2nl, 2020-02-13:
@janlinders 
it would be better you had added the link to the exsisting library ;-)
https://github.com/wemos/LOLIN_I2C_MOTOR_Library

And the wiki for more information
old wiki: https://wiki.wemos.cc/products:d1_mini_shields:motor_shield
new wiki: https://docs.wemos.cc/en/latest/d1_mini_shiled/motor.html

But be aware to sak for a driver wher nobody can get the hardware in 
the LOLIN/WEMOS anymore. Looks like they are gone.
https://lolin.nl.aliexpress.com/store/1331105

ascillato, 2020-02-13:
@mike2nl 

I found that it is available for buying at https://www.banggood.com/Motor-Shield-For-D1-Mini-I2C-Dual-Motor-Driver-TB6612FNGV1_0_0-p-1102387.html

But if this module is going to be deprecated, there is no point on working on a driver for this.

ascillato2, 2020-02-13:
Closing as this hardware module is deprecated.

If there is a new module, let's reopen this or provide a PR. Thanks.

closed 2020-02-13 by ascillato2

mike2nl, 2020-02-20:
@janlinders @ascillato2 
There is the newer version 2.0.0

https://wiki.wemos.cc/products:d1_mini_shields:motor_shield

PR #7764 (2020-02-22): Add Wemos DC mottor shild (v1.0.0) support
## Description:

**Related issue (if applicable):** fixes #7719

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).