arendst/Tasmota #7720: missmatching UI using slider

discip, 2020-02-11:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
**Please refer to the following link [https://github.com/stefanbode/Sonoff-Tasmota/issues/172]**

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-11:
https://github.com/stefanbode/Sonoff-Tasmota/issues/172#issuecomment-517993695

closed 2020-02-11 by ascillato2

discip, 2020-02-11:
@ascillato2 
Sorry, but did I do anything wrong, or why was this issue closed without further explanation?

ascillato, 2020-02-11:
Stefan has already explained that to you in the linked issue.

discip, 2020-02-11:
Sorry sir, but if you read carefully, you will find, that he never addressed the actual thing I described to him thoroughly, even twice. And he did not even bother to answer my latest comment. I even tried it via email, unfortunately with no luck at all.
Please do not take this as an offence, I am just trying to be objective.
Please let others look into this as well before closing this issue, because it is not solved yet.
Thank you in advance
kind regards

ascillato2, 2020-02-11:
No offense taken. No problem at all.

Stefan is the author of the shutters driver and sorry but he did answer to you and also he said in the previously linked comment that he is not going to make your requested change. Sorry.

Anyway, thanks for sharing your ideas.