arendst/Tasmota #7721: 8.1.0 schemes not working? (RGBW bulb)

foxsam21, 2020-02-11:
### PROBLEM DESCRIPTION
When I do `scheme 4` in previous versions the colors flashed rapidly. After upgrading to 8.1.0 (and 8.1.0.7) nothing happens.
I am using a lohas bulb configured with its template. This is the bulb: https://templates.blakadder.com/lohas-ZN001.html

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): see above
- [X] Tasmota binary firmware version number used: 8.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
20:38:32 CMD: Backlog Template; Module; GPIO 255
20:38:32 RSL: stat/lohas3/RESULT = {"NAME":"Lohas RGBW","GPIO":[0,0,0,0,0,0,0,0,0,143,0,144,0],"FLAG":0,"BASE":18}
20:38:32 RSL: stat/lohas3/RESULT = {"Module":{"0":"Lohas RGBW"}}
20:38:33 RSL: stat/lohas3/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"143":"MY92x1 DI"},"GPIO14":{"0":"None"},"GPIO15":{"144":"MY92x1 DCKI"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
20:39:31 CMD: Status 0
20:39:31 RSL: stat/lohas3/STATUS = {"Status":{"Module":0,"FriendlyName":["Lohas 3"],"Topic":"lohas3","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:39:31 RSL: stat/lohas3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:34","StartupUTC":"2020-02-11T19:30:57","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2020-02-11T20:14:12","SaveCount":24,"SaveAddress":"F4000"}}
20:39:31 RSL: stat/lohas3/STATUS2 = {"StatusFWR":{"Version":"8.1.0.7(9814468-tasmota)","BuildDateTime":"2020-02-10T23:00:10","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"357/699"}}
20:39:31 RSL: stat/lohas3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FurryTiger","SlipperySnail"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
20:39:31 RSL: stat/lohas3/STATUS4 = {"StatusMEM":{"ProgramSize":569,"Free":432,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
20:39:31 RSL: stat/lohas3/STATUS5 = {"StatusNET":{"Hostname":"lohas3-3288","IPAddress":"192.168.1.113","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:97:AC:D8","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
20:39:31 RSL: stat/lohas3/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_97ACD8","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:39:31 RSL: stat/lohas3/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 11 19:39:31 2020","Local":"Tue Feb 11 20:39:31 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:05","Sunset":"18:03"}}
20:39:32 RSL: stat/lohas3/STATUS10 = {"StatusSNS":{"Time":"2020-02-11T20:39:31"}}
20:39:32 RSL: stat/lohas3/STATUS11 = {"StatusSTS":{"Time":"2020-02-11T20:39:32","Uptime":"0T00:08:35","UptimeSec":515,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"000000FF","HSBColor":"213,0,0","White":100,"Channel":[0,0,0,100],"Scheme":4,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"SlipperySnail","BSSId":"C2:4A:00:7B:63:9C","Channel":9,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
`scheme 4`

### EXPECTED BEHAVIOUR
as above

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
`scheme 3` also has no effect when current color is 000000FF (white) but if color is changed i.e `color 1` then `scheme 3` does slow change and `scheme 4` also does slow.

BTW I had this working on one of the bulbs and the only thing changed was the firmware upgrade.

**(Please, remember to close the issue when the problem has been addressed)**
> **(Please, remember to close the issue when the problem has been addressed)**

OK

Jason2866, 2020-02-11:
Duplicate of #7676

foxsam21, 2020-02-11:
Maybe, but I think I was unclear. In previous versions the colors changed *very* fast. Now they don't behave like they did previously ever. The closest is that they change at all if there is a color already set (as in #7676) but even then the behavior is not like it was before upgrade.

Jason2866, 2020-02-11:
Yes, `scheme 4` is changed see https://github.com/arendst/Tasmota/commit/edf15844310f77d0d1311f94cfb6ef44b9f39de3

foxsam21, 2020-02-11:
I see. But the speed of the changes was supposed to change also?

ascillato2, 2020-02-11:
Yes

closed 2020-02-11 by ascillato2