arendst/Tasmota #7724: White Channel missing in 8.1.0.7 HA Discovery

patrickdk77, 2020-02-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
White channel is missing in home assistant when using auto discovery. There used to be a white slider that would reset the bulb to white, now it stays on colors with no way to reset to white. On RGBW bulbs even using the tasmota web control, there is no way to reset it to use the white channel. When using RGBCCT bulbs you can use the white balance slider to reset it to white only when using the tasmota web interface.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Lohas zx006 and fiet om60 bulbs
- [x] Tasmota binary firmware version number used: 8.1.0.7
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Lohas100 RGBW","GPIO":[0,0,0,0,0,0,0,0,0,143,0,144,0],"FLAG":0,"BASE":18}
{"Module":{"0":"Lohas100 RGBW"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"143":"MY92x1 DI"},"GPIO14":{"0":"None"},"GPIO15":{"144":"MY92x1 DCKI"},"GPIO16":{"0":"None"}}

{"NAME":"FE-OM60-15K-AG","GPIO":[0,0,0,0,141,140,0,0,37,142,38,0,0],"FLAG":0,"BASE":18}
{"Module":{"0":"FE-OM60-15K-AG"}}
{"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"141":"SM16716 DAT"},"GPIO5":{"140":"SM16716 CLK"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"37":"PWM1"},"GPIO13":{"142":"SM16716 PWR"},"GPIO14":{"38":"PWM2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
22:58:06 MQT: mylight/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["pdk-iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","00800200","00000000"]}}
22:58:06 MQT: mylight/stat/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FC000C3","04068001","00000085","000001C0","00004881","00000000"],"Drivers":"1,2,3,4,8,9,10,12,16,18,19,24,29","Sensors":"1,2,3,6"}}
22:58:06 MQT: mylight/stat/STATUS5 = {"StatusNET":{"Hostname":"mylight-4204","IPAddress":"192.168.11.116","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"DC:4F:22:AD:90:6C","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
22:58:06 MQT: mylight/stat/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.dswett.patrickdk.com","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AD906C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:58:06 MQT: mylight/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Feb 12 03:58:06 2020","Local":"Tue Feb 11 22:58:06 2020","StartDST":"Sun Mar 08 02:00:00 2020","EndDST":"Sun Nov 01 03:00:00 2020","Timezone":99,"Sunrise":"07:10","Sunset":"18:14"}}
22:58:06 MQT: mylight/stat/STATUS10 = {"StatusSNS":{"Time":"2020-02-11T22:58:06"}}
22:58:06 MQT: mylight/stat/STATUS11 = {"StatusSTS":{"Time":"2020-02-11T22:58:06","Uptime":"0T08:33:56","UptimeSec":30836,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER":"OFF","Dimmer":47,"Color":"0,0,0,88,32","HSBColor":"325,100,0","White":47,"CT":245,"Channel":[0,0,0,35,13],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"pdk-iot","BSSId":"86:2A:A8:81:3F:0F","Channel":11,"RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
setoption19 1
change light to any color
attempt to reset to white channel

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Used to have a white slider that acted as the white channel brightness, could be used to reset the bulb back to white channel.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

patrickdk77, 2020-02-12:
Going back to commit df35a54824462ceacc890b623f16fad2df6ee9b5 corrects the issue, a 8.1.0.6 tagged version.
[image]
[image]

Jason2866, 2020-02-12:
Yes, there where changes in HA AD after this commit. @effelle can you take a look?

effelle, 2020-02-12:
@patrickdk77 yes, I'm working on the new light module for HA. 
Ping me on Discord, I could use your help to do some tests.

effelle, 2020-02-12:
> On RGBW bulbs even using the tasmota web control, there is no way to reset it to use the white channel. 

meaning the web interface has issue too and is not just HA related?

patrickdk77, 2020-02-12:
The web interface I just never noticed before, this was an issue unrelated to these changes. I believe it has always been an issue. I have 3 sliders, change color, change color intensity from white to max color, and change brightness. I just overly noticed it, since I was attempting to reset to white, and couldn't

But the change color slider from white to color setting changes it using rgb white to rgb color, and white channel is ignored. I don't think this is exactly wrong though. On CCT bulbs and in HomeAssistant there is a 4th slider for white color, and using CCT this is white tempature, and resets the bulb to using the white channels as expected, just missing on RGBW. Home Assistant uses the same thing, but instead of using it as a white tempature slider, it's a white channel brightness slider, basically a duplicate of the brightness slider, except it resets the light to white channel.

I thought I remembered this in an older tasmota version, and if I did migh have to go down to v7 to see, if it existed back then. But this is my only RGBW bulb, as the rest I have are RGBCCT.
[image]

effelle, 2020-02-12:
Got it. I'll have a look on it.

effelle, 2020-02-12:
@patrickdk77 please test a development version >= commit `8079ee9` with both RGBW and RGBCCT. Thanks.

patrickdk77, 2020-02-12:
Yes, this has the desired result on both types of lights.

The web interface still remains, but that is really a separate issue I noticed cause of this issue.

effelle, 2020-02-12:
ok. Thanks.
@ascillato didn't find an issue about the slider. If you find it could you gently reference it?

Jason2866, 2020-02-12:
Now i know what is the reason for increasing mqtt message size. Was on the way to ask Theo 😀

ascillato2, 2020-02-12:
> Yes, this has the desired result on both types of lights.

Great to know that now it is working for you.

closed 2020-02-12 by ascillato2