arendst/Tasmota #7737: poweronstate 1 not working for Sonoff Dual R2 Module both releais

riker65, 2020-02-14:
### PROBLEM DESCRIPTION
poweronstate 1 not working for Sonoff Dual R2 Module for both relais only relais 2 is on after powerup.
powerupstate 1 is set

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____
- [ x] Tasmota binary firmware version number used: _____
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```

Program Version | 8.1.0(tasmota)
-- | --
2019-12-25T12:33:25
2_6_1/2.2.2-dev(38a443e)
0T00:07:33
61 at 0xF7000
16
Software/System restart

  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
`08:31:08 MQT: tele/TA_SOR2/STATE = {"Time":"2020-02-14T08:31:08","Uptime":"0T00:05:13","UptimeSec":313,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"TR7272","BSSId":"36:A4:3C:DD:17:E1","Channel":6,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:06"}}
08:34:37 MQT: stat/TA_SOR2/STATUS = {"Status":{"Module":39,"FriendlyName":["TA_SOR2_1_KH","TA_SOR2_2_Anbau"],"Topic":"TA_SOR2","ButtonTopic":"0","Power":2,"PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
08:34:37 MQT: stat/TA_SOR2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:42","StartupUTC":"2020-02-14T07:25:55","Sleep":50,"CfgHolder":4617,"BootCount":16,"SaveCount":61,"SaveAddress":"F7000"}}
08:34:37 MQT: stat/TA_SOR2/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"386/699"}}
08:34:37 MQT: stat/TA_SOR2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TR7272","TR_ESP"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
08:34:37 MQT: stat/TA_SOR2/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
08:34:37 MQT: stat/TA_SOR2/STATUS5 = {"StatusNET":{"Hostname":"TA_SOR2-7721","IPAddress":"192.168.1.155","Gateway":"192.168.0.31","Subnetmask":"255.255.240.0","DNSServer":"1.1.1.1","Mac":"2C:F4:32:A5:3E:29","Webserver":2,"WifiConfig":4}}
08:34:37 MQT: stat/TA_SOR2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.9","MqttPort":1893,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A53E29","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
08:34:37 MQT: stat/TA_SOR2/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 14 07:34:37 2020","Local":"Fri Feb 14 08:34:37 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:00","Sunset":"18:08"}}
08:34:37 MQT: stat/TA_SOR2/STATUS10 = {"StatusSNS":{"Time":"2020-02-14T08:34:37"}}
08:34:37 MQT: stat/TA_SOR2/STATUS11 = {"StatusSTS":{"Time":"2020-02-14T08:34:37","Uptime":"0T00:08:42","UptimeSec":522,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"TR7272","BSSId":"36:A4:3C:DD:17:E1","Channel":6,"RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:06"}}}`

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-14:
Confirmated bug.

`Poweronstate 1` on a sonoff dual R2 has no effect when rebooting the device.

Thanks for reporting.

ascillato2, 2020-02-14:
For generic module it is not working neither.

arendst, 2020-02-14:
Poweronstate is designed to (re)set the power state to the state before the power was removed.

It is NOT designed to (re)set the power state on a restart.

arendst, 2020-02-14:
Just checked with and it works as designed.

ascillato2, 2020-02-14:
Yes, I totally forgot about that. It is working as designed. Thanks.

closed 2020-02-14 by ascillato2

riker65, 2020-02-14:
Hi I do nbot understand the docu than.

https://tasmota.github.io/docs/#/PowerOnState

-> 1 / ON = turn relay(s) ON after power up

> Poweronstate is designed to (re)set the power state to the state before the power was removed.

thisit not stated this way in the wiki, what am I missing.

I understood you comment as state 3!

3 = switch relay(s) to their last saved state (default)

so how to enure that relais are on after powerup?

Thanks

ascillato, 2020-02-14:
Power up is not a restart.

riker65, 2020-02-14:
maybe I phase it the other way.

After an power outage I need the relais to be set to on.

I assumed poweronstate 1 does ensure relais to be on after power up.

this is stated in the wiki.

-> 1 / ON = turn relay(s) ON after power up

but this is not happening. 

I think I was not asking for restart but powerup.

Thanks for clarification

arendst, 2020-02-14:
In that case make sure interlock is off

riker65, 2020-02-14:
Hi, I did not specify an interlock 

`5:33:28 CMD: interlock
15:33:28 MQT: stat/TA_SOR2/RESULT = {"Interlock":"OFF","Groups":"1,2"}`

so your saying. poweronstate 1 should switch all releais on when interlock is off?

arendst, 2020-02-14:
In my test, half an hour ago, both releays were powered on after re-applying power. Before the power drop I had them turned off.

So yes, all relays should come on with ``PowerOnState 1``

riker65, 2020-02-14:
ok will retest. That is what I also did.