arendst/Tasmota #7738: WeMo Emulation Port assignment/visiblity

hyukishi, 2020-02-14:
**Have you looked for this feature in other issues and in the docs?**  
Yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Somewhat, the issue is that my WeMo emulated devices are not easily found even when using the static configuration because the port is required. So finding the port is the problem.
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
The solution would be to add the ability to assign a port to the WeMo emulated device or at least making the port visible on the same configuration page that the enable WeMo emulation feature is on.
**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I've tried using nmap, however since the WeMo emulation uses UDP ports, they are not caught in the scan of the target device.
**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-15:
If you change the port of the Emulation, the rest of standard softwares won't be able to find your device. Seems not to be a good option to change that.

As in older issues, seems that your issue is that multicast is not enabled in your router or it is using an old version of igmp. Try changing your router configuration.

hyukishi, 2020-02-15:
Made sure multicast is enabled on my router. Still unable to add the WeMo emulated device to my home assistant installation. I'm trying to get away from mqtt whenever its not absolutely necessary for my use case.

Jason2866, 2020-02-15:
How have you checked that the udp traffic is reaching the endpoint(s)?

hyukishi, 2020-02-15:
To elaborate on the issue, WeMo emulated devices show up in Alexa just fine. The issue comes with trying to discover them in home assistant. So yes, UDP traffic is reaching the endpoints for certain.

In Home Assistant it is strongly recommended to define the port used by the emulated WeMo device that you are trying to discover in it. Does anyone know of a default port that is assigned for WeMo emulation in Tasmota?

Jason2866, 2020-02-15:
Do you have a Alexa device (hardware)?  Without it will NOT work.
WeMo and HueEmulation from Tasmota is designed and tested only to work with Alexa (hardware).

hyukishi, 2020-02-15:
Yes, I have it discovered with my echo dots... Discovery with Alexa is NOT the issue. Discovery in home assistant is the issue and reason for suggesting display of the wemo emulation port in the information section.

Jason2866, 2020-02-15:
As said before you can NOT discover or use with home assistant. ONLY with Alexa.

closed 2020-02-15 by hyukishi