arendst/Tasmota #7739: after activating #define USE_GPS, ESP8266 goes into reset loop

adezerega, 2020-02-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Trying to setup GPS-NTP-Server. Uncommented #USE_GPS. After flashing, ESP8266 board goes into reset loop.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NodeMCU board
- [X] Tasmota binary firmware version number used: 8.1
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Arduino 1.8.10
- [X] Flashing tools used: Arduino 1.8.10
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
Can't execute command - device continuously resets

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
Not applicable

```
- [ ] Provide the output of this command: `Status 0`:
```
Not applicable - device continuously resets
  STATUS 0 output here:

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
Not applicable - device continuously resets

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compile Tasmota 8.1 with #define USE_GPS option

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Tasmota running

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Serial monitor shows:

00:00:00 CFG: Loaded from flash at FB, Count 1
00:00:00 APP: Some settings have been reset (244)
00:00:00 Project tasmota Tasmota Version 8.1.0(tasmota)-2_6_1

Exception (28):
epc1=0x4020f4fc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd70 end: 3fffffc0 offset: 01a0
3fffff10:  00000000 00000004 3fff1a9a 402236ac  
3fffff20:  00000000 00000004 00000012 4022c61b  
3fffff30:  3fff1a90 3fff1fec 00000010 4020f92e  
3fffff40:  00000004 0000004c 00000004 00000001  
3fffff50:  3fff1a90 3fff1fec 3fff1cec 4021afcc  
3fffff60:  6d736174 0061746f feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  3fff1d75 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff3bdc  
3fffffa0:  3fffdad0 00000000 3fff3b9c 402413e0  
3fffffb0:  feefeffe feefeffe 3ffe86f8 40101981  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld

00:00:00 CFG: Loaded from flash at FB, Count 1
00:00:00 APP: Some settings have been reset (245)
00:00:00 Project tasmota Tasmota Version 8.1.0(tasmota)-2_6_1

Exception (28):
epc1=0x4020f4fc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd70 end: 3fffffc0 offset: 01a0
3fffff10:  00000000 00000004 3fff1a9a 402236ac  
3fffff20:  00000000 00000004 00000012 4022c61b  
3fffff30:  3fff1a90 3fff1fec 00000010 4020f92e  
3fffff40:  00000004 0000004c 00000004 00000001  
3fffff50:  3fff1a90 3fff1fec 3fff1cec 4021afcc  
3fffff60:  6d736174 0061746f feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  3fff1d75 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff3bdc  
3fffffa0:  3fffdad0 00000000 3fff3b9c 402413e0  
3fffffb0:  feefeffe feefeffe 3ffe86f8 40101981  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v482516e3
~ld

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I have successfully flashed self-compiled versions with other options (such as #define USE_MCP230xx or #define USE_PCF8574) 

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-15:
Sorry, I can't reproduce your issue. It works on my setup. Please erase flash before uploading your self compiled version.

adezerega, 2020-02-15:
As suggested, tried erasing flash and with a brand-new NodeMCU. Getting the same error. Maybe some issue with my Arduinosetup. Thanks for the help - I will restart from scratch with another compile/flashing platform. Weird issue is that self-compile without the #define USE_GPS works perfectly.

closed 2020-02-15 by adezerega

adezerega, 2020-02-15:
I recompiled Tasmota 8.1 with Tasmocompiler (from gitpod.io), and had the same issue. Attached please find the user_config_override.h file used
[user_config_override.txt](https://github.com/arendst/Tasmota/files/4207616/user_config_override.txt)
I must be doing something wrong, but with the automated tool should work.

adezerega, 2020-02-15:
Finally, it worked with the development version (8.1.0.8). Same user_config_override.h file, but now it runs without the reset.