arendst/Tasmota #7740: sonoff basic loosing time settings on power fail

fracalifa, 2020-02-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

The "sonoff basic" looses all time settings by interrupting the main power supply, as if data is only stored in ram and not in eeprom or flash.  After powering on all time settings must set again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff Basic
- [X] Tasmota binary firmware version number used: 8.1.x
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:39:13 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff_Basic_AUX"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":0,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:39:13 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:43:58","StartupUTC":"2020-02-15T17:55:15","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":20,"SaveAddress":"F8000"}}
19:39:13 RSL: STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019.12.25 12:47:07","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"347/699"}}
19:39:13 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MxWLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000001","00000000"]}}
19:39:13 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
19:39:13 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-3615","IPAddress":"192.168.0.196","Gateway":"192.168.0.244","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.244","Mac":"DC:4F:22:DC:4E:1F","Webserver":2,"WifiConfig":4}}
19:39:13 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Feb 15 18:39:13 2020","Local":"Sat Feb 15 19:39:13 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:59","Sunset":"18:09"}}
19:39:13 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-02-15T19:39:13"}}
19:39:13 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-02-15T19:39:13","Uptime":"0T00:43:58","UptimeSec":2638,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"MxWLAN","BSSId":"1A:E8:29:9A:FE:BF","Channel":6,"RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-02-15:
Check SaveData

ascillato, 2020-02-15:
Go to the console and type :

```yaml
Savedata 1
Setoption36 0
Setoption65 1
```

And try again.

ascillato2, 2020-02-15:
If that don't solve your issue, please set weblog to 4 and report back the console output when restarting.

Or if you prefer, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-15 by ascillato2