arendst/Tasmota #7742: Blinds are not completely accurate while moving via HTTP requests

buenyamin-olgun, 2020-02-15:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When I start from ShutterPosition 12 and set ShutterPosition 9 afterwards via HTTP Request it is responding that the current value is 8.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): WF-CS02 Curtain Switch
- [X] Tasmota binary firmware version number used: 8.1.0
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Gitpod
- [X] Flashing tools used: tuya-convert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:00:27 CMD: Backlog Template; Module; GPIO 255
00:00:28 RSL: stat/tasmota/RESULT = {"NAME":"WF-CS02 Tuya","GPIO":[157,0,53,11,23,18,0,0,17,21,54,22,52],"FLAG":0,"BASE":18}
00:00:28 RSL: stat/tasmota/RESULT = {"Module":{"0":"WF-CS02 Tuya"}}
00:00:28 RSL: stat/tasmota/RESULT = {"GPIO0":{"157":"LedLink"},"GPIO1":{"0":"None"},"GPIO2":{"53":"Led2"},"GPIO3":{"11":"Switch3"},"GPIO4":{"23":"Relay3"},"GPIO5":{"18":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"17":"Button1"},"GPIO13":{"21":"Relay1"},"GPIO14":{"54":"Led3"},"GPIO15":{"22":"Relay2"},"GPIO16":{"52":"Led1"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
00:01:31 CMD: Backlog Rule1; Rule2; Rule3
00:01:31 RSL: stat/tasmota/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":454,"Rules":"on Power3#State=1 do backlog ShutterStop1; Power3 0 endon"}
00:01:31 RSL: stat/tasmota/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
00:01:32 RSL: stat/tasmota/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:01:55 CMD: Status 0
00:01:55 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["kueche-jalousie","kueche-jalousie2","kueche-jalousie3"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,4,4,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
00:01:55 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://192.168.0.100/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:56","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":226,"SaveAddress":"FA000"}}
00:01:55 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2020-01-04T16:52:58","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"421/699"}}
00:01:55 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["UPC3D1EAF3_2.4GHz",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000E02B","2805C8000100060000005A00000000000000","40000000","00000000"]}}
00:01:55 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8EC86383","043683A1","200217C5","010008C0","00007981"],"Drivers":"1,2,3,4,6,8,10,12,16,18,19,20,21,22,24,26,27,29","Sensors":"1,2,3,4,5,6,34"}}
00:01:55 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6624","IPAddress":"192.168.0.94","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"68:C6:3A:E1:F9:E0","Webserver":2,"WifiConfig":4}}
00:01:55 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E1F9E0","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:01:55 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:01:55 1970","Local":"Thu Jan 01 00:01:55 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00"}}
00:01:55 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:01:55","Switch3":"OFF","Shutter1":{"Position":8,"direction":0}}}
00:01:55 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:01:55","Uptime":"0T00:01:56","UptimeSec":116,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"UPC3D1EAF3_2.4GHz","BSSId":"DC:53:7C:F5:07:AC","Channel":11,"RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
23:41:34 HTP: Command
23:41:34 SRC: WebCommand from 192.168.0.122
23:41:34 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data ""
23:41:34 SHT: Pos. in: payload  (0), payload -99, idx 1, src 8
23:41:34 RSL: stat/tasmota/RESULT = {"ShutterPosition1":12}
23:41:38 HTP: Command
23:41:38 SRC: WebCommand from 192.168.0.122
23:41:38 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "9"
23:41:38 SHT: Pos. in: payload 9 (1), payload 9, idx 1, src 8
23:41:38 SHT: lastsource 8:, real 16142, target 15248, payload 9
23:41:38 SHT: Start in dir 0
23:41:38 SRC: Shutter
23:41:38 SHT: Switched relay: 2 by Shutter
23:41:38 RSL: stat/tasmota/RESULT = {"POWER2":"ON"}
23:41:38 RSL: stat/tasmota/POWER2 = ON
23:41:38 RSL: stat/tasmota/RESULT = {"ShutterPosition1":9}
23:41:39 SRC: Shutter
23:41:39 SHT: Switched relay: 2 by Shutter
23:41:39 RSL: stat/tasmota/RESULT = {"POWER2":"OFF"}
23:41:39 RSL: stat/tasmota/POWER2 = OFF
23:41:39 SHT: Shutter 1: Real. 15192, Start: 16142, Stop: 15248, dir -1, motordelay 0, rtc: 0.5 [s], freq 952
23:41:39 RSL: stat/tasmota/SHUTTER1 = 8
23:41:39 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":8,"direction":0}}
23:41:39 CFG: Saved to flash at F4, Count 224, Bytes 4096
23:41:43 HTP: Command
23:41:43 SRC: WebCommand from 192.168.0.122
23:41:43 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data ""
23:41:43 SHT: Pos. in: payload  (0), payload -99, idx 1, src 8
23:41:43 RSL: stat/tasmota/RESULT = {"ShutterPosition1":8}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Go to ShutterPosition 12 by:
http://192.168.0.94/cm?cmnd=ShutterPosition%2012

Then go to ShutterPosition 9 by
http://192.168.0.94/cm?cmnd=ShutterPosition%209
You'll see that it is reporting ShutterPosition == 9:
```
{"POWER1":"ON","ShutterPosition1":9}
```

But if you request the current position again than it is reporting 8:
http://192.168.0.94/cm?cmnd=ShutterPosition
```
{"ShutterPosition1":8}
```

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
ShutterPosition should return 9 if position was previously set to 9

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I didn't test it too often because the motor gets overheated by time so I don't know it this is reproducible in every case.

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-02-16:
Yes this is absolutely correct. With shutterposition you set the position you want to have, but you get reported the actual position. This can be a slightly different, but correct one. It can be rounding issues. Internally the shutter is much more accurate than 1..100. but also it can be an issue because you only make this small changes. This stresses the software a bit, because it cannot start and stop that fast.if you start with 50 and then go to 9 it should work if there is no rounding issue.

ascillato2, 2020-02-24:
Closing this issue as it is solved.

Thanks for sharing your ideas.

closed 2020-02-24 by ascillato2