arendst/Tasmota #7745: Stack Mode

ITTV-tools, 2020-02-16:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
If you have to control lots of Relays in one cabinet (multiple Sonoff 4CH for example) you need many wifi connections and configure them individuality. This costs "much" time to maintain and can be a disadvantage for the access point. Especially if you have also many other devices connected to one AP. 
**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I'd thought about a stacking system (like it can be done with Cisco Switches for example). We could use 1-Wire, I2C or someting similar to bind multiple esp/sonoffs together and make one master. 
The master ESP gets the wifi connection and controls all other ESPs/Sonoffs.
All Relays can be configured via one Webinterface and MQTT Messages are getting forwarded to the right slave ESP

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2020-02-16:
The code impact would be massive. There is little chance this will happen unless you propose a PR.

mgerald21, 2020-02-16:
What about using WiFi.mode(WIFI_AP_STA); 
One device can serve as soft access point for the others?

ascillato2, 2020-02-16:
That is mesh network and also requires a lot of coding, flash and RAM.

ascillato, 2020-02-16:
So far, an approach with limited control can be done if you connect all your devices by serial (tx line of master to the rx lines of the rest) and using rules event you can receive commands in the master that can be retransmitted by serial. Then also using rules in your slave devices, you can distinguish if a power on or off must be performed and in which device.

ascillato, 2020-02-16:
Another approach is to use only 1 nodemcu and a 8 relays board.

Another approach is to use only 1 nodemcu and a port expander (already supported in Tasmota) and several relay boards

ascillato2, 2020-02-20:
Closing as nobody is working on this.

Thanks for sharing your ideas.

closed 2020-02-20 by ascillato2