arendst/Tasmota #7746: SetOption57 changing APs with only 1dbm difference instead of 10dbm

DavinKD, 2020-02-16:
### PROBLEM DESCRIPTION
SetOption57 is causing the device to switch APs when there is only a 1dbm difference instead of 10

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ X] Device used (e.g., Sonoff Basic): Generic bulb
- [X ] Tasmota binary firmware version number used: 8.1.0.8
  - [ X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ X] Flashing tools used: TuyaConvert
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

12:58:49 MQT: stat/0827/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"0827","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:58:49 MQT: stat/0827/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"BedroomLamps","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T11:37:00","StartupUTC":"2020-02-16T08:21:49","Sleep":50,"CfgHolder":4617,"BootCount":38,"BCResetTime":"2020-02-11T15:31:04","SaveCount":697,"SaveAddress":"F7000"}}
12:58:49 MQT: stat/0827/STATUS2 = {"StatusFWR":{"Version":"8.1.0.7(9814468-tasmota)","BuildDateTime":"2020-02-10T23:00:10","Boot":4,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"326/699"}}
12:58:49 MQT: stat/0827/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<<redacted>>",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000080","00000000"]}}
12:58:49 MQT: stat/0827/STATUS4 = {"StatusMEM":{"ProgramSize":569,"Free":432,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:58:49 MQT: stat/0827/STATUS5 = {"StatusNET":{"Hostname":"0827-0827","IPAddress":"192.168.0.235","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"CC:50:E3:50:E3:3B","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
12:58:49 MQT: stat/0827/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.40","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_50E33B","MqttUser":"DVES_USER","MqttCount":11,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:58:49 MQT: stat/0827/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 16 19:58:49 2020","Local":"Sun Feb 16 12:58:49 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"-07:00","Sunrise":"23:57","Sunset":"10:11"}}
12:58:49 MQT: stat/0827/STATUS10 = {"StatusSNS":{"Time":"2020-02-16T12:58:49"}}
12:58:49 MQT: stat/0827/STATUS11 = {"StatusSTS":{"Time":"2020-02-16T12:58:49","Uptime":"0T11:37:00","UptimeSec":41820,"Heap":24,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":11,"POWER":"ON","Dimmer":5,"Color":"0D00000000","HSBColor":"0,100,5","White":0,"CT":153,"Channel":[5,0,0,0,0],"Scheme":0,"Fade":"OFF","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"<<redacted>>","BSSId":"86:83:C2:03:48:EC","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":4,"Downtime":"0T00:00:31"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

12:41:45 MQT: tele/0827/STATE = {"Time":"2020-02-16T12:41:45","Uptime":"0T11:19:56","UptimeSec":40796,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":10,"POWER":"ON","Dimmer":5,"Color":"0D00000000","HSBColor":"0,100,5","White":0,"CT":153,"Channel":[5,0,0,0,0],"Scheme":0,"Fade":"OFF","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"<<redacted>>","BSSId":"<<redacted>>","Channel":6,"RSSI":76,"Signal":-62,"LinkCount":3,"Downtime":"0T00:00:28"}}
12:46:25 WIF: Checking connection...
12:46:29 WIF: Network (re)scan started...
12:46:30 WIF: Checking connection...
12:46:31 WIF: Network 3, AP1, SSId <<redacted>>, Channel 1, BSSId <<redacted>>, RSSI -56, Encryption 1
12:46:31 WIF: Network 6, AP1, SSId <<redacted>>, Channel 6, BSSId <<redacted>>, RSSI -56, Encryption 1
12:46:31 WIF: Network 12, AP1, SSId <<redacted>>, Channel 10, BSSId <<redacted>>, RSSI -80, Encryption 1
12:46:31 WIF: Network 16, AP1, SSId <<redacted>>, Channel 11, BSSId <<redacted>>, RSSI -55, Encryption 1
12:46:31 WIF: Connecting to AP1 <<redacted>> Channel 11 BSSId <<redacted>> in mode 11N as 0827-0827...
12:46:35 MQT: Attempting connection...
12:46:35 MQT: Connected

```

### TO REPRODUCE
Turn on setoption57

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
It should only switch APs if there is a 10dbm difference in RSSI

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-02-17:
Agree.

Will check.

arendst, 2020-02-17:
It seems to be 5 now instead of 1.

Unless you defined ``WIFI_RSSI_THRESHOLD`` somewhere else as 1.

Pls check.

DavinKD, 2020-02-17:
I'm using pre-compliled version and did not change that constant.

DavinKD, 2020-02-17:
Also, I cut out all the APs that were not my SSID above.  Let me know if that information would be helpful and I can re-create.

DavinKD, 2020-02-17:
I think part of the issue is the data may be stale.  From looking at support_wifi.ino, it seems to not be using the dbm from the currently connected AP for comparison, it's using the dbm from some other point in time?  (Sorry, I'm not super familiar with the code).  So in my case, the last teleperiod dbm was -62 and there were other APs with a 5dbm better signal than 062, but during the scan my currently connected AP was -56 which was actually better than the -55 of the channel 11 AP it switched to.  Maybe move the calculation of Wifi.bes_network_db inside the for loop and use the dbm of the currently connected AP as found in the scan?

DavinKD, 2020-02-17:
Maybe something like this around line 289 ish?

     for (uint32_t i = 0; i < wifi_scan_result; ++i) {
        String ssid_scan;
        int32_t rssi_scan;
        uint8_t sec_scan;
        uint8_t* bssid_scan;
        int32_t chan_scan;
        bool hidden_scan;

        WiFi.getNetworkInfo(i, ssid_scan, sec_scan, rssi_scan, bssid_scan, chan_scan, hidden_scan);

        if (bssid_scan == Wifi.BSSID()) {  // SSID match
        	Wifi.best_network_db = rssi_scan
		    if (Wifi.best_network_db < -WIFI_RSSI_THRESHOLD) {
		      Wifi.best_network_db += WIFI_RSSI_THRESHOLD;
		    }
	}        	
     }
     for (uint32_t i = 0; i < wifi_scan_result; ++i) {
	......

DavinKD, 2020-02-19:
Here's another thought.  The reason the threshold was changed from 10 to 5 was for devices with a weak connection, right?  Maybe make the threshold variable depending on the range of the rssi.  So, if the rssi is in the good range, the threshold is 10.  If the rssi is in a bad range, the threshold is 5.

https://www.metageek.com/training/resources/understanding-rssi.html

DavinKD, 2020-02-19:
Here's the code that seems to work good in my environment.  There may be ways to make it more efficient, but again, I'm not that familiar with the entire code base.  The code below was inserted at line 289 in support_wifi.ino.

Let me know if  you'd like me to do a pull request.

DavinKD, 2020-02-19:
for (uint32_t i = 0; i < wifi_scan_result; ++i) {
			String ssid_scan;
			int32_t rssi_scan;
			uint8_t sec_scan;
			uint8_t* bssid_scan;
			int32_t chan_scan;
			bool hidden_scan;
			bool bMatch;

			WiFi.getNetworkInfo(i, ssid_scan, sec_scan, rssi_scan, bssid_scan, chan_scan, hidden_scan);

			bMatch=false;
			for (uint32_t j = 0; j < MAX_SSIDS; j++) {
          		if (ssid_scan == SettingsText(SET_STASSID1 + j)) {
          			bMatch=true;
          		}
          	}
          	if (bMatch){
				for (uint32_t i = 0; i < sizeof(Wifi.bssid); i++) {
				  if (bssid_scan[i] != Wifi.bssid[i]) {
					bMatch=false;
				  }
				}
			}
       		if (bMatch) {  // SSID match
				Wifi.best_network_db = rssi_scan;
				if (Wifi.best_network_db < -WIFI_RSSI_THRESHOLD) {
					if (Wifi.best_network_db > -70) {
						Wifi.best_network_db += WIFI_RSSI_THRESHOLD;
					}
					else {
						Wifi.best_network_db += WIFI_RSSI_THRESHOLD-5;
					}
				}
 			   AddLog_P2(LOG_LEVEL_DEBUG, "best_network_db set to %i)", Wifi.best_network_db);
			}
		}

arendst, 2020-02-19:
Thx. I'll investigate your suggestion

arendst, 2020-02-19:
Could you pls check the following code for same functionality. It replaces the current lines 248 until 267 in the current released file.
```
  // Init scan when connected
  if (2 == Wifi.scan_state) {
    uint8_t* bssid = WiFi.BSSID();                  // Get current bssid
    memcpy((void*) &Wifi.bssid, (void*) bssid, sizeof(Wifi.bssid));
    Wifi.best_network_db = WiFi.RSSI();             // Get current rssi and add threshold
    if ((Wifi.best_network_db < -WIFI_RSSI_THRESHOLD) && (Wifi.best_network_db > -70)) {
      Wifi.best_network_db += WIFI_RSSI_THRESHOLD;
    }
    Wifi.scan_state = 3;
  }
  // Init scan
  if (3 == Wifi.scan_state) {
    if (WiFi.scanComplete() != WIFI_SCAN_RUNNING) {
      WiFi.scanNetworks(true);                      // Start wifi scan async
      Wifi.scan_state++;
      AddLog_P2(LOG_LEVEL_DEBUG, PSTR(D_LOG_WIFI "Network (re)scan started (%i)..."), Wifi.best_network_db);
      return;
    }
  }
  int8_t wifi_scan_result = WiFi.scanComplete();
```

sfromis, 2020-02-19:
I don't know if this is really related, but I've noticed another signal strength issue recently. When doing a "Scan for wifi networks", I get a value like 94% (-59 dBm). Reading the code in WifiGetRssiAsQuality, I found that the conversion is rssi% = 2 * ( rssi:dBm + 100 ), for values from -50 to -100. According to that, -59 dBm should be 82% instead of 94%. While this does not make me loose sleep, it does indicate that something "less obvious" is going on around signal quality.

While my "off" value is not too bad, what donald reported recently looks a **lot** more strage.
https://discordapp.com/channels/479389167382691863/479389167382691865/679347156557234177

On the same screenshot, he has
-47 dBm giving both 14%, 16%, 18% and 26%
-50 dBm giving both 44%, 62% and 100%

If signal quality is handled very weirdly in one place, it occurred to be that funny stuff could also be happening in other places, like here.

DavinKD, 2020-02-19:
@arendst I feel the best_network_db should be done during the scan in wifi.state=5.  I've seen as much as a 6db difference in the currently connected AP from the time of state 2 to the time of state 5.  Doing it in state 5 is the best apples to apples comparison.

arendst, 2020-02-20:
I reverted the changes made in version 8.1.0.5 as others also reported different issues.

The reverted code should adhere to the 10db difference in RSSI and should work as before.

@sfromis regrading the difference between rssi and signal strength it is caused by different calls to Wifi.RSSI(). Every call would return different values so the relation between the two would seem wrong. This has been solved by not calling the RSSI() again but using the same value for both RSSI and signal strength.

Remember, RSSI on ESP8266 is not exact science, every call returns another value in a range of -+10

ascillato2, 2020-02-21:
Closing this issue as the changes affecting this has been reverted.

closed 2020-02-21 by ascillato2