arendst/Tasmota #7749: Configuring Relay1(21) impacts I2CScan ability to read devices on bus.

politick, 2020-02-17:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
If I configure a " Relay1 (21) ", then the HTU21 on the I2C is no longer detected.  If I issue an I2CScan withOUT the Relay1 configured, then the HTU21 is detected properly.  Once the Relay1 (21) is configured, I2Cscan no longer detects any device on the bus.  If I configure the Relay1 as a LED1 (52) instead, then I can issue in the console:  "LedPower1 1"  and "LedPower1 0" and that turns On/Off the relay with no problem and I can still read the temperature and the humidity from the HTU21.  I've discussed this on DiscordApp and I've followed the suggestions of @barbudor and @sfromis.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____ Wemos D1 Mini knockoff:  "IZOKEE D1 Mini"
- [x ] Tasmota binary firmware version number used: _____ 8.1.0 Doris
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ x ] Flashing tools used: _____ C:\Data\HomeAssistant\sonoff>"c:\Program Files\esptool-2.8\esptool.py" --port COM6 erase_flash
C:\Data\HomeAssistant\sonoff>"c:\Program Files\esptool-2.8\esptool.py" --port COM6 write_flash -fs 1MB -fm dout 0x0 tasmota-V8.1.0-Doris.bin
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
17:00:58 CMD: Backlog Template; Module; GPIO 255
17:00:58 MQT: experi1/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
17:00:58 MQT: experi1/stat/RESULT = {"Module":{"18":"Generic"}}
17:00:58 MQT: experi1/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"5":"I2C SCL"},"GPIO5":{"6":"I2C SDA"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
17:02:13 CMD: Backlog Rule1; Rule2; Rule3
17:02:14 MQT: experi1/stat/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
17:02:14 MQT: experi1/stat/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
17:02:14 MQT: experi1/stat/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:02:57 CMD: Status 0
17:02:57 MQT: experi1/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["experi1"],"Topic":"experi1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:02:57 MQT: experi1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:21","StartupUTC":"2020-02-17T16:00:36","Sleep":50,"CfgHolder":4617,"BootCount":66,"SaveCount":211,"SaveAddress":"F9000"}}
17:02:57 MQT: experi1/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"393/699"}}
17:02:57 MQT: experi1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Politick",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000088","00000000"]}}
17:02:57 MQT: experi1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
17:02:57 MQT: experi1/stat/STATUS5 = {"StatusNET":{"Hostname":"experi1-4793","IPAddress":"192.168.10.138","Gateway":"192.168.10.2","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.2","Mac":"EC:FA:BC:C0:92:B9","Webserver":2,"WifiConfig":4}}
17:02:57 MQT: experi1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.8","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C092B9","MqttUser":"hass","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:02:57 MQT: experi1/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Feb 17 16:02:57 2020","Local":"Mon Feb 17 17:02:57 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:55","Sunset":"18:13"}}
17:02:57 MQT: experi1/stat/STATUS10 = {"StatusSNS":{"Time":"2020-02-17T17:02:57"}}
17:02:57 MQT: experi1/stat/STATUS11 = {"StatusSTS":{"Time":"2020-02-17T17:02:57","Uptime":"0T00:02:21","UptimeSec":141,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Politick","BSSId":"82:2A:A8:54:6B:A5","Channel":1,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Solder HTU21 to D2 & D1 as I2C SCL & I2C SDA
Configure module as Generic (18) and 
D2 GPIO4 as  I2C SCL(5)
D1 GPIO5 as  I2C SDA(6)
Everything works OK, then NO NEED to install physical relay, just need to configure
Relay1 (21) on any free PIN example: D8 GPIO15
Then the HTU21 is no longer detected on the I2C bus

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The HTU21 should still be detected on the I2C bus when a relay (21) is configured

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Please visit this page near the bottom, just above the RFID
https://wiki.politick.ca/pages/viewpage.action?pageId=8454628

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

politick, 2020-02-19:
[image]

A picture of the 2 units that are reacheable

arendst, 2020-02-19:
Cannot reproduce:
```
10:08:05 CMD: gpio
10:08:05 MQT: stat/wemos7/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"56":"Led1i"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"51":"IRrecv"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}
10:08:19 CMD: i2cscan
10:08:19 MQT: stat/wemos7/RESULT = {"I2CScan":"Device(s) found at 0x23 0x40"}
10:08:23 CMD: status 8
10:08:23 MQT: stat/wemos7/STATUS8 = {"StatusSNS":{"Time":"2020-02-19T10:08:23","HTU21":{"Temperature":22.3,"Humidity":40.3},"BH1750":{"Illuminance":45},"TempUnit":"C"}}
```
In the above case both a BH1750 and a HTU21D are connected to the I2C bus. On your request a relay is configured but not connected. Wemos is powered by an external 5V USB power supply and monitors the livingroom while also accepting Remote Control IR commands.

politick, 2020-02-19:
Can you try on my device? Try at
https://wsgi.politick.ca/esp3/
Or
https://wsgi.politick.ca/dinette/
User: admin
Pass:  barbudor

barbudor, 2020-02-19:
Hi Theo @arendst 
I connected to @politick's device and was able to see it by myself : with 2ndary GPIO as 'relay1' no HTU sensor, as 'led1' it works. 
When I disabled all 'I2Cdriver' but the HTU one (9) it started to work (originaly 7,8,10,11,15 and 20 were also enabled). 
I believe that something in driver 9 gets confused by Relay1 and prevent detection. 
For now @politick issue is fixed.

arendst, 2020-02-19:
Apparently the SHT1X driver is spoiling your environment. The SHT1X can be found all over the I2C bus and I don't like the driver as it is. 

By disabling the driver using command ``I2cDriver8 0`` you're good to go.

politick, 2020-02-19:
???? I am utterly confused???? Just tried again on esp3 and works well now. Did someone for something in the console as I was typing the last comment?

arendst, 2020-02-19:
We must have been typing all three. Good luck with your new environment

barbudor, 2020-02-19:
@politick not me. I tested only on 'dinette'

politick, 2020-02-19:
Ok thx! Can I remove the devices from the internet more?  Thx very much!

arendst, 2020-02-19:
I was on esp3

arendst, 2020-02-19:
Alright with me

ascillato2, 2020-02-20:
@politick 

Any news on this? Is your problem solved?

politick, 2020-02-20:
Yes it was solved by disabling the I2CDriver8 with the command:
i2cdriver8 0

Thank you, Great product & Support !
Martin Politick.

closed 2020-02-20 by politick