arendst/Tasmota #7751: PCF8574 shutter - 12 (or more) shutters per one esp module - plizzz

abaszczok, 2020-02-18:
Hi

I'm using following setup
1) Wemos D1 min + Tasmota 8.1.0.8 (self compiled with #shutter and #pcf8574 options)

I'm using at home some old version of Stefan_Bode fork (6,3) for front gate..and I'm completely satisfied...
What I discovered lately is that Stefan completely moved to main code (Tasmota) .. and deleted his pages and issue sites.. (sad)

Currently I wanted to switch my shutters from pure I2C +RPI to Tasmota/shutter/Pcf8574

I managed to startup PCF options.. I set all of them as output and managed to setup 4 shutters..
And than surprize... thats all folks - "shutterrelay5" - command error .

I'm using 3xPCF connected to  D1mini.  - wanted to have 12 shutters controlled from one esp.

My request is - could you please add more shutter devices into one project...

PS If I remember correctly Stefan stated that up to 32 shutters were available in his version via PCF..

Below you can see what I managed to confiure... (only 4xshutters)

[image]

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

TrDA-hab, 2020-02-19:
@abaszczok 
- where did you find the information that shutterrelay5 should work? ... or rather give a link to an article on which you did the firmware setup?
- can you add an electric circuit to your blinds?

Jason2866, 2020-02-19:
Stefan did not delete. This seems to be his latest not sync`d version
https://github.com/stefanbode/Sonoff-Tasmota/tree/f4c5f5568f44edd714e0ad44cce7dd28c725b19d

abaszczok, 2020-02-20:
Hi. I know that tasmota is providing only solution for 4 shutters. But i the past on Stefan bode wiki there was information that multiple shutters controll was available via pcf I2C modulles. Today there is only link to tasmota master wiki. Also i can't find issue list form Stefan. Therefore I requested for extended functionality.

Jason2866, 2020-02-20:
Controlling 4 shutters in Tasmota is a design decision, was made because of limited flash and ram.

arendst, 2020-02-20:
As said by @Jason2866 it's a Tasmota design decision to "only" support four shutters.

You'll have to revert to Stefan's code if more are needed.

abaszczok, 2020-02-20:
Guys.. The problem is that Stefan Code is not available... Need to ask Stef' to get project. @arendst - thx a lot. Regarding flash ram  - In case of  D1 mini (4mb) - shouldn't be a problem to get it working. I'm just wondering how many bytes of payload is it possible to transfer via mqtt..

meingraham, 2020-02-20:
Adam,

In the ESP82xx, there is flash memory, RAM, and iRAM. The 4MB D1 Mini you reference is the size of flash memory. That's not the issue. As stated prior, the issue is the volatile RAM available. That is "fixed" in the ESP82xx design. This "memory" is something you probably don't need to try to "educate" @arendst on as I'm fairly certain he understands the options and constraints better than you or me.

With broader use of the shutter feature in Tasmota, additional considerations were made which affect a larger share of users and thus support was constrained to four shutters per ESP device. Stefan participated in porting the code and supports the direction Tasmota took in this regard.

Stefan's fork now makes use of the shutter features in Tasmota master rather than continue to support it in his fork. However, all the commits from his prior versions are still present in his repository. @Jason2866 provided you with the link to the code commit that you are looking for (i.e., it IS available). You can compile that codebase an use it for what you are requesting. Just don't expect support from Stefan at this point.

Since your use case appears to be an extreme edge case, further dialog for your request for support of more shutters than four with a single ESP device will most likely be fruitless for you. One recommendation for you to meet your requirements is to use multiple ESP devices and perform any shutter synchronization via Tasmota rules or via a central home automation hub.

If you have questions on how to do this, please visit the Tasmota Discord chat to discuss. Please close this issue.

Regards.

abaszczok, 2020-02-20:
Thx. Again for explaonation.. Of course i mixed memory types.. Etc.. In this case i will switch to more ESP modulles.. Thx guys for your support... Lets close this.. Regards

closed 2020-02-20 by abaszczok

stefanbode, 2020-03-19:
Sorry missed that mail. Will not support more than 4 shutter because there is plenty of space needed for every shutter to store all the variable. You can build your own version. Anyhow this will become incompatible with the main branch. The amount of changes are somehow 10-15 lines. Maximum are 16 shutters because of 32bit maximum number size.