arendst/Tasmota #7753: S20: no power off with timer at sunrise

Jambola77, 2020-02-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I want to start the timer at 06:30 to power on and then sun rise time to power off the lamp.
Power on works great, but power off does not work, the switch stays in power on state.

What do I have to configure to switch off the lamp at sunrise?
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): S20_____
- [ x] Tasmota binary firmware version number used: 8.1.0_____
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
13:09:54 CMD: Backlog Template; Module; GPIO 255
13:09:54 RSL: stat/sonoff/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
13:09:54 RSL: stat/sonoff/RESULT = {"Module":{"8":"Sonoff S2X"}}
13:09:54 RSL: stat/sonoff/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
13:10:54 CMD: Backlog Rule1; Rule2; Rule3;
13:10:54 RSL: stat/sonoff/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
13:10:54 RSL: stat/sonoff/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
13:10:55 RSL: stat/sonoff/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
13:11:47 RSL: stat/sonoff/STATUS = {"Status":{"Module":8,"FriendlyName":["Sonoff1"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:11:47 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-DE.bin","RestartReason":"Power on","Uptime":"1T17:28:30","StartupUTC":"2020-02-16T18:43:17","Sleep":50,"CfgHolder":4617,"BootCount":61,"SaveCount":708,"SaveAddress":"F6000"}}
13:11:47 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019.12.25 12:47:07","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"320/699"}}
13:11:47 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":1,"LogHost":"","LogPort":514,"SSId":["HalloNachbar",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000001","00000000"]}}
13:11:47 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
13:11:47 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1630","IPAddress":"192.168.1.85","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:79:86:5E","Webserver":2,"WifiConfig":2}}
13:11:47 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.45","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_79865E","MqttUser":"piet","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
13:11:47 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 18 12:11:47 2020","Local":"Tue Feb 18 13:11:47 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":99,"Sunrise":"07:33","Sunset":"17:34"}}
13:11:47 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2020-02-18T13:11:47"}}
13:11:47 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2020-02-18T13:11:47","Uptime":"1T17:28:30","UptimeSec":149310,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MySSID","BSSId":"38:10:D5:B9:56:6C","Channel":1,"RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The lamp should switch off at sunrise.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Th0maz, 2020-02-18:
You set the timer to add 7 hours and 5 minutes to the sunrise time.
Change it to + 00:00 and it will work ;)

Jambola77, 2020-02-18:
Hmm, that seems to be logical ;-)
Great thanks for the help!

closed 2020-02-18 by ascillato2