arendst/Tasmota #7754: Extending the ADS115 support - measurement of differential signals, using another gain

cph25, 2020-02-18:
**Have you looked for this feature in other issues and in the docs?**  
yes, didn' find a solution

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
the ads1115 implementation supports measurement of single ended signals only, in my project I need the measurement of differential signals based on another gain

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Additional configuration panel to switch between the measurement of single-ended and differential signals as well as the option to modify the gain.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
changing the code, i'm not sure if I can keep the pace of your changing process

**Additional context**  
_Add any other context or screenshots about the feature request here._
The ADS1115 allows measuring single-ended as well as differential signals.
As far I understand, the current Tasmota implementation supports the single-ended signal only. 
In order to measure wind speed based on a DC motor generating DC voltage (0V to 1.2V), I extended (hard coded) the xsns_12_ads1115_i2cdev.ino of Tasmota 6.6.  In the migration progress from Tasmota 6.6 to Tasmota 8.1 I learned:
1.	The module xsns_12_ads1115_i2cdev.ino has been removed, just the xsns_12_ads1115.ino survived.
2.	The xsns_12_ads1115.ino implementation strategy is completely different from the strategy of the xsns_12_ads1115_i2cdev.ino.
3.	The implementation uses As I have to measure the voltage between 0V and 1.2V only, I need the gain 2 rather than 2/3.

**(Please, remember to close the issue when the problem has been addressed)**

closed 2020-03-19 by stale[bot]

spcqike, 2020-06-23:
Hey,

I also play around with the ADS1115 and would like to see a possibility of using differential inputs.
looking for anwers, i came across this thread.

one possibility for the gain would be some kind of autorange, like it's done [here](http://shelvin.de/den-ads1115-16-bit-ad-wandler-am-arduino-mit-autorange-betreiben/)

of course, it would be better/useful to set the gain manually and, in case of autorange, it would be neccessary to know, what gain is used at the moment. except if the value isn't given in digits but in mV or V.

maybe someone how can do this could add something like this? :)

amayii0, 2022-01-02:
Hi,

Just found this by searching for differential inputs.
[xsns_12_ads1115.ino ](https://github.com/arendst/Tasmota/blob/development/tasmota/xsns_12_ads1115.ino) is a limited version based on [Adafruit ADS1X15](https://github.com/adafruit/Adafruit_ADS1X15).

The [sample provided by Adafruit ADS1X15](https://github.com/adafruit/Adafruit_ADS1X15/blob/master/examples/differential/differential.ino) is working just fine for my use case (NB: sample uses default I2C address 0x48).

I had a look at compiling my own firmware, but as of today, I don't yet have skills to "upgrade" the current xsns_12_ads1115.ino.

Adding a few links in case I've enough time to dev that:
https://github.com/adafruit/Adafruit_ADS1X15
https://tasmota.github.io/docs/Sensor-API/

Purpose is to integrate CT sensors for grid and solar panels in Home Assistant
https://www.home-assistant.io/blog/2021/08/04/home-energy-management/
They document use of ESPHome : https://esphome.io/components/sensor/ads1115.html
https://github.com/esphome/esphome/tree/dev/esphome/components/ads1115

amayii0, 2022-06-18:
Implemented by FransOv, to be tested :)
https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xsns_sensor/xsns_12_ads1115.ino#L293-L296