arendst/Tasmota #7755: Power dropout when KNX bus is scanned

Winni66, 2020-02-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I'm using Tasmota 8.1.0 and 8.1.8 on Teckin and Gosund SP111 device. Everything works fine with one exception. As KNX visualization I'm using edomi (edomi.de).
This software scans all KNX devices when restarting, to get the actual status of the device. Unfortunately during this process, the tasmota software seems to restart (even device was excluded from this initial scan). I already lowered data rate from 15/s to 7/s in edomi, but there was no change.
I don't know how long this behaviour is existing, because up to know there was no impact on the connected devices, during the reset of tasmota. There is a power loss of a few ms and now I have a device which don't compense this drop out.
Unfortunately there is no way for you to reproduce this problem I fear. Nevertheless I hope the information from console will help to find the issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund SP111, Teckin SS42
- [x] Tasmota binary firmware version number used: 8.1.0 and 8.1.8
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: over the air
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:37:09 CMD: Backlog Template; Module; GPIO 255
20:37:09 RSL: RESULT = {"NAME":"Teckin SS42","GPIO":[255,255,255,255,56,57,0,0,21,17,22,255,255],"FLAG":15,"BASE":18}
20:37:09 RSL: RESULT = {"Module":{"0":"Teckin SS42"}}
20:37:10 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"56":"Led1i"},"GPIO5":{"57":"Led2i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"17":"Button1"},"GPIO14":{"22":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
20:38:05 RSL: RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
20:38:05 RSL: RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
20:38:05 RSL: RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:38:28 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:38:28 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-knx.bin","RestartReason":"Exception","Uptime":"0T00:19:32","StartupUTC":"2020-02-18T19:18:56","Sleep":50,"CfgHolder":4617,"BootCount":32,"SaveCount":160,"SaveAddress":"F4000"}}
20:38:28 RSL: STATUS2 = {"StatusFWR":{"Version":"8.1.0(knx)","BuildDateTime":"2019-12-25T12:37:34","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"354/699"}}
20:38:28 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HACKERBOX-CAM",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["02008001","2805C8000100060000005A00000000000000","00000000","00000000"]}}
20:38:28 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":576,"Free":424,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","9FDAE197","042683B0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
20:38:28 RSL: STATUS5 = {"StatusNET":{"Hostname":"Teckin 01","IPAddress":"192.168.4.110","Gateway":"192.168.5.99","Subnetmask":"255.255.248.0","DNSServer":"8.8.8.8","Mac":"D8:F1:5B:CB:34:52","Webserver":2,"WifiConfig":4}}
20:38:28 RSL: STATUS7 = {"StatusTIM":{"UTC":"Tue Feb 18 19:38:28 2020","Local":"Tue Feb 18 20:38:28 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:53","Sunset":"18:14"}}
20:38:28 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-02-18T20:38:28"}}
20:38:28 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-02-18T20:38:28","Uptime":"0T00:19:32","UptimeSec":1172,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"HACKERBOX-CAM","BSSId":"02:9F:C2:2A:0F:7E","Channel":1,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}}
20:38:28 RSL: STATUS12 = {"StatusSTK":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025de03","4025dd98","4025dd3f","4025cea0","4025cec9","4025a934","40256c9d","4010170a","4025b62b","40000f58","4025b09e","402647ef","402640af","4026a157","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","40262f94","40262f45","40247b1e","40247b1e","401009b4","401024ac","4010575f","40106005","40106005","40247b1e"]}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:00 CFG: Loaded from flash at F9, Count 155
00:00:00 Project tasmota Tasmota Version 8.1.0(knx)-2_6_1
00:00:00 WIF: Connecting to AP1 HACKERBOX-CAM in mode 11N as Teckin 01...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on Teckin 01 with IP address 192.168.4.110
00:00:05 RSL: INFO1 = {"Module":"Teckin SS42","Version":"8.1.0(knx)","FallbackTopic":"cmnd/DVES_CB3452_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"Teckin 01","IPAddress":"192.168.4.110"}
00:00:05 RSL: INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025de03","4025dd98","4025dd3f","4025cea0","4025cec9","4025a934","40256c9d","4025b62b","40000f58","4025b09e","402647ef","402640af","4024111f","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","40262f94","40262f45","401009b4","401024ac","4010575f","40103043","40103224","401036e6","4000050c","40103408","402474ae"]}}
00:00:05 RSL: RESULT = {"POWER1":"ON"}
00:00:05 RSL: POWER1 = ON
00:00:05 RSL: RESULT = {"POWER2":"ON"}
00:00:05 RSL: POWER2 = ON
19:16:02 RSL: STATE = {"Time":"2020-02-18T19:16:02","Uptime":"0T00:00:11","UptimeSec":11,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"HACKERBOX-CAM","BSSId":"02:9F:C2:2A:0F:7E","Channel":1,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}
[... 20 lines cut ...]
00:00:05 RSL: INFO1 = {"Module":"Teckin SS42","Version":"8.1.0(knx)","FallbackTopic":"cmnd/DVES_CB3452_fb/","GroupTopic":"cmnd/tasmotas/"}
00:00:05 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"Teckin 01","IPAddress":"192.168.4.110"}
00:00:05 RSL: INFO3 = {"RestartReason":{"Exception":29,"Reason":"Exception","EPC":["4000df64","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4025de03","4025dd98","4025dd3f","4025cea0","4025cec9","4025a934","40256c9d","4010170a","4025b62b","40000f58","4025b09e","402647ef","402640af","4026a157","40000f49","40000f49","40000e19","40001878","40105865","4010586b","4010000d","40262f94","40262f45","4010307e","401031eb","401036e6","4000050c","40103408","402474cb","40247b1e","40280825"]}}
00:00:05 RSL: RESULT = {"POWER1":"ON"}
00:00:05 RSL: POWER1 = ON
00:00:05 RSL: RESULT = {"POWER2":"ON"}
00:00:05 RSL: POWER2 = ON
[... cut ...] F:C2:2A:0F:7E","Channel":1,"RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:06"}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
not possible without edomi

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
no power dropout when restarting edomi with bus scan

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-19:
Hi,

That is not a power dropout, you are having a restart because an exception 29 have occurred.

When your device restarts and it has the output turned on, the esp8266 turn some gpio to 0v. That is a hardware known issue. So far, that blink can't be solved by software.

Can you use any KNX snoop software to check which command is sending edomi?

Tasmota is prepared to answer inquiries, so that shouldn't be an issue.

First, let's try to clean the flash for any wrong data being left.

In the console do `reset 3` (this won't erase any setting, just a forced wifi parameters recalibration) and then power cycle your device (a restart is not enough, it needs to be de-energized and energized again).

Then, if the issue remains, please try `reset 5` (this will erase your settings leaving your wifi credentials only)

If that doesn't work, we will need to scan the network to see the actual telegram that is sending edomi.

Winni66, 2020-02-19:
Thank for you answer. I will do the checks as soon as I'm back home.

For snooping KNX bus I'm using ETS software, but there is not a single telegram adressed to the tasmota devices when starting edomi (after disabling initscan for those). But in that starting phase, edomi does a request for round about 350 GAs, and of course, the answers are also floodfilling the bus.
I fear that tasmota has a problem with that many telegrams in that short time.
The devices are brand new, so I don't expect changes in behaviour after reset, but I will do, for sure.

If we need to investigate ETS log, please be aware, that I don't want to post all my GAs of my domicil here in public, due to security reason. But I can send you directly.

best regards, I will keep you updated and thanks for support

Winni66, 2020-02-19:
I did the test, both resets didn't change anything. 
I have exported XML and CSV data from ETS from the beginning of the restart until the first reset.
How can I send you this information (only if you are willing to inspect)

kind regards
   Winni

ascillato, 2020-02-20:
You can send that to me on a private message on Discord.

ascillato, 2020-02-20:
I have downloaded your files and will try to reproduce it. Thanks.

Winni66, 2020-02-20:
Just one additional comment: Sometimes there are several restarts, sometimes none. In most cases it is one restart. If you think that I should log the whole sequence of restarting edomi, I will do.
Teckin device is listening on 8/6/8 and 8/6/9 and physical adress is 2.0.30

Jason2866, 2020-02-26:
Can you try this build http://thehackbox.org/tasmota/pre-2.7/

It uses Arduino feature/stage core (from yesterday). Many changes are done since the core 2.6.1,
which is the standard core Tasmota is build

Winni66, 2020-02-26:
Great, it seems to work. I was only able to do a remote test. The device with new version was stable, no  restart in console, the old once still do a restart.  I hope that therefore also the relais will be stable, I will check in the evening.

Winni66, 2020-02-26:
So, back home and I did some additional tests. Now 3 devices (2Gosund 1Teckin) is on new version. Bus scanning issue seems to be gone, but the reliability of KNX Wifi telegrams goes down from about 95% to 50% when triggering in frequence of 1/second. Hopefully this will be stable again in release! If you need any additional tests, please contact me. This issue can be closed.
Thanks a lot !

Jason2866, 2020-02-26:
Glad we found a solution!
You can close the issue since you opened.

closed 2020-02-26 by Winni66