arendst/Tasmota #7757: Aqara Wall Switch Not Responding

russellhq, 2020-02-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have an Aqara Wall switch, single switch no neutral and I have it paired with zigbee2tasmota. When I click the switch I can see the output on the console and it works fine. But when I try to send a power command, I get a response on the console but the switch doesn't turn on.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Here is the output from the console when I try to toggle the power on and off

`00:27:23 CMD: ZigbeeSend { "device":"0x46D6", "endpoint":"0x02", "send":{"Power": true} } 00:27:23 MQT: stat/tasmota/RESULT = {"ZigbeeSend":"Done"} 00:27:24 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"0006/F000":184549377,"Power":true,"LinkQuality":101}}} 00:27:25 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"Power":true,"LinkQuality":101}}} 00:27:42 CMD: ZigbeeSend { "device":"0x46D6", "endpoint":"0x02", "send":{"Power": false} } 00:27:42 MQT: stat/tasmota/RESULT = {"ZigbeeSend":"Done"} 00:27:43 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"0006/F000":184549377,"Power":false,"LinkQuality":96}}} 00:27:43 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"Power":false,"LinkQuality":96}}}`

Here's the output from the console when I control the switch from the wall:

`00:45:35 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"0006/F000":54973953,"Power":true,"LinkQuality":65}}} 00:45:35 MQT: tele/tasmota/RESULT = {"ZigbeeState":{"Status":32,"ActiveEndpoints":["0x03"]}} 00:45:36 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"Power":false,"Power2":true,"LinkQuality":54}}} 00:45:36 MQT: tele/tasmota/RESULT = {"ZigbeeState":{"Status":33,"Endpoint":"0x03","ProfileId":"0xC05E","DeviceId":"0x0210","DeviceVersion":2"InClusters":["0x1000","0x0000","0x0003","0x0004","0x0005","0x0006","0x0008","0x0300","0xFC0F"],"OutClusters":["0x0019"]}} 00:45:36 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x6EB0":{"Manufacturer":"OSRAM","ModelId":"CLA60 RGBW OSRAM","LinkQuality":46}}} 00:45:43 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"0006/F000":54973953,"Power":false,"LinkQuality":70}}} 00:45:44 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"Power":false,"Power2":true,"LinkQuality":65}}}`

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-19:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-19 by ascillato2

s-hadinger, 2020-02-19:
The switch responds normally to the Power command from a Zigbee perspective. 

Can you set `weblog 3` and post logs here? Maybe it’s the wrong endpoint.

s-hadinger, 2020-02-19:
Also please try with endpoint 3: `ZigbeeSend { "device":"0x46D6", "endpoint":"0x03", "send":{"Power": true} }`

russellhq, 2020-02-19:
Thanks. I tried all endpoints reported form the probe and only 2 got a response from the switch:

Here's the Weblog 3 output from endpoint 2.

```
09:41:28 CMD: Zigbeesend { "device":"0x46D6", endpoint: 2, "send": {"Power": true}}
09:41:28 SRC: WebConsole from 192.168.1.193
09:41:28 CMD: Group 0, Index 1, Command "ZIGBEESEND", Data "{ "device":"0x46D6", endpoint: 2, "send": {"Power": true}}"
09:41:28 ZigbeeSend: command_template = 0006!xx
09:41:28 SendZCLCommand_P: zcl_cmd = 0006!01
09:41:28 ZigbeeSend: command_final    = 0006!01
09:41:28 ZigbeeCmd_actual: ZigbeeZCLSend {"device":"0x46D6","endpoint":2,"send":"0006!01"}
09:41:28 ZigbeeSend: dstAddr 0x46D6, cluster 0x0006, endpoint 0x02, cmd 0x01, data 
09:41:28 ZIG: ZigbeeZNPSent 2401D6460201060001301E03110101
09:41:28 MQT: stat/tasmota/RESULT = {"ZigbeeSend":"Done"}
09:41:28 ZIG: ZigbeeZNPReceived 640100
09:41:28 ZIG: ZigbeeZNPSent 2401D6460201060001301E050001000000
09:41:28 ZIG: ZigbeeZNPReceived 640100
09:41:29 ZIG: ZigbeeZNPReceived 4480000101
09:41:29 ZIG: ZigbeeZNPReceived 448100000600D6460201006500F686A700000E18930A0000100100F0230100000BD6461D
09:41:29 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":147,"cmdid":"0x0A","payload":"0000100100F0230100000B"}}
09:41:29 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{"0006/0000":true,"0006/F000":184549377}}
09:41:29 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"0006/F000":184549377,"Power":true,"LinkQuality":101}}}
09:41:30 ZIG: ZigbeeZNPReceived 4480000101
09:41:30 ZIG: ZigbeeZNPReceived 448100000600D64602010065006F92A70000081801010000001001D6461D
09:41:30 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"0000001001"}}
09:41:30 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{"0006/0000":true}}
09:41:30 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"Power":true,"LinkQuality":101}}}
09:42:04 ZIG: ZigbeeZNPReceived 448100000A00D64601010065002932A90000051094000000D6461D
09:42:04 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":10,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x10","manuf":"0x0000","transact":148,"cmdid":"0x00","payload":"0000"}}
09:42:04 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{}}
09:42:04 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"LinkQuality":101}}}
```

And here's the output from endpoint 3

```
09:49:52 ZigbeeCmd_actual: ZigbeeZCLSend {"device":"0x46D6","endpoint":3,"send":"0006!00"}
09:49:52 ZigbeeSend: dstAddr 0x46D6, cluster 0x0006, endpoint 0x03, cmd 0x00, data 
09:49:52 ZIG: ZigbeeZNPSent 2401D6460301060001301E03110100
09:49:52 MQT: stat/tasmota/RESULT = {"ZigbeeSend":"Done"}
09:49:53 ZIG: ZigbeeZNPReceived 640100
09:49:53 ZIG: ZigbeeZNPSent 2401D6460301060001301E050001000000
09:49:53 ZIG: ZigbeeZNPReceived 640100
09:49:53 ZIG: ZigbeeZNPReceived 4480000101
09:49:54 ZIG: ZigbeeZNPReceived 4480000101
09:49:54 ZIG: ZigbeeZNPReceived 448100000600D6460301006500F29BBF0000081801010000001000D6461D
09:49:54 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"0000001000"}}
09:49:54 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{"0006/0000":false}}
09:49:54 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"Power":false,"LinkQuality":101}}}
09:50:20 ZIG: ZigbeeZNPReceived 448100000A00D646010100650008D7C0000005109D000000D6461D
09:50:20 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":10,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x10","manuf":"0x0000","transact":157,"cmdid":"0x00","payload":"0000"}}
09:50:20 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{}}
09:50:20 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"LinkQuality":101}}}
```

s-hadinger, 2020-02-19:
Thanks. Can you now press on the switch and report what is received with same log level?

russellhq, 2020-02-19:
Sure, I'll need to do that when I get home from work ;)

russellhq, 2020-02-19:
Ok, here's the output when I push the switch and the light turns on, then when I push it again and the light turns off:

Turn on
```17:17:05 ZIG: ZigbeeZNPReceived 448100000600D64602010031004A03C400000E18990A0000100100F02301D64603D6461D
17:17:05 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":153,"cmdid":"0x0A","payload":"0000100100F02301D64603"}}
17:17:05 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{"0006/0000":true,"0006/F000":54973953}}
17:17:05 ZIG: ZigbeeZNPReceived 45C1B06EB06E8AAFA800AA3EB07C8E
17:17:05 MQT: tele/tasmota/RESULT = {"ZigbeeState":{"Status":30,"IEEEAddr":"7CB03EAA00A8AF8A","ShortAddr":"0x6EB0","PowerSource":true,"ReceiveWhenIdle":true,"Security":false}}
17:17:05 ZIG: ZigbeeZNPSent 2505B06EB06E
17:17:05 ZIG: ZigbeeZNPReceived 650500
17:17:05 ZIG: ZigbeeZNPReceived 448100000600D6460401003600CA07C400000B189A0A0000100000001001D6461D
17:17:05 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":154,"cmdid":"0x0A","payload":"0000100000001001"}}
17:17:05 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{"0006/0000":false,"0006/0000+2":true}}
17:17:05 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"0006/F000":54973953,"Power":true,"LinkQuality":49}}}
17:17:06 ZIG: ZigbeeZNPReceived 4585B06E00B06E0103
17:17:06 ZIG: ZigbeeZNPSent 2504B06EB06E03
17:17:06 MQT: tele/tasmota/RESULT = {"ZigbeeState":{"Status":32,"ActiveEndpoints":["0x03"]}}
17:17:06 ZIG: ZigbeeZNPReceived 650400
17:17:06 ZIG: ZigbeeZNPReceived 4584B06E00B06E1C035EC010020209001000000300040005000600080000030FFC011900
17:17:06 MQT: tele/tasmota/RESULT = {"ZigbeeState":{"Status":33,"Endpoint":"0x03","ProfileId":"0xC05E","DeviceId":"0x0210","DeviceVersion":2"InClusters":["0x1000","0x0000","0x0003","0x0004","0x0005","0x0006","0x0008","0x0300","0xFC0F"],"OutClusters":["0x0019"]}}
17:17:06 ZIG: ZigbeeZNPSent 2401B06E0301000001301E0700010004000500
17:17:06 ZIG: ZigbeeZNPReceived 640100
17:17:07 ZIG: ZigbeeZNPReceived 4480000101
17:17:07 ZIG: ZigbeeZNPReceived 448100000000B06E03010029001716C400002218010104000042054F5352414D0500004210434C4136302052474257204F5352414DB06E1D
17:17:07 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"04000042054F5352414D0500004210434C4136302052474257204F5352414D"}}
17:17:07 ZIG: ZigbeeZCLRawReceived: {"0x6EB0":{"0000/0004":"OSRAM","0000/0005":"CLA60 RGBW OSRAM"}}
17:17:07 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x6EB0":{"Manufacturer":"OSRAM","ModelId":"CLA60 RGBW OSRAM","LinkQuality":41}}}
```

Turn Off
```17:18:52 ZIG: ZigbeeZNPReceived 448100000600D6460201003100401BC900000E18A10A0000100000F02301D64603D6461D
17:18:52 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":161,"cmdid":"0x0A","payload":"0000100000F02301D64603"}}
17:18:52 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{"0006/0000":false,"0006/F000":54973953}}
17:18:52 ZIG: ZigbeeZNPReceived 448100000600D64604010031000C20C900000B18A20A0000100000001001D6461D
17:18:52 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":162,"cmdid":"0x0A","payload":"0000100000001001"}}
17:18:52 ZIG: ZigbeeZCLRawReceived: {"0x46D6":{"0006/0000":false,"0006/0000+2":true}}
17:18:52 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"0006/F000":54973953,"Power":false,"LinkQuality":49}}}
'''

s-hadinger, 2020-02-19:
Thanks. I need the raw hex just before your logs, containing `ZigbeeZNPReceived`

Can you also upgrade to the latest version ? It will make it easier to analyze logs (there was a bug in this version)

russellhq, 2020-02-19:
Turn on
`20:27:29 ZIG: ZigbeeZNPReceived 448100000600D64602010031006CC2E600000E18860A0000100100F02301D64603D6461D`

Turn off
`20:27:35 ZIG: ZigbeeZNPReceived 448100000600D64602010031004E05E700000E18880A0000100000F02301D64603D6461D`

I'll update tasmota next

s-hadinger, 2020-02-19:
Thanks. This confirms that the endpoint is 2. 

I’m out of ideas here. All logs look good and zigbee2mqtt seems to send the same commands. 

There is an internal operation_mode parameter. Could you read parameter 0xFF22 from cluster 0, endpoint 1 or 2 (not sure). 

`ZbRead { "device":"0x46D6", "endpoint":1, "cluster":0, "read":"0xFF22" }`

`ZbRead { "device":"0x46D6", "endpoint":2, "cluster":0, "read":"0xFF22" }`

russellhq, 2020-02-19:
```

22:22:43 CMD: ZbRead { "device":"0x46D6", "endpoint":1, "cluster":0, "read":"0xFF22" }
22:22:43 SRC: WebConsole from 192.168.1.229
22:22:43 CMD: Group 0, Index 1, Command "ZBREAD", Data "{ "device":"0x46D6", "endpoint":1, "cluster":0, "read":"0xFF22" }"
22:22:43 ZIG: ZbZNPSent 2401D6460101000001301E0500010022FF
22:22:43 MQT: stat/tasmota/RESULT = {"ZbRead":"Done"}
22:22:43 ZIG: {"ZbZNPReceived":"640100"}
22:22:44 ZIG: {"ZbZNPReceived":"4480000101"}
22:22:44 ZIG: {"ZbZNPReceived":"448100000000D6460101006D00A3101400000618010122FF86D6461D"}
22:22:44 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x46D6","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":109,"securityuse":0,"seqnumber":0,"timestamp":1314979,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"22FF86"}}
22:22:44 ZIG: ZbZCLRawReceived: {"0x46D6":{}}
22:22:44 MQT: tele/tasmota/SENSOR = {"ZbReceived":{"0x46D6":{"LinkQuality":109}}}
22:22:44 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"LinkQuality":109}}}
```

```

22:23:32 CMD: ZbRead { "device":"0x46D6", "endpoint":2, "cluster":0, "read":"0xFF22" }
22:23:32 SRC: WebConsole from 192.168.1.229
22:23:32 CMD: Group 0, Index 1, Command "ZBREAD", Data "{ "device":"0x46D6", "endpoint":2, "cluster":0, "read":"0xFF22" }"
22:23:32 ZIG: ZbZNPSent 2401D6460201000001301E0500010022FF
22:23:32 MQT: stat/tasmota/RESULT = {"ZbRead":"Done"}
22:23:32 ZIG: {"ZbZNPReceived":"640100"}
22:23:32 ZIG: {"ZbZNPReceived":"4480000101"}
```

s-hadinger, 2020-02-19:
Thanks. The first one on endpoint 1 reports that the attribute is unsupported. 
I don't see any response for endpoint 2. Don't you have a ZbReceived just after the command?

russellhq, 2020-02-19:
This arrived a little later, check the time stamps
```

22:23:27 ZIG: {"ZbZNPReceived":"448100000A00D6460101006D004122160000051015000000D6461D"}
22:23:27 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":10,"srcaddr":"0x46D6","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":109,"securityuse":0,"seqnumber":0,"timestamp":1450561,"fc":"0x10","manuf":"0x0000","transact":21,"cmdid":"0x00","payload":"0000"}}
22:23:27 ZIG: ZbZCLRawReceived: {"0x46D6":{}}
22:23:27 MQT: tele/tasmota/SENSOR = {"ZbReceived":{"0x46D6":{"LinkQuality":109}}}
22:23:27 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"LinkQuality":109}}}
```

And this
```

22:24:29 ZIG: {"ZbZNPReceived":"448100000A00D64601010065002E17190000051016000000D6461D"}
22:24:29 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":10,"srcaddr":"0x46D6","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":101,"securityuse":0,"seqnumber":0,"timestamp":1644334,"fc":"0x10","manuf":"0x0000","transact":22,"cmdid":"0x00","payload":"0000"}}
22:24:29 ZIG: ZbZCLRawReceived: {"0x46D6":{}}
22:24:29 MQT: tele/tasmota/SENSOR = {"ZbReceived":{"0x46D6":{"LinkQuality":101}}}
22:24:29 MQT: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x46D6":{"LinkQuality":101}}}
```

s-hadinger, 2020-02-20:
Thanks. Those two are unrelated messages. Can you try again ZbRead for endpoint 2?

russellhq, 2020-02-20:
Tried again, but seem to get the same result:
```
09:45:57 CMD: ZbRead { "device":"0x46D6", "endpoint":2, "cluster":0, "read":"0xFF22" }
09:45:57 SRC: WebConsole from 192.168.1.193
09:45:57 CMD: Group 0, Index 1, Command "ZBREAD", Data "{ "device":"0x46D6", "endpoint":2, "cluster":0, "read":"0xFF22" }"
09:45:57 ZIG: ZbZNPSent 2401D6460201000001301E0500010022FF
09:45:57 MQT: stat/tasmota_14273C/RESULT = {"ZbRead":"Done"}
09:45:58 ZIG: {"ZbZNPReceived":"640100"}
09:45:58 ZIG: {"ZbZNPReceived":"4480000101"}
```

russellhq, 2020-02-20:
Also, here's the results from a probe:
```
09:52:39 CMD: Zbprobe 1
09:52:39 SRC: WebConsole from 192.168.1.193
09:52:39 CMD: Group 0, Index 1, Command "ZBPROBE", Data "1"
09:52:39 ZIG: ZbZNPSent 2505D646D646
09:52:39 MQT: stat/tasmota_14273C/RESULT = {"ZbProbe":"Done"}
09:52:39 ZIG: {"ZbZNPReceived":"650500"}
09:52:40 ZIG: {"ZbZNPReceived":"4585D64600D6460701020304050608"}
09:52:40 ZIG: ZbZNPSent 2504D646D64601
09:52:40 ZIG: ZbZNPSent 2504D646D64602
09:52:40 ZIG: ZbZNPSent 2504D646D64603
09:52:40 ZIG: ZbZNPSent 2504D646D64604
09:52:40 ZIG: ZbZNPSent 2504D646D64605
09:52:40 ZIG: ZbZNPSent 2504D646D64606
09:52:40 ZIG: ZbZNPSent 2504D646D64608
09:52:40 MQT: tele/tasmota_14273C/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x01","0x02","0x03","0x04","0x05","0x06","0x08"]}}
09:52:40 ZbInput discarding byte C0
09:52:40 ZbInput discarding byte D9
09:52:40 ZbInput discarding byte 04
09:52:40 ZbInput discarding byte 00
09:52:40 ZbInput discarding byte 60
09:52:40 ZIG: {"ZbZNPReceived":"650400"}
09:52:40 ZIG: {"ZbZNPReceived":"650400"}
09:52:40 ZIG: {"ZbZNPReceived":"650400"}
09:52:40 ZIG: {"ZbZNPReceived":"650400"}
09:52:40 ZIG: {"ZbZNPReceived":"650400"}
09:52:40 ZIG: {"ZbZNPReceived":"650400"}
09:52:41 ZIG: {"ZbZNPReceived":"4584D64600D6461A01040106000206000003000100020019000A000300000A001900"}
09:52:41 MQT: tele/tasmota_14273C/RESULT = {"ZbState":{"Status":33,"Endpoint":"0x01","ProfileId":"0x0104","DeviceId":"0x0006","DeviceVersion":2"InClusters":["0x0000","0x0003","0x0001","0x0002","0x0019","0x000A"],"OutClusters":["0x0000","0x000A","0x0019"]}}
09:52:41 ZIG: ZbZNPSent 2401D6460101000001301E0700010004000500
09:52:41 ZIG: {"ZbZNPReceived":"640100"}
09:52:42 ZIG: {"ZbZNPReceived":"4480000101"}
09:52:42 ZIG: {"ZbZNPReceived":"448100000000D6460101005800D5138000002318010104000042044C554D4905000042126C756D692E6374726C5F6E65757472616C31D6461D"}
09:52:42 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x46D6","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":88,"securityuse":0,"seqnumber":0,"timestamp":8393685,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"04000042044C554D4905000042126C756D692E6374726C5F6E65757472616C31"}}
09:52:42 ZIG: ZbZCLRawReceived: {"0x46D6":{"0000/0004":"LUMI","0000/0005":"lumi.ctrl_neutral1"}}
09:52:42 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x46D6":{"Manufacturer":"LUMI","ModelId":"lumi.ctrl_neutral1","LinkQuality":88}}}
09:52:42 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x46D6":{"Manufacturer":"LUMI","ModelId":"lumi.ctrl_neutral1","LinkQuality":88}}}
```

s-hadinger, 2020-02-20:
Thanks. I don't have a clue why you can't turn on/Off remotely. Everything looks good. 

I just realized there is a problem with ZbProbe. There are 6 requests sent but they seem to get lost. Maybe I should add delays between each request

russellhq, 2020-02-20:
I reset the CC2530 and re-paired it with zb2t. I then sent the power on command and the light switch turned on. But every command I tried after that didn't work. Here's the weblog 3 output:

```

17:50:12 CMD: Zigbeesend { "device":"0x4D9A", endpoint: 2, "send": {"Power": true}}
17:50:12 SRC: WebConsole from 192.168.1.229
17:50:12 CMD: Group 0, Index 1, Command "ZIGBEESEND", Data "{ "device":"0x4D9A", endpoint: 2, "send": {"Power": true}}"
17:50:12 ZbSend: command_template = 0006!xx
17:50:12 SendZCLCommand_P: zcl_cmd = 0006!01
17:50:12 ZbSend: command_final    = 0006!01
17:50:12 ZbCmd_actual: ZigbeeZCLSend {"device":"0x4D9A","endpoint":2,"send":"0006!01"}
17:50:12 ZbSend: dstAddr 0x4D9A, cluster 0x0006, endpoint 0x02, cmd 0x01, data 
17:50:12 ZIG: ZbZNPSent 24019A4D0201060001301E03110101
17:50:12 MQT: stat/tasmota_14273C/RESULT = {"ZigbeeSend":"Done"}
17:50:12 ZIG: {"ZbZNPReceived":"640100"}
17:50:12 ZIG: ZbZNPSent 24019A4D0201060001301E050001000000
17:50:12 ZIG: {"ZbZNPReceived":"640100"}
17:50:12 ZIG: {"ZbZNPReceived":"4480000101"}
17:50:12 ZIG: {"ZbZNPReceived":"4481000006009A4D0201005B00E97B0700000E18030A0000100100F023010000079A4D1D"}
17:50:12 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x4D9A","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":91,"securityuse":0,"seqnumber":0,"timestamp":490473,"fc":"0x18","manuf":"0x0000","transact":3,"cmdid":"0x0A","payload":"0000100100F02301000007"}}
17:50:12 ZIG: ZbZCLRawReceived: {"0x4D9A":{"0006/0000":true,"0006/F000":117440513}}
17:50:13 ZIG: {"ZbZNPReceived":"45C1B06EB06E8AAFA800AA3EB07C8E"}
17:50:13 MQT: tele/tasmota_14273C/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"7CB03EAA00A8AF8A","ShortAddr":"0x6EB0","PowerSource":true,"ReceiveWhenIdle":true,"Security":false}}
17:50:13 ZIG: ZbZNPSent 2505B06EB06E
17:50:13 ZIG: {"ZbZNPReceived":"650500"}
17:50:13 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x4D9A":{"0006/F000":117440513,"Power":true,"LinkQuality":91}}}
17:50:13 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x4D9A":{"0006/F000":117440513,"Power":true,"LinkQuality":91}}}
17:50:13 ZIG: {"ZbZNPReceived":"4585B06E00B06E0103"}
17:50:13 ZIG: ZbZNPSent 2504B06EB06E03
17:50:13 MQT: tele/tasmota_14273C/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x03"]}}
17:50:13 ZIG: {"ZbZNPReceived":"650400"}
17:50:13 ZIG: {"ZbZNPReceived":"4480000101"}
17:50:13 ZIG: {"ZbZNPReceived":"4481000006009A4D0201005B00F3860700000818010100000010019A4D1D"}
17:50:13 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x4D9A","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":91,"securityuse":0,"seqnumber":0,"timestamp":493299,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"0000001001"}}
17:50:13 ZIG: ZbZCLRawReceived: {"0x4D9A":{"0006/0000":true}}
17:50:13 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x4D9A":{"Power":true,"LinkQuality":91}}}
17:50:13 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x4D9A":{"Power":true,"LinkQuality":91}}}
17:50:14 ZIG: {"ZbZNPReceived":"4584B06E00B06E1C035EC010020209001000000300040005000600080000030FFC011900"}
17:50:14 MQT: tele/tasmota_14273C/RESULT = {"ZbState":{"Status":33,"Endpoint":"0x03","ProfileId":"0xC05E","DeviceId":"0x0210","DeviceVersion":2"InClusters":["0x1000","0x0000","0x0003","0x0004","0x0005","0x0006","0x0008","0x0300","0xFC0F"],"OutClusters":["0x0019"]}}
17:50:14 ZIG: ZbZNPSent 2401B06E0301000001301E0700010004000500
17:50:14 ZIG: {"ZbZNPReceived":"640100"}
17:50:14 ZIG: {"ZbZNPReceived":"4480000101"}
17:50:14 ZIG: {"ZbZNPReceived":"448100000000B06E0301003100BB8D0700002218010104000042054F5352414D0500004210434C4136302052474257204F5352414DB06E1D"}
17:50:14 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":0,"srcaddr":"0x6EB0","srcendpoint":3,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":49,"securityuse":0,"seqnumber":0,"timestamp":495035,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"04000042054F5352414D0500004210434C4136302052474257204F5352414D"}}
17:50:14 ZIG: ZbZCLRawReceived: {"0x6EB0":{"0000/0004":"OSRAM","0000/0005":"CLA60 RGBW OSRAM"}}
17:50:14 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x6EB0":{"Manufacturer":"OSRAM","ModelId":"CLA60 RGBW OSRAM","LinkQuality":49}}}
17:50:14 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x6EB0":{"Manufacturer":"OSRAM","ModelId":"CLA60 RGBW OSRAM","LinkQuality":49}}}

17:51:39 CMD: Zigbeesend { "device":"0x4D9A", endpoint: 2, "send": {"Power": false}}
17:51:39 SRC: WebConsole from 192.168.1.229
17:51:39 CMD: Group 0, Index 1, Command "ZIGBEESEND", Data "{ "device":"0x4D9A", endpoint: 2, "send": {"Power": false}}"
17:51:39 ZbSend: command_template = 0006!xx
17:51:39 SendZCLCommand_P: zcl_cmd = 0006!00
17:51:39 ZbSend: command_final    = 0006!00
17:51:39 ZbCmd_actual: ZigbeeZCLSend {"device":"0x4D9A","endpoint":2,"send":"0006!00"}
17:51:39 ZbSend: dstAddr 0x4D9A, cluster 0x0006, endpoint 0x02, cmd 0x00, data 
17:51:39 ZIG: ZbZNPSent 24019A4D0201060001301E03110100
17:51:39 MQT: stat/tasmota_14273C/RESULT = {"ZigbeeSend":"Done"}
17:51:39 ZIG: {"ZbZNPReceived":"640100"}
17:51:39 ZIG: {"ZbZNPReceived":"4480000101"}
17:51:39 ZIG: ZbZNPSent 24019A4D0201060001301E050001000000
17:51:39 ZIG: {"ZbZNPReceived":"640100"}
17:51:39 ZIG: {"ZbZNPReceived":"4481000006009A4D0201005D001A9E0B00000E18060A0000100000F0230100000B9A4D1D"}
[... cut ...]
17:51:40 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x4D9A","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":93,"securityuse":0,"seqnumber":0,"timestamp":764287,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"0000001000"}}
17:51:40 ZIG: ZbZCLRawReceived: {"0x4D9A":{"0006/0000":false}}
17:51:40 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x4D9A":{"Power":false,"LinkQuality":93}}}
17:51:40 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x4D9A":{"Power":false,"LinkQuality":93}}}
```

s-hadinger, 2020-02-20:
Interesting. Can you try a toggle command?

`Zigbeesend { "device":"0x4D9A", endpoint: 2, "send": {"Power":2}}`

s-hadinger, 2020-02-20:
Can you also try the following command, it should shut down the switch:

`ZbZNPSend 24019A4D0201060009301E03110900`

I changed the transaction ID. Just to be sure that's not the issue, but it could explain why the first command worked and not the others.

russellhq, 2020-02-20:
That's done something! I modified your command because I had reset the module again and the switch received a new address. Anyway, your command switches off the switch! And I can also switch it back on with the normal power true command. Log below. is there a way I can repeat this now via MQTT?

```
20:10:23 CMD: ZbZNPSend 2401D20D0201060009301E03110900
20:10:23 SRC: WebConsole from 192.168.1.229
20:10:23 CMD: Group 0, Index 1, Command "ZBZNPSEND", Data "2401D20D0201060009301E03110900"
20:10:23 ZIG: ZbZNPSent 2401D20D0201060009301E03110900
20:10:23 MQT: stat/tasmota_14273C/RESULT = {"ZbZNPSend":"Done"}
20:10:23 ZIG: {"ZbZNPReceived":"640100"}
20:10:23 ZIG: {"ZbZNPReceived":"4480000109"}
20:10:23 ZIG: {"ZbZNPReceived":"448100000600D20D0201005D001218B200000E18B60A0000100000F02309000007D20D1D"}
20:10:23 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0DD2","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":93,"securityuse":0,"seqnumber":0,"timestamp":11671570,"fc":"0x18","manuf":"0x0000","transact":182,"cmdid":"0x0A","payload":"0000100000F02309000007"}}
20:10:23 ZIG: ZbZCLRawReceived: {"0x0DD2":{"0006/0000":false,"0006/F000":117440521}}
20:10:24 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x0DD2":{"0006/F000":117440521,"Power":false,"LinkQuality":93}}}
20:10:24 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x0DD2":{"0006/F000":117440521,"Power":false,"LinkQuality":93}}}
20:10:36 CMD: Zigbeesend { "device":"0x0DD2", endpoint: 2, "send": {"Power": true}}
20:10:36 SRC: WebConsole from 192.168.1.229
20:10:36 CMD: Group 0, Index 1, Command "ZIGBEESEND", Data "{ "device":"0x0DD2", endpoint: 2, "send": {"Power": true}}"
20:10:36 ZbSend: command_template = 0006!xx
20:10:36 SendZCLCommand_P: zcl_cmd = 0006!01
20:10:36 ZbSend: command_final    = 0006!01
20:10:36 ZbCmd_actual: ZigbeeZCLSend {"device":"0x0DD2","endpoint":2,"send":"0006!01"}
20:10:36 ZbSend: dstAddr 0x0DD2, cluster 0x0006, endpoint 0x02, cmd 0x01, data 
20:10:36 ZIG: ZbZNPSent 2401D20D0201060001301E03110101
20:10:36 MQT: stat/tasmota_14273C/RESULT = {"ZigbeeSend":"Done"}
20:10:37 ZIG: {"ZbZNPReceived":"448100000600D20D0201005D00DBBBB200000E18B70A0000100100F02301000007D20D1D"}
20:10:37 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0DD2","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":93,"securityuse":0,"seqnumber":0,"timestamp":11713499,"fc":"0x18","manuf":"0x0000","transact":183,"cmdid":"0x0A","payload":"0000100100F02301000007"}}
20:10:37 ZIG: ZbZCLRawReceived: {"0x0DD2":{"0006/0000":true,"0006/F000":117440513}}
20:10:37 ZIG: {"ZbZNPReceived":"45C1B06EB06E8AAFA800AA3EB07C8E"}
20:10:37 MQT: tele/tasmota_14273C/RESULT = {"ZbState":{"Status":30,"IEEEAddr":"7CB03EAA00A8AF8A","ShortAddr":"0x6EB0","PowerSource":true,"ReceiveWhenIdle":true,"Security":false}}
20:10:37 ZIG: ZbZNPSent 2505B06EB06E
20:10:37 ZIG: {"ZbZNPReceived":"650500"}
20:10:37 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x0DD2":{"0006/F000":117440513,"Power":true,"LinkQuality":93}}}
20:10:37 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x0DD2":{"0006/F000":117440513,"Power":true,"LinkQuality":93}}}
20:10:37 ZIG: {"ZbZNPReceived":"4585B06E00B06E0103"}
20:10:37 ZIG: ZbZNPSent 2504B06EB06E03
20:10:37 MQT: tele/tasmota_14273C/RESULT = {"ZbState":{"Status":32,"ActiveEndpoints":["0x03"]}}
20:10:37 ZIG: {"ZbZNPReceived":"650400"}
20:10:37 ZIG: {"ZbZNPReceived":"4584B06E00B06E1C035EC010020209001000000300040005000600080000030FFC011900"}
20:10:37 MQT: tele/tasmota_14273C/RESULT = {"ZbState":{"Status":33,"Endpoint":"0x03","ProfileId":"0xC05E","DeviceId":"0x0210","DeviceVersion":2"InClusters":["0x1000","0x0000","0x0003","0x0004","0x0005","0x0006","0x0008","0x0300","0xFC0F"],"OutClusters":["0x0019"]}}
20:10:37 ZIG: ZbZNPSent 2401B06E0301000001301E0700010004000500
20:10:37 ZIG: {"ZbZNPReceived":"448100000A00D20D0101005D00CCC6B200000510B8000000D20D1D"}
20:10:37 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":10,"srcaddr":"0x0DD2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":93,"securityuse":0,"seqnumber":0,"timestamp":11716300,"fc":"0x10","manuf":"0x0000","transact":184,"cmdid":"0x00","payload":"0000"}}
20:10:37 ZIG: ZbZCLRawReceived: {"0x0DD2":{}}
20:10:37 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x0DD2":{"LinkQuality":93}}}
20:10:37 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x0DD2":{"LinkQuality":93}}}
20:10:37 ZIG: {"ZbZNPReceived":"4480000101"}
20:10:38 ZIG: {"ZbZNPReceived":"448100000600D20D0201005D00FDC6B20000081801010000001001D20D1D"}
20:10:38 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0DD2","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":93,"securityuse":0,"seqnumber":0,"timestamp":11716349,"fc":"0x18","manuf":"0x0000","transact":1,"cmdid":"0x01","payload":"0000001001"}}
20:10:38 ZIG: ZbZCLRawReceived: {"0x0DD2":{"0006/0000":true}}
20:10:38 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x0DD2":{"Power":true,"LinkQuality":93}}}
20:10:38 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x0DD2":{"Power":true,"LinkQuality":93}}}
20:10:38 ZIG: {"ZbZNPReceived":"640100"}
20:10:38 ZIG: {"ZbZNPReceived":"4480000101"}
20:10:38 ZIG: {"ZbZNPReceived":"448100000000B06E030100290086CBB200002218010104000042054F5352414D0500004210434C4136302052474257204F5352414DB06E1D"}
[... cut ...]
20:11:39 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":10,"srcaddr":"0x0DD2","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":93,"securityuse":0,"seqnumber":0,"timestamp":11910057,"fc":"0x10","manuf":"0x0000","transact":186,"cmdid":"0x00","payload":"0000"}}
20:11:39 ZIG: ZbZCLRawReceived: {"0x0DD2":{}}
20:11:39 MQT: tele/tasmota_14273C/SENSOR = {"ZbReceived":{"0x0DD2":{"LinkQuality":93}}}
20:11:39 MQT: tele/tasmota_14273C/SENSOR = {"ZigbeeReceived":{"0x0DD2":{"LinkQuality":93}}}
```

s-hadinger, 2020-02-20:
That's good news. It means the device does pay attention to the transaction ID. Unfortunately you cannot set it by Mqtt.  

I need to add a counter to increment the ids with each request. It will require a code change, hopefully next weekend.

russellhq, 2020-02-20:
Thanks, sounds good. I gave the following a try and it worked for turning the light off from MQTT

`
cmnd/tasmota_14273C/ZbZNPSend 2401D20D0201060009301E03110900`

I'll use that for now until the the code is changed

s-hadinger, 2020-02-20:
Yes indeed. And changing the last byte to 01 will turn it on. Actually there are two transaction ids, they both have values 09 here. Transaction ids are single byte and supposed to increment with each request. 

So it's possible that future calls need to increase it. I'm also interested in knowing which one makes it working. If you have time to explore it.

russellhq, 2020-02-20:
Sure, just let me know what you want me to try and I'll let you know how I get on.

s-hadinger, 2020-02-20:
Just change the first 09 to 01 and try if switching off works. Then try replacing only the second 09 with 01.

russellhq, 2020-02-20:
This turned off the light but I couldn't turn it back on with the power command:
ZbZNPSend 2401D20D0201060009301E03110100

This didn't work:
ZbZNPSend 2401D20D0201060001301E03110900

This didn't either:
ZbZNPSend 2401D20D0201060001301E03110100

s-hadinger, 2020-02-20:
Thanks a lot. That's exactly what I needed.

russellhq, 2020-02-26:
Thanks all, tried the new version and the switch is now responding 👍

PR #7773 (2020-02-23): Fix Zigbee auto-increment transaction number (#7757)
## Description:

Auto-increment the sequence and transaction number for each outgoing message. Sequence numbers are per-device. It seems that Aqara switches do care about seq number.

Also add `Device` attribute to `ZbReceived` message. Some multi-button switches use source endpoint to indicate which button was pressed.

Exemple:

`
MQT: tele/<topic>/SENSOR = {"ZigbeeReceived":{"Kitchen_presence":{"Device":"0xE07E", "Illuminance":37, "Endpoint":1, "LinkQuality":59}}}
`

**Related issue (if applicable):** fixes #7757 

## Checklist:
  - [x] T [... cut]