arendst/Tasmota #7758: Get all the options

timmaeh, 2020-02-19:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
Is there an option to get all my setting like "setoption 80 1" down to a document? Months after programming i forget what my settings are.

**Describe the solution you'd like**  
An option do get a list of all my settings ive done.

**Describe alternatives you've considered**  

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-02-19:
Use the ``decode-status.py`` tool like
```
.\decode-status.py -d sonoff3

*** decode-status.py v20200210 by Theo Arends and Jacek Ziolkowski ***
Decoding information for device Sonoff3 from status report taken at 2020-02-19T09:59:33

Options
   0 (ON ) Save power state and use after restart
   1 (OFF) Restrict button actions to single, double and hold
   2 (OFF) (not used) Show value units in JSON messages
   3 (ON ) MQTT enabled
   4 (OFF) Respond as Command topic instead of RESULT
   5 (OFF) MQTT retain on Power
   6 (OFF) MQTT retain on Button
   7 (OFF) MQTT retain on Switch
   8 (OFF) Convert temperature to Fahrenheit
   9 (OFF) MQTT retain on Sensor
  10 (OFF) MQTT retained LWT to OFFLINE when topic changes
  11 (OFF) Swap Single and Double press Button
  12 (OFF) Do not use flash page rotate
  13 (OFF) Button single press only
  14 (OFF) Power interlock mode
  15 (ON ) Do not allow PWM control
  16 (OFF) Reverse clock
  17 (OFF) Allow entry of decimal color values
  18 (OFF) CO2 color to light signal
  19 (OFF) HASS discovery
  20 (OFF) Do not control Power with Dimmer
  21 (OFF) Energy monitoring while powered off
  22 (OFF) MQTT serial
  23 (OFF) MQTT serial binary
  24 (OFF) Convert pressure to mmHg
  25 (OFF) KNX enabled
  26 (OFF) Use Power device index on single relay devices
  27 (OFF) KNX enhancement
  28 (OFF) RF receive decimal
  29 (OFF) IR receive decimal
  30 (OFF) Enforce HASS light group
  31 (OFF) Do not show Wifi and Mqtt state using Led
  32 ( 40) Key hold time (ms)
  33 (  5) Sonoff POW Max_Power_Retry
  34 (200) Backlog delay (ms)
  35 (  0) (not used) mDNS delayed start (Sec)
  36 (  1) Boot loop retry offset (0 = disable)
  37 (  0) RGBWW remap
  38 (  6) IR Unknown threshold
  39 (  0) CSE7766 invalid power margin
  40 (  0) Ignore hold time (s)
  41 (  0) (not used) Number of Tuya MCU relays
  42 ( 90) Over temperature threshold (celsius)
  43 (  0) (not used) Tuya MCU max dimmer value
  44 (  0) (not used) Tuya MCU voltage Id
  45 (  0) (not used) Tuya MCU current Id
  46 (  0) (not used) Tuya MCU power Id
  47 ( 23) (not used) Energy Tariff1 start hour
  48 (  7) (not used) Energy Tariff2 start hour
  49 (  0)
  50 (OFF) Timers enabled
  51 (OFF) Generic ESP8285 GPIO enabled
  52 (OFF) Add UTC time offset to JSON message
  53 (OFF) Show hostname and IP address in GUI
  54 (OFF) Apply SetOption20 to Tuya
  55 (OFF) mDNS enabled
  56 (ON ) Use wifi network scan at restart
  57 (ON ) Use wifi network rescan regularly
  58 (OFF) Add IR raw data to JSON message
  59 (OFF) Change state topic from tele/STATE to stat/RESULT
  60 (OFF) Enable normal sleep instead of dynamic sleep
  61 (OFF) Force local operation when button/switch topic is set
  62 (OFF) Do not use retain flag on HOLD messages
  63 (OFF) Do not scan relay power state at restart
  64 (OFF) Use _ instead of - as sensor index separator
  65 (OFF) Disable fast power cycle detection for device reset
  66 (OFF) Enable TuyaMcuReceived messages over Mqtt
  67 (OFF) Enable buzzer when available
  68 (OFF) Enable multi-channels PWM instead of Color PWM
  69 (OFF) (not used) Limits Tuya MCU dimmers to minimum of 10% (25) when enabled
  70 (OFF) Enable Weekend Energy Tariff
  71 (OFF) Select different Modbus registers for Active Energy
  72 (OFF) Enable hardware energy total counter as reference
  73 (ON ) Enable HTTP CORS
  74 (OFF) Enable internal pullup for single DS18x20 sensor
  75 (OFF) GroupTopic replaces %topic% (0) or fixed topic cmnd/grouptopic (1)
  76 (OFF) Enable incrementing bootcount when deepsleep is enabled
  77 (OFF) Do not power off if slider moved to far left
  78 (OFF) Bypass Compatibility check
  79 (OFF) Enable resetting of counters after telemetry was sent
  80 (OFF) Enable shutter support
  81 (OFF) Invert PCF8574 ports
  82 (OFF) Reduced CT range for Alexa
  83 (OFF) Use FriendlyNames instead of ShortAddresses when possible
  84 (OFF) (AWS IoT) publish MQTT state to a device shadow
  85 (OFF)
  86 (OFF)
  87 (OFF)
  88 (OFF)
  89 (OFF)
  90 (OFF)
  91 (OFF)
  92 (OFF)
  93 (OFF)
  94 (OFF)
  95 (OFF)
  96 (OFF)
  97 (OFF)
  98 (OFF)
  99 (OFF)
  100 (OFF)
  101 (OFF)
  102 (OFF)
  103 (OFF) 
  104 (OFF)
  105 (OFF)
  106 (OFF)
  107 (OFF)
  108 (OFF)
  109 (OFF)
  110 (OFF)
  111 (OFF)
  112 (OFF)
  113 (OFF)

Features
  Language LCID = 2057
  MQTT_HOST_DISCOVERY
  USE_ADE7953
  USE_BH1750
  USE_BME680
  USE_BMP
  USE_BUZZER
  USE_CONFIG_OVERRIDE
  USE_CSE7766
  USE_DHT
  USE_DISPLAY_LCD
  USE_DISPLAY_MATRIX
  USE_DISPLAY_MODES1TO5
  USE_DISPLAY_SSD1306
  USE_DOMOTICZ
  USE_DS18x20
  USE_ENERGY_SENSOR
  USE_HLW8012
  USE_HM10
  USE_HTU
  USE_HX711
  USE_I2C
  USE_IR_RECEIVE
  USE_IR_REMOTE
  USE_LIGHT
  USE_MCP39F501
  USE_PZEM004T
  USE_PZEM_AC
  USE_RULES
  USE_SDM120
  USE_SM16716
  USE_SM2135
  USE_SONOFF_IFAN
  USE_SONOFF_L1
  USE_SONOFF_RF
  USE_SUNRISE
  USE_TIMERS
  USE_TIMERS_WEB
  USE_WEBSERVER
  USE_WS2812
  VTABLES_IN_FLASH
  WEBSERVER_ADVERTISE
```

timmaeh, 2020-02-19:
So i just write this in the console of my device?

arendst, 2020-02-19:
Nope.

It's a python script to be run on a windows or linux desktop. If you have platformio installed just execute it from the terminal page.

timmaeh, 2020-02-19:
Ah ok. Ive found an executable for win. I will give it a try [https://github.com/tasmota/Tasmota-decode-status](https://github.com/tasmota/Tasmota-decode-status)

arendst, 2020-02-19:
Yes that should work too.

timmaeh, 2020-02-19:
ty

closed 2020-02-19 by timmaeh