arendst/Tasmota #7760: TuyaMCU Button press not publishing to MQTT

wb-ha, 2020-02-19:
**Have you looked for this feature in other issues and in the docs?**  
Yes and forums

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
Yes. I am using the TuyaMCU module for 1-4 gang light switches and am unable to define rules or publish MQTT messages on button (or switch) press.
eg. ON button1#state DO POWER1 TOGGLE ENDON

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I'd like to be able to trigger rules and/or have the devices publish MQTT messages when the button is pressed as per normal Tasmota devices.
A MQTT publish like the below when the relay is activated by the local button
root/mobile/devices/b_units/b_S1/button1 on
root/mobile/devices/b_units/b_S1/button1 off
(where on or off is the state of the associated relay)

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
I do get power and result MQTT message
07:48:02 MQT: root/mobile/devices/b_units/b_G08/RESULT = {“POWER1”:“ON”}
07:48:02 MQT: root/mobile/devices/b_units/b_G08/POWER1 = ON (retained)
so have tried to code automations in HomeAssistant and NodeRED to get around this but I am trying to configure the switches to be interlinked (2 way and 3 way)and I keep getting into an infinite loop as I can't tell if the power state change was caused by a button press, web activation or MQTT command.

**Additional context**  
_Add any other context or screenshots about the feature request here._
The TuyaMCU module is configured and works fine standalone and via MQTT.
I have set a switchtopic and button topic (since I wasn't sure which might work)-
07:47:56 CMD: switchtopic
07:47:56 MQT: root/mobile/devices/b_units/b_G08/RESULT = {“SwitchTopic”:“b_G08”}
07:48:00 CMD: buttontopic
07:48:00 MQT: root/mobile/devices/b_units/b_G08/RESULT = {“ButtonTopic”:“b_G08”}

If I set verbose logging I can see in the logs (note the SRC line)
 
A websend from another unit looks like
21:03:54 HTP: Command
**21:03:54 SRC: WebCommand from 10.1.5.167**
21:03:54 CMD: Group 0, Index 1, Command “POWER”, Data “0”
21:03:54 TYA: Send “55aa0006000501010001000d”
21:03:54 MQT: root/mobile/devices/b_units/b_S1/RESULT = {“POWER1”:“OFF”}
21:03:54 MQT: root/mobile/devices/b_units/b_S1/POWER1 = OFF (retained)
21:03:54 MQT: root/mobile/devices/b_units/b_S1/RESULT = {“TuyaReceived”:{“Data”:“55AA0107000501010001000F”,“Cmnd”:7,“CmndData”:“0101000100”,“DpId”:1,“DpIdType”:1,“DpIdData”:“00”}}

A local switching (pressing the button on the unit looks like) looks like
21:03:54 TYA: RX Relay-1 --> MCU State: Off Current State:On
**21:03:54 SRC: Switch**
21:03:54 MQT: root/mobile/devices/b_units/b_S1/RESULT = {“POWER1”:“OFF”}
21:03:54 MQT: root/mobile/devices/b_units/b_S1/POWER1 = OFF (retained)
21:03:54 RUL: POWER1#STATE performs “WebSend [10.1.5.167,usr:pwd] POWER1 0”

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-02-19:
There is no button for Tuya modules only switches.
See https://tasmota.github.io/docs/#/TuyaMCU

wb-ha, 2020-02-19:
Thanks Jason - I did try both Buttons and Switches, but no matter what I did I couldn't get a MQTT topic to publish (thats not to say I still haven't missed something).  
The log still displays a source even without the switch linked to a DPId so I presume it's possible for Tasmota to identify the source and publish?
The switches have dpid 7,8,9,10 and 101 that I haven't been able to identify a function for yet.  I have tried dpid7=1 without any luck.
I can edit my above to remove references to buttons and make it switches.

ascillato2, 2020-02-20:
Please try rules as in https://github.com/arendst/Tasmota/issues/7646

Thanks.

ascillato2, 2020-02-23:
Please, when you have time to test the provided rules, tell us if this solves your request. Thanks.

closed 2020-02-23 by ascillato2

wb-ha, 2020-02-24:
Hi 

Have tried these rules as referenced in #7646 but doesn't do anything - I'm sorry but I'm not understanding how those rules would solve the issue as I think they should report the state of the relays but not the source of the activation.  I already get the state of the relay (through Power and Result MQTT messages) but I need to know the source of the activation and publish that to MQTT (or allow me to write a rule) 

Here is the rule I tried as requested
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":269,"Rules":"on TuyaReceived#Cmnd7DpId3DpType4 do Publish stat/%topic%/SPEED %value% endon   on TuyaReceived#Cmnd7DpId2DpType4 do Publish stat/%topic%/FANMODE %value% endon   on TuyaReceived#Cmnd7DpId6DpType4 do Publish stat/%topic%/FANTIMER %value% endon"}

I have done very little with rules but for my config I think it should be
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":434,"Rules":"on TuyaReceived#Cmnd7DpId1DpType1 do Publish stat/%topic%/SPEED %value% endon"}
but couldn't get that to work either.

I have gathered the logs for 3 different activation types - as you can see the TuyaReceived is the same for all ON events and all OFF events regardless of source.  The only difference I can see is that remote sources have a TuyaSend and the SRC log entry is different (is there any way to write a rule to look for a log string??)

### TurnOFF   via Switch
22:04:23   {"TuyaReceived":{"Data":"55AA0107000501010001000F","Cmnd":7,"CmndData":"0101000100","DpId":1,"DpIdType":1,"DpIdData":"00"}}
22:04:23 TYA: fnId=11 is set for dpId=1
22:04:23 TYA: RX Relay-1 --> MCU   State: Off Current State:On
22:04:23 SRC: Switch
22:04:23 MQT:   root/mobile/devices/b_units/b_S1/RESULT =   {"POWER1":"OFF"}
22:04:23 MQT:   root/mobile/devices/b_units/b_S1/POWER1 = OFF (retained)
 
 
### Turn ON Via Switch
22:06:00   {"TuyaReceived":{"Data":"55AA01070005010100010110","Cmnd":7,"CmndData":"0101000101","DpId":1,"DpIdType":1,"DpIdData":"01"}}
22:06:00 TYA: fnId=11 is set for dpId=1
22:06:00 TYA: RX Relay-1 --> MCU   State: On Current State:Off
22:06:00 SRC: Switch
22:06:00 MQT:   root/mobile/devices/b_units/b_S1/RESULT = {"POWER1":"ON"}
22:06:00 MQT:   root/mobile/devices/b_units/b_S1/POWER1 = ON (retained)
 
 
### Turn ON via Console
22:11:17 HTP: Main Menu
22:11:19 SRC: WebGui from 10.1.1.75
22:11:19 TYA: Send   "55aa0006000501010001010e"
22:11:19 MQT:   root/mobile/devices/b_units/b_S1/RESULT = {"POWER1":"ON"}
22:11:19 MQT:   root/mobile/devices/b_units/b_S1/POWER1 = ON (retained)
22:11:19   {"TuyaReceived":{"Data":"55AA01070005010100010110","Cmnd":7,"CmndData":"0101000101","DpId":1,"DpIdType":1,"DpIdData":"01"}}
22:11:19 TYA: fnId=11 is set for dpId=1
22:11:19 TYA: RX Relay-1 --> MCU   State: On Current State:On
22:11:20 CFG: Saved to flash at FB, Count   41, Bytes 4096
22:11:20 HTP: Console
 
 
### Turn OFF via Console
22:15:16 HTP: Main Menu
22:15:18 SRC: WebGui from 10.1.1.75
22:15:18 TYA: Send   "55aa0006000501010001000d"
22:15:18 MQT:   root/mobile/devices/b_units/b_S1/RESULT =   {"POWER1":"OFF"}
22:15:18 MQT:   root/mobile/devices/b_units/b_S1/POWER1 = OFF (retained)
22:15:18   {"TuyaReceived":{"Data":"55AA0107000501010001000F","Cmnd":7,"CmndData":"0101000100","DpId":1,"DpIdType":1,"DpIdData":"00"}}
22:15:18 TYA: fnId=11 is set for dpId=1
22:15:18 TYA: RX Relay-1 --> MCU   State: Off Current State:Off
22:15:18 CFG: Saved to flash at FA, Count   42, Bytes 4096
22:15:19 HTP: Console
 
 
### Turn ON via MQTT
22:20:42 MQT: Received Topic   "root/mobile/devices/b_units/b_S1/cmnd/power1", Data Size 2, Data   "on"
22:20:42 SRC: MQTT
22:20:42 CMD: Group 0, Index 1, Command   "POWER", Data "on"
22:20:42 TYA: Send   "55aa0006000501010001010e"
22:20:42 MQT:   root/mobile/devices/b_units/b_S1/RESULT = {"POWER1":"ON"}
22:20:42 MQT:   root/mobile/devices/b_units/b_S1/POWER1 = ON (retained)
22:20:42   {"TuyaReceived":{"Data":"55AA01070005010100010110","Cmnd":7,"CmndData":"0101000101","DpId":1,"DpIdType":1,"DpIdData":"01"}}
22:20:42 TYA: fnId=11 is set for dpId=1
22:20:42 TYA: RX Relay-1 --> MCU   State: On Current State:On
22:20:42 CFG: Saved to flash at F7, Count   45, Bytes 4096
 
 
### Turn OFF via MQTT
22:23:08 MQT: Received Topic   "root/mobile/devices/b_units/b_S1/cmnd/power1", Data Size 3, Data   "off"
22:23:08 SRC: MQTT
22:23:08 CMD: Group 0, Index 1, Command   "POWER", Data "off"
22:23:08 TYA: Send   "55aa0006000501010001000d"
22:23:08 MQT:   root/mobile/devices/b_units/b_S1/RESULT =   {"POWER1":"OFF"}
22:23:08 MQT:   root/mobile/devices/b_units/b_S1/POWER1 = OFF (retained)
22:23:08   {"TuyaReceived":{"Data":"55AA0107000501010001000F","Cmnd":7,"CmndData":"0101000100","DpId":1,"DpIdType":1,"DpIdData":"00"}}
22:23:08 TYA: fnId=11 is set for dpId=1
22:23:08 TYA: RX Relay-1 --> MCU   State: Off Current State:Off
22:23:09 CFG: Saved to flash at F6, Count   46, Bytes 4096

wb-ha, 2020-02-25:
Hi @ascillato2 - do you have any further thoughts on this issue pls?

ascillato, 2020-02-25:
Hi, have you updated to latest Tasmota? That has been added very recently.

wb-ha, 2020-02-25:
Hi - I was on 8.1.0.2 and just updated my test unit to 8.1.0 but the issue appears to be the same - will go through my config again tonight to make sure I haven't changed anything with all the testing.  Am I right in assuming I should get a MQTT switch message by just defining a switch topic?  Do I need to do anything else? (MQTT is configured and working for power, state and sensor topics)

meingraham, 2020-02-25:
You need to update to a later version than 8.1.0.2. 8.1.0 is actually an earlier version. Use 

```Backlog OTAURL http://thehackbox.org/tasmota/tasmota/bin; Upgrade 1```

to load the latest version. Allow at least 5 minutes for the update to complete as the process requires two firmware uploads and associated restarts.

ascillato, 2020-02-26:
Latest Tasmota is 8.1.0.9

Please update (http://thehackbox.org/tasmota/tasmota.bin)

wb-ha, 2020-02-26:
Hi, thanks for your help on this

I have done the upgrade as recommended and tried again with no change - now on version 8.1.0.9.

To be clear 
- There is no DpId configured for a switch (only the relays) -
- I have added a switchtopic (and buttontopic just in case)
- MQTT is working correctly
I am not sure if I need to do anything else?

I still do not get a MQTT message when changing state with the switch even though it shows as the SRC in the weblog (as per attached)

As you can see from the below - the Tuyareceived includes information on the state of the RELAY but I am trying to get the SRC: which caused the relay to change state - which only appears in weblog with the SRC: header.  Is there a way to get the information from the SRC: log line into a MQTT publish ?

21:07:31   {"TuyaReceived":{"Data":"55AA0107000501010001000F","Cmnd":7,"CmndData":"0101000100","DpType1Id1":0,"1":{"DpId":1,"DpIdType":1,"DpIdData":"00"}}}
21:07:31 TYA: fnId=11 is set for dpId=1
21:07:31 TYA: RX Relay-1 --> MCU   State: Off Current State:On
**21:07:31 SRC: Switch**
21:07:31 MQT:   root/mobile/devices/b_units/b_S1/RESULT =   {"POWER1":"OFF"}
21:07:31 MQT:   root/mobile/devices/b_units/b_S1/POWER1 = OFF (retained)
21:07:35   {"TuyaReceived":{"Data":"55AA01070005010100010110","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
21:07:35 TYA: fnId=11 is set for dpId=1
21:07:35 TYA: RX Relay-1 --> MCU   State: On Current State:Off
**21:07:35 SRC: Switch**
21:07:35 MQT:   root/mobile/devices/b_units/b_S1/RESULT = {"POWER1":"ON"}
21:07:35 MQT:   root/mobile/devices/b_units/b_S1/POWER1 = ON (retained)

[image]

wb-ha, 2020-03-07:
Hi all , thanks for suggestions so far but I'm not sure where to go with this as the issue is now listed as closed.  Is anything else I can do here or do I continue with workarounds.