arendst/Tasmota #7763: WiFi light settings: always on when connected to MQTT

glennmurley15, 2020-02-20:
**Have you looked for this feature in other issues and in the docs?**  yes

**Is your feature request related to a problem? Please describe.**  no
_A clear and concise description of what the problem is._ 

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

I am requesting an extra setting for the wifi connection light (I use T1 switches) - I would like the wifi connection light to have a setting to always be lit up when connected to MQTT broker. Current default setting the wifi light is on when connected to wifi, and it turns off when a switch LED comes on. I would like to be able to have the wifi light always turned on when the device is connected to MQTT broker, and only turn off when connection breaks.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

Checked settings for wifi light

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-20:
Already implemented. See commands in the docs. Thanks

ascillato2, 2020-02-20:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-20 by ascillato2

glennmurley15, 2020-02-20:
I have searched the docs numerous times and the only thing I can find is LedState and LedPower, neither of which seem to do what I want

I literally want the wifi light to only turn on when connected to MQTT broker, and not to flash or blink when any relay state changes or any MQTT command issued etc, and then the light to turn off if MQTT connection is lost

If the commands can already do this, please can you specify which options I need? Thanks