arendst/Tasmota #7767: Tasmota bluetooth NRF24L01 with MJ-HT-V1/ LYWSD02 / CGG1

kroon040, 2020-02-21:
### PROBLEM DESCRIPTION
I have an wmos d1 mini and an NRF24L01 connected. I have 3 sensors for Xiaomi  MJ-HT-V1/ LYWSD02 / CGG1 , 1 meter away, only 1 sensor works (LYWDSD02)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X ] Device used (wmemos d1 mini with NRF24L01 and  MJ-HT-V1/ LYWSD02 / CGG1): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ X] Self-compiled
    - [ ] IDE / Compiler used:  tasmota 8.1.0.9 (atom / 2_6_1/2.2.2-dev)
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
#define USE_SPI                              
#ifdef USE_SPI
  #define USE_NRF24                             
  #ifdef USE_NRF24
    #define USE_MIBLE 

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:13:20 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sensors","kroonen"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:13:20 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":472,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8F4000CF","04000000","200617CD","01001BC0","03000180","00000000"],"Drivers":"1,2,3,4,9,10,12,27,33","Sensors":"1,2,3,5,6,7,8,9,10,14,15,26,34,61"}}
11:13:20 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-1816","IPAddress":"192.168.180.76","Gateway":"192.168.180.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.180.150","Mac":"5C:CF:7F:0B:67:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
11:13:20 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.180.5","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B6718","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
11:13:20 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-02-21T10:13:20","Local":"2020-02-21T11:13:20","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:48","Sunset":"18:19"}}
11:13:20 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-02-21T11:13:20","LYWSD02-b14c68":{"Temperature":8.9,"Humidity":79.0},"TempUnit":"C"}}
11:13:20 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-02-21T11:13:20","Uptime":"0T03:00:20","UptimeSec":10820,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sensors","BSSId":"48:4A:E9:35:2F:84","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
At least the MJ_HT_V1 should work to, and maybe we I can help to tupprtt the cgg1 (cleargrass)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
I can help with debug?

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-21:
Hi,

Your status 0 is incomplete, please add the missing information. 

Also please provide the output of the Console log output when you experience your issue. (Please use weblog 4 for more debug information)

kroon040, 2020-02-21:
status 0:
13:00:13 MQT: stat/wemosbt/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sensors","kroonen"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
13:00:13 MQT: stat/wemosbt/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":472,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8F4000CF","04000000","200617CD","01001BC0","03000180","00000000"],"Drivers":"1,2,3,4,9,10,12,27,33","Sensors":"1,2,3,5,6,7,8,9,10,14,15,26,34,61"}}
13:00:13 MQT: stat/wemosbt/STATUS5 = {"StatusNET":{"Hostname":"wemosbt-1816","IPAddress":"192.168.180.76","Gateway":"192.168.180.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.180.150","Mac":"5C:CF:7F:0B:67:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:00:13 MQT: stat/wemosbt/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.180.5","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B6718","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:00:13 MQT: stat/wemosbt/STATUS7 = {"StatusTIM":{"UTC":"2020-02-21T12:00:13","Local":"2020-02-21T13:00:13","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:48","Sunset":"18:19"}}
13:00:13 MQT: stat/wemosbt/STATUS10 = {"StatusSNS":{"Time":"2020-02-21T13:00:13","LYWSD02-b14c68":{"Temperature":8.6,"Humidity":80.0},"TempUnit":"C"}}
13:00:13 MQT: stat/wemosbt/STATUS11 = {"StatusSTS":{"Time":"2020-02-21T13:00:13","Uptime":"0T01:16:40","UptimeSec":4600,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sensors","BSSId":"48:4A:E9:35:2F:84","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:06"}}}

kroon040, 2020-02-21:
```
13:02:07 CMD: weblog 4
13:02:07 MQT: stat/wemosbt/RESULT = {"WebLog":4}
13:02:07 CFG: Saved to flash at F7, Count 72, Bytes 4096
13:02:11 MQT: stat/wemosbt/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sensors","kroonen"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
13:02:11 MQT: stat/wemosbt/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":472,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8F4000CF","04000000","200617CD","01001BC0","03000180","00000000"],"Drivers":"1,2,3,4,9,10,12,27,33","Sensors":"1,2,3,5,6,7,8,9,10,14,15,26,34,61"}}
13:02:11 MQT: stat/wemosbt/STATUS5 = {"StatusNET":{"Hostname":"wemosbt-1816","IPAddress":"192.168.180.76","Gateway":"192.168.180.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.180.150","Mac":"5C:CF:7F:0B:67:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
13:02:11 MQT: stat/wemosbt/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.180.5","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B6718","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
13:02:11 MQT: stat/wemosbt/STATUS7 = {"StatusTIM":{"UTC":"2020-02-21T12:02:11","Local":"2020-02-21T13:02:11","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:48","Sunset":"18:19"}}
13:02:11 MQT: stat/wemosbt/STATUS10 = {"StatusSNS":{"Time":"2020-02-21T13:02:11","LYWSD02-b14c68":{"Temperature":8.6,"Humidity":80.0},"TempUnit":"C"}}
13:02:11 MQT: stat/wemosbt/STATUS11 = {"StatusSTS":{"Time":"2020-02-21T13:02:11","Uptime":"0T01:18:38","UptimeSec":4718,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sensors","BSSId":"48:4A:E9:35:2F:84","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:06"}}}
13:02:20 WIF: Checking connection...
13:02:40 WIF: Checking connection...
13:03:00 WIF: Checking connection...
13:03:20 WIF: Checking connection...
13:03:40 WIF: Checking connection...
13:03:45 MQT: tele/wemosbt/STATE = {"Time":"2020-02-21T13:03:45","Uptime":"0T01:20:12","UptimeSec":4812,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sensors","BSSId":"48:4A:E9:35:2F:84","Channel":1,"RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:06"}}
13:03:45 MQT: tele/wemosbt/SENSOR = {"Time":"2020-02-21T13:03:45","LYWSD02-b14c68":{"Temperature":8.7,"Humidity":80.0},"TempUnit":"C"}
```

ascillato, 2020-02-21:
Status 0 is still incomplete

kroon040, 2020-02-21:
runnning twice but only get this

```
14:27:21 MQT: stat/wemosbt/STATUS2 = {"StatusFWR":{"Version":"8.1.0.9(tasmota)","BuildDateTime":"2020-02-21T08:08:59","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"361/699"}}
14:27:21 MQT: stat/wemosbt/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sensors","kroonen"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
14:27:21 MQT: stat/wemosbt/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":472,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8F4000CF","04000000","200617CD","01001BC0","03000180","00000000"],"Drivers":"1,2,3,4,9,10,12,27,33","Sensors":"1,2,3,5,6,7,8,9,10,14,15,26,34,61"}}
14:27:21 MQT: stat/wemosbt/STATUS5 = {"StatusNET":{"Hostname":"wemosbt-1816","IPAddress":"192.168.180.76","Gateway":"192.168.180.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.180.150","Mac":"5C:CF:7F:0B:67:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:27:21 MQT: stat/wemosbt/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.180.5","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B6718","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:27:21 MQT: stat/wemosbt/STATUS7 = {"StatusTIM":{"UTC":"2020-02-21T13:27:21","Local":"2020-02-21T14:27:21","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:48","Sunset":"18:19"}}
14:27:21 MQT: stat/wemosbt/STATUS10 = {"StatusSNS":{"Time":"2020-02-21T14:27:21"}}
14:27:21 MQT: stat/wemosbt/STATUS11 = {"StatusSTS":{"Time":"2020-02-21T14:27:21","Uptime":"0T00:00:19","UptimeSec":19,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sensors","BSSId":"48:4A:E9:35:2F:84","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:06"}}}
14:27:24 MQT: stat/wemosbt/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["sensors","kroonen"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
14:27:24 MQT: stat/wemosbt/STATUS4 = {"StatusMEM":{"ProgramSize":532,"Free":472,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8F4000CF","04000000","200617CD","01001BC0","03000180","00000000"],"Drivers":"1,2,3,4,9,10,12,27,33","Sensors":"1,2,3,5,6,7,8,9,10,14,15,26,34,61"}}
14:27:24 MQT: stat/wemosbt/STATUS5 = {"StatusNET":{"Hostname":"wemosbt-1816","IPAddress":"192.168.180.76","Gateway":"192.168.180.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.180.150","Mac":"5C:CF:7F:0B:67:18","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
14:27:24 MQT: stat/wemosbt/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.180.5","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0B6718","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
14:27:24 MQT: stat/wemosbt/STATUS7 = {"StatusTIM":{"UTC":"2020-02-21T13:27:24","Local":"2020-02-21T14:27:24","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:48","Sunset":"18:19"}}
14:27:24 MQT: stat/wemosbt/STATUS10 = {"StatusSNS":{"Time":"2020-02-21T14:27:24","LYWSD02-b14c68":{"Temperature":8.7},"TempUnit":"C"}}
14:27:24 MQT: stat/wemosbt/STATUS11 = {"StatusSTS":{"Time":"2020-02-21T14:27:24","Uptime":"0T00:00:22","UptimeSec":22,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Wifi":{"AP":1,"SSId":"sensors","BSSId":"48:4A:E9:35:2F:84","Channel":1,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:06"}}}
14:27:29 WIF: Checking connection...
```

ascillato, 2020-02-21:
As the example of my previous comment, there are some lines missing

Staars, 2020-02-21:
Okay:
The CGG1 simply can not work, because it must be added to the driver code. I do not have this sensor.

The MJ-HT-V1 should work, but of course the are lots of opportunities to fail.

For both cases it would be helpful for me to get the BT-adress (MAC). A generic smartphone-tool like nRF-connect should do the job.

kroon040, 2020-02-21:
is this ok. the 3 sensors (a rpi was also in close there)

sudo hcitool lescan
LE Scan ...
4C:65:A8:D1:7D:DC (unknown)
4C:65:A8:D1:7D:DC MJ_HT_V1
58:2D:34:10:C8:6B (unknown)
58:2D:34:10:C8:6B ClearGrass Temp & RH
E7:2E:00:B1:4C:68 (unknown)
E7:2E:00:B1:4C:68 LYWSD02

Staars, 2020-02-21:
Interesting and that explains the problem for me.
The good thing is, that this will be probably solvable. (Bad thing is, I do not have much spare time in the coming days).

It does not come to a surprise, that not all assumption I made were correct. Here we see, that the OUI of the MAC is not safe to use as a discriminator between the different sensor types. The OUI of my MJ_HT_V1 is equal to the OUI of your CGG1. So this must fail with the current driver.

Please try to catch an BLE-Advertisement packet for the two „problematic“ sensors and post the result.

I will have to re-write some part of the code and this will take a while.

kroon040, 2020-02-21:
How do I catch those? 

with gattool?  or?

Staars, 2020-02-21:
On the Mac I use the developer tools from Xcode oder sometimes "BlueSee".
For the other platforms I would have to google it myself.

kroon040, 2020-02-21:
I installed bluesee, is this the right info?

**first 4C:65:A8:D1:7D:DC MJ_HT_V1**

```
18:52:34.1120: connected in underlying BLE layer.
18:52:34.1150: discovered services: (
    "Device Information",
    FE95,
    "226C0000-6476-4566-7562-66734470666D",
    Battery,
    "00001530-1212-EFDE-1523-785FEABCD123"
)
18:52:34.2900: discovered characteristics for service 226C0000-6476-4566-7562-66734470666D: (
    "226CAA55-6476-4566-7562-66734470666D",
    "226CBB55-6476-4566-7562-66734470666D"
)
18:52:34.3500: discovered characteristics for service 180F: (
    "Battery Level"
)
18:52:34.4410: discovered characteristics for service 180A: (
    "Manufacturer Name String",
    "Model Number String",
    "Serial Number String",
    "Hardware Revision String",
    "Firmware Revision String"
)
18:52:34.5600: discovered characteristics for service 00001530-1212-EFDE-1523-785FEABCD123: (
    "00001532-1212-EFDE-1523-785FEABCD123",
    "00001531-1212-EFDE-1523-785FEABCD123",
    "00001534-1212-EFDE-1523-785FEABCD123"
)
18:52:34.6960: discovered characteristics for service FE95: (
    0001,
    0002,
    0004,
    0010,
    0013,
    0014
)
18:52:34.6960: state changed to 'Connected'.
18:52:34.9650: received update from characteristic 2A19___180F: <56>
18:52:35.0250: received update from characteristic 2A19___180F: <56>
18:52:35.3340: received update from characteristic 2A25___180A: <4c65a8d1 7ddcf7b3 2f65f116 c25e>
18:52:35.3550: updated subscription state for characteristic 2A19___180F: subscribed
18:52:35.3850: received update from characteristic 2A27___180A: <322e3030>
18:52:35.4150: received update from characteristic 2A29___180A: <436c6561 72677261 73732049 6e63>
18:52:35.4450: received update from characteristic 2A24___180A: <4475636b 5f52656c 65617365>
18:52:35.4520: received update from characteristic 2A25___180A: <4c65a8d1 7ddcf7b3 2f65f116 c25e>
18:52:40.2950: disconnected in underlying BLE layer with error: Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.}
```
**second : 8:2D:34:10:C8:6B ClearGrass Temp & RH**

```
18:53:22.1510: connected in underlying BLE layer.
18:53:22.1530: discovered services: (
    "Device Information",
    FE59,
    "22210000-554A-4546-5542-46534450464D",
    FE95
)
18:53:22.5250: discovered characteristics for service FE59: (
    "8EC90003-F315-4F60-9FB8-838830DAEA50"
)
18:53:22.7500: discovered characteristics for service 22210000-554A-4546-5542-46534450464D: (
    0001,
    0002,
    0003,
    0004,
    0100
)
18:53:22.9760: discovered characteristics for service 180A: (
    "Manufacturer Name String",
    "Model Number String",
    "Serial Number String",
    "Hardware Revision String",
    "Firmware Revision String"
)
18:53:23.3510: discovered characteristics for service FE95: (
    0001,
    0002,
    0004,
    0005,
    0010,
    0013,
    0014
)
18:53:23.3510: state changed to 'Connected'.
18:53:24.1000: received update from characteristic 2A25___180A: <36424338 31303334 32443538>
18:53:24.7000: received update from characteristic 2A27___180A: <30303031>
18:53:24.8500: received update from characteristic 2A29___180A: <51696e67 70696e67 20546563 686e6f6c 6f677920 28426569 6a696e67 2920436f 2e2c204c 74642e>
18:53:24.9250: received update from characteristic 2A24___180A: <43474731>
```

Staars, 2020-02-21:
No, only the advertisement is needed. In BLUESEE there should be a view, where you see all kind of devices including their ADV packet.

kroon040, 2020-02-21:
This value?

[image]

kroon040, 2020-02-21:
and the other one

[image]

Staars, 2020-02-21:
Yes, but for the MJ_HT_V1 you should get Svc: FE95 most of the time too and that is the one we need. Please send one.

Besides for the CGG1 more work is needed later on, I hope to get the MJ_HT_V1 fixed in the next few days and after that we take a look at the CGG1.

kroon040, 2020-02-21:
then this should be the right one

[image]

Staars, 2020-02-21:
Yes, thats right and looks as expected.
Now it is my turn, maybe I am able to do a bit of coding on sunday. Please have some patience.

kroon040, 2020-02-21:
I have patience, thnx for the support already!

Staars, 2020-02-22:
Time for some tests:
https://gist.github.com/Staars/16c8c45edfee4a4f97787dd753474786

Just replace the file and compile as usual.

1. For me this is functional as before.
2. This should fix your MJ_HT_V1 problem.
3. I would be surprised, if the CGG1 works too, because I could only do some very rough "guess-timations" for the PDU-types and fitting LSFR-bytes.

kroon040, 2020-02-22:
Well the MJ_HT_V1 is working, the CGG1, not yet

`21:46:37 MQT: tele/wemosbt/SENSOR = {"Time":"2020-02-22T21:46:37","MJ_HT_V1-d17ddc":{"Temperature":9.1,"Humidity":87.1,"Battery":85},"LYWSD02-b14c68":{"Temperature":9.2,"Humidity":88.0},"TempUnit":"C"}`

kroon040, 2020-02-22:
Als some info for the cleargrass cgg1 in this post

https://github.com/esphome/feature-requests/issues/381

Staars, 2020-02-22:
Cool (including your environment).
I will prepare a debug version of the driver to get the raw byte stream. With a bit of luck we have a working driver tomorrow.

kroon040, 2020-02-22:
It is my garage so it can be cool-;)

Good work!

Staars, 2020-02-23:
Here is the debug-driver:
https://gist.github.com/Staars/823aae89589ac1dc3fab3fa7b517f72d

You must uncomment `#define DEBUG_TASMOTA_SENSOR ` in my_user_config.h.

It will make life easier, if you remove all unnecessary BT-devices. Then open the serial console and watch out for printings, that contain `95 fe 47 03` in the last line (of 3). In an ideal scenario we get 3 different sorts of it (with different  lines 1 and 2 above it).
We need this data! (Fingers crossed ...)

kroon040, 2020-02-23:
Hi, 

I think this is the correct logging (with webklog 4), almost 95 fe 47 03 , but  95 fe 50 30 47 03

```
2:58:11 MINRF: Buffer: 6b 5b f4 17 92 57 33 eb 98 25 95 ea b6 db cc 0c db a4 84 c1 e2 e4 07 cf 05 51 5a 8d 9c 23 66 51 
12:58:11 MINRF: Buffer: 6d 64 20 11 2e 69 77 43 0d 57 52 23 3e 25 60 33 72 32 10 4c 17 70 7f 65 42 6f 29 55 1f 56 30 5d 
12:58:11 MINRF: Buffer: 00 1f 6b c8 10 34 2d 58 02 01 06 15 16 95 fe 50 30 47 03 13 6b c8 10 34 2d 58 0d 10 04 63 00 3a 
12:58:11 MINRF: Type: 3050
12:58:11 MINRF: unknown Packet
12:58:13 WIF: Checking connection...
12:58:23 MINRF: Buffer: b1 f3 3c 96 b4 c9 d2 17 ee 0c e8 da 01 47 60 cd 6e 75 15 63 af d9 c4 37 e9 03 70 2c 93 80 ff bb 
12:58:23 MINRF: Buffer: 42 6f 29 55 1f 56 30 5d 39 19 08 26 4f 38 7b 76 21 73 50 6e 4b 2b 18 6a 58 48 04 13 63 1c 79 3b 
12:58:23 MINRF: Buffer: 00 1d 6b c8 10 34 2d 58 02 01 06 13 16 95 fe 50 30 47 03 16 6b c8 10 34 2d 58 06 10 02 3a 03 ac 
12:58:23 MINRF: Type: 3050
12:58:23 MINRF: unknown Packet
12:58:31 MINRF: Buffer: 6b 9b f4 17 92 57 33 eb 98 25 95 0a b6 db cc 0c db a4 84 11 e2 e4 07 cf 05 51 ba 8d 3c 2f 33 6b 
12:58:31 MINRF: Buffer: 6d 64 20 11 2e 69 77 43 0d 57 52 23 3e 25 60 33 72 32 10 4c 17 70 7f 65 42 6f 29 55 1f 56 30 5d 
12:58:31 MINRF: Buffer: 00 1c 6b c8 10 34 2d 58 02 01 06 12 16 95 fe 50 30 47 03 18 6b c8 10 34 2d 58 0a 10 01 53 aa 66 
12:58:31 MINRF: Type: 3050
12:58:31 MINRF: unknown Packet
12:58:33 WIF: Checking connection...
12:58:39 MINRF: Buffer: f8 54 84 e9 a9 43 8d 43 c3 eb c3 ea 6c 33 04 8d fd 3a 65 ad 94 cd 7a 9f b2 cd 26 4c 49 68 37 7a 
12:58:39 MINRF: Buffer: 0f 1a 27 2d 46 7c 4a 49 66 6d 64 20 11 2e 69 77 43 0d 57 52 23 3e 25 60 33 72 32 10 4c 17 70 7f 
12:58:39 MINRF: Buffer: 00 1d 6b c8 10 34 2d 58 02 01 06 13 16 95 fe 50 30 47 03 1a 6b c8 10 34 2d 58 06 10 02 3a 03 ae 
12:58:39 MINRF: Type: 3050
12:58:39 MINRF: unknown Packet
12:58:43 MINRF: Buffer: f8 14 84 e9 a9 43 8d 43 c3 eb c3 8a 6c 33 04 8d fd 3a 65 2d 94 cd 7a 9f b2 cd f6 4c 29 f2 f7 53 
12:58:43 MINRF: Buffer: 0f 1a 27 2d 46 7c 4a 49 66 6d 64 20 11 2e 69 77 43 0d 57 52 23 3e 25 60 33 72 32 10 4c 17 70 7f 
12:58:43 MINRF: Buffer: 00 1f 6b c8 10 34 2d 58 02 01 06 15 16 95 fe 50 30 47 03 1b 6b c8 10 34 2d 58 0d 10 04 63 00 3a 
12:58:43 MINRF: Type: 3050
12:58:43 MINRF: unknown Packet
12:58:51 MINRF: Buffer: f8 54 84 e9 a9 43 8d 43 c3 eb c3 ea 6c 33 04 8d fd 3a 65 4d 94 cd 7a 9f b2 cd 66 4c 49 f2 f7 c0 
12:58:51 MINRF: Buffer: 0f 1a 27 2d 46 7c 4a 49 66 6d 64 20 11 2e 69 77 43 0d 57 52 23 3e 25 60 33 72 32 10 4c 17 70 7f 
12:58:51 MINRF: Buffer: 00 1d 6b c8 10 34 2d 58 02 01 06 13 16 95 fe 50 30 47 03 1d 6b c8 10 34 2d 58 04 10 02 63 00 f3 
12:58:51 MINRF: Type: 3050
12:58:51 MINRF: unknown Packet
12:58:53 WIF: Checking connection...
12:58:55 MINRF: Buffer: f8 54 84 e9 a9 43 8d 43 c3 eb c3 ea 6c 33 04 8d fd 3a 65 8d 94 cd 7a 9f b2 cd 26 4c 49 68 37 fa 
12:58:55 MINRF: Buffer: 0f 1a 27 2d 46 7c 4a 49 66 6d 64 20 11 2e 69 77 43 0d 57 52 23 3e 25 60 33 72 32 10 4c 17 70 7f 
12:58:55 MINRF: Buffer: 00 1d 6b c8 10 34 2d 58 02 01 06 13 16 95 fe 50 30 47 03 1e 6b c8 10 34 2d 58 06 10 02 3a 03 af 
12:58:55 MINRF: Type: 3050
12:58:55 MINRF: unknown Packet
12:59:13 WIF: Checking connection...
12:59:23 MINRF: Buffer: f8 54 84 e9 a9 43 8d 43 c3 eb c3 ea 6c 33 04 8d fd 3a 65 51 94 cd 7a 9f b2 cd 66 4c 49 f2 f7 90 
12:59:23 MINRF: Buffer: 0f 1a 27 2d 46 7c 4a 49 66 6d 64 20 11 2e 69 77 43 0d 57 52 23 3e 25 60 33 72 32 10 4c 17 70 7f 
12:59:23 MINRF: Buffer: 00 1d 6b c8 10 34 2d 58 02 01 06 13 16 95 fe 50 30 47 03 25 6b c8 10 34 2d 58 04 10 02 63 00 f9 
12:59:23 MINRF: Type: 3050
12:59:23 MINRF: unknown Packet 
```

Staars, 2020-02-23:
Yes, was a typo.
Stay tuned ....

Staars, 2020-02-23:
I updated:
https://gist.github.com/Staars/16c8c45edfee4a4f97787dd753474786

Please try it.

kroon040, 2020-02-23:
Good Job, it works.

`16:32:15 MQT: tele/wemosbt/SENSOR = {"Time":"2020-02-23T16:32:15","LYWSD02-b14c68":{"Temperature":9.8,"Humidity":89.0},"MJ_HT_V1-d17ddc":{"Temperature":9.8,"Humidity":88.4,"Battery":85},"CGG1-10c86b":{"Temperature":10.2,"Humidity":83.0,"Battery":83},"TempUnit":"C"}`

Staars, 2020-02-23:
Nice!
I will do the PR.

kroon040, 2020-02-23:
this issue can be closed,  added as PR en it works fine

closed 2020-02-23 by kroon040