arendst/Tasmota #7772: Reading power status end delay in Backlog

Fiege, 2020-02-22:
### PROBLEM DESCRIPTION
I've multiple Sonof basic switches running Tasmota 7.1.2.6 .
All behave the same when I run commands in Backlog and try to read the power state.
The delay command in the backlog seems to be aborted.

I turn on the loght for 60 seconds:
http://192.168.99.171/cm?cmnd=Backlog%20Power%20On%3bDelay%20600%3bPower%20Off

Which works okay, but when I read the relais status the delay seems to be ended and the power is going off.
Tried commands:
http://192.168.99.171/cm?cmnd=status4
http://192.168.99.171/cm?cmnd=power

Both switch off the light.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ ] Tasmota binary firmware version number used: 7.1.2.6
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Start Firefox and  use the provided links in the description.

### EXPECTED BEHAVIOUR
Light should stay on for 60 seconds.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-22:
As designed.

use instead command pulsetime.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-22 by ascillato2

Klaws--, 2020-12-16:
Well, works "As designed.", fine.

I question the design.

I have the issue that the delay (and the delay only!) is cancelled but the rest of the backlog commands continue to run. So there is a number of devices (controlled by an equal  number of Tasmota smart plugs) which need to be turned off in a specific order, after a specific delay. I use the backlog feature for that.

Occasionally, all devices are powered off nearly simultaneously, too early and definitely not in the required order. This appears to happen when some application running elsewhere (like on a mobile/cell phone), with a different IP address, polls the status of the devices. This is counter-intuitive and (in my case) counter-productive.