arendst/Tasmota #7775: blind with venetian and domoticz

megamarco833, 2020-02-23:
### PROBLEM DESCRIPTION
1. using tasmota blind also with venetian and buttons
2. report in domoticz the status of tasmota blinds

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): nodemcu with esp8266
- [ x] Tasmota binary firmware version number used: 8.1
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: tasmocompiler (used just to enable blinds module, nothing else changed)
- [ x] Flashing tools used: esp8266flasher
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
02:55:12 MQT: stat/tasmota_blind/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
02:55:12 MQT: stat/tasmota_blind/RESULT = {"Module":{"0":"Generic"}}
02:55:13 MQT: stat/tasmota_blind/RESULT = {"GPIO0":{"0":"Nessuno"},"GPIO1":{"0":"Nessuno"},"GPIO2":{"17":"Button1"},"GPIO3":{"0":"Nessuno"},"GPIO4":{"30":"Relay2i"},"GPIO5":{"29":"Relay1i"},"GPIO9":{"0":"Nessuno"},"GPIO10":{"0":"Nessuno"},"GPIO12":{"32":"Relay4i"},"GPIO13":{"18":"Button2"},"GPIO14":{"31":"Relay3i"},"GPIO15":{"0":"Nessuno"},"GPIO16":{"0":"Nessuno"}}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
02:56:20 MQT: stat/tasmota_blind/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota_blind","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmota_blind","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:56:20 MQT: stat/tasmota_blind/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:09:09","StartupUTC":"2020-02-23T00:47:11","Sleep":50,"CfgHolder":4617,"BootCount":15,"SaveCount":170,"SaveAddress":"3F6000"}}
02:56:20 MQT: stat/tasmota_blind/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2020-02-05T20:25:29","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"361/699"}}
02:56:20 MQT: stat/tasmota_blind/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NETGEAR",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000000","00000000"]}}
02:56:20 MQT: stat/tasmota_blind/STATUS4 = {"StatusMEM":{"ProgramSize":551,"Free":2520,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000410","8FDAE387","043683A1","22B617CF","01001BC0","00007981"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29","Sensors":"1,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
02:56:20 MQT: stat/tasmota_blind/STATUS5 = {"StatusNET":{"Hostname":"tasmota_blind-0912","IPAddress":"192.168.0.158","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"3C:71:BF:32:83:90","Webserver":2,"WifiConfig":4}}
02:56:20 MQT: stat/tasmota_blind/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.105","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_328390","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
02:56:20 MQT: stat/tasmota_blind/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 23 01:56:20 2020","Local":"Sun Feb 23 02:56:20 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:44","Sunset":"18:22"}}
02:56:20 MQT: stat/tasmota_blind/STATUS10 = {"StatusSNS":{"Time":"2020-02-23T02:56:20","Shutter1":{"Position":13,"direction":0},"Shutter2":{"Position":47,"direction":0}}}
02:56:20 MQT: stat/tasmota_blind/STATUS11 = {"StatusSTS":{"Time":"2020-02-23T02:56:20","Uptime":"0T01:09:09","UptimeSec":4149,"Vcc":2.946,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"NETGEAR","BSSId":"80:3F:5D:6E:37:67","Channel":13,"RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

 **question1**
i have a motor that control **UP** and **Down** of a venetian blind. I have two venetian blind to be controlled in this way
**two relay**: relay1= up ; relay2=down
**two buttons**: button1= single press = 0,5sec relay1=on => orienting up venetian up
button1=long press(>2seconds) => relay1=on for ShutterOpenDuration1 30 (30seconds continuos triggering relay1 to open completly)
button2=single press = 0,5sec relay2=on => orienting up venetian down
button2=long press(>2seconds) => relay2=on for ShutterCloseDuration1 25 (25seconds continuos triggering relay2 to close completly)

same stuff for venetian number 2

i set in this way: inside console:
SetOption80 1
ShutterRelay1 1
ShutterRelay2 3

Backlog PulseTime1 0; PulseTime2 0; PulseTime3 0; PulseTime4 0
Backlog Interlock 1,2 3,4; Interlock ON
ShutterOpenDuration1 30
ShutterOpenDuration2 30
ShutterCloseDuration1 25
ShutterCloseDuration2 25

all works fine using web page and slide inside web page.

the issue is how i have to set inside the console the:
button1 and button 2 for venetian 1
and
button 3 and button 4 for venetian 2

to have the behaviour descrived above??
is it possible?
i do not succeed...:( is it possible to have it?
thanks to anyone for help!

**question2**
second part of the issue is related to domoticz.
is it possible to report in domoticz the percentage/status of open/close blind?
how?
i try to create two dummy devices in domoticz and set as blind, and inside tasmota, at domoticz configuration, insert the idx of devices, but nothing change inside domoticz, why?

### EXPECTED BEHAVIOUR
orienting venetian using short press button
open or close venenita using long press button
report in domoticz the status of venetian with percentages

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-23:
Yes, the behaviour you want can be done using rules. Please, check the examples on the docs at rules cookbook.

Question 1, you need to define thw buttons 3 and 4 in module config. You have just defined buttons 1 and 2.
Then those behaviors need to be configured using rules. 

Question 2, search in issues for examples. Several users have configured rules to publish data to domoticz.

If you need further assistant on the rules, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-23 by ascillato2

megamarco833, 2020-02-23:
thanks, question1: during my tentative i delete button2 and button3 to be concentrate on "how works" one blind (relay1 and relay2 button1 and button2) and avoid to create mess :-)

i tryed to use rules for this scope but i do not succeed on send to domoticz mqtt to update position of venetian blind 
i asked to chat comunity, but till now not receive any answer / help
thanks again for support