arendst/Tasmota #7778: Switch behaviour differs from documentation for switchmode 6

Phonog, 2020-02-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
In switchmode 6 the HOLD state is generated 4s after the button is released, instead of 4s of holding the button as the documentation says.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Dual R2
- [ x] Tasmota binary firmware version number used: 8.1.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: ESP Easy Flasher 0.040.007
- [ x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
11:37:06 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
11:37:07 RSL: stat/tasmota/RESULT = {"Module":{"39":"Sonoff Dual R2"}}
11:37:07 RSL: stat/tasmota/RESULT = {"GPIO0":{"10":"Switch2"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"11":"Switch3"},"GPIO10":{"17":"Button1"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
11:39:10 RSL: stat/tasmota/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
11:39:10 RSL: stat/tasmota/RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":463,"Rules":"on switch2#state == 3 do power1 on endon rule2 1"}
11:39:10 RSL: stat/tasmota/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [ x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
11:39:48 RSL: stat/tasmota/STATUS = {"Status":{"Module":39,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,5,6,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:39:48 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:10:05","StartupUTC":"2020-02-23T09:29:43","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":228,"SaveAddress":"F8000"}}
11:39:48 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(knx)","BuildDateTime":"2019-12-25T12:37:34","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"319/699"}}
11:39:48 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["bridge",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
11:39:48 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":576,"Free":424,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","9FDAE197","042683B0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,11,12,16,18,19,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
11:39:48 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-2225","IPAddress":"192.168.1.98","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"60:01:94:E5:C8:B1","Webserver":2,"WifiConfig":4}}
11:39:48 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E5C8B1","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:39:48 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Sun Feb 23 10:39:48 2020","Local":"Sun Feb 23 11:39:48 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:44","Sunset":"18:22"}}
11:39:48 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-02-23T11:39:48","Switch2":"ON","Switch3":"OFF"}}
11:39:48 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-02-23T11:39:48","Uptime":"0T01:1

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Configure GPIOS as given above. Attach to button0 and button1 terminals of the board two normally open pushbuttons, which will be given the logical names button2 and button3. Connect a lamp at each of out1 and out2.
On the console type:
switchmode2 6
switchmode3 6
on switch2#state == 3 do power1 on endon
rule2 1
Play with the buttons as follows:
Press button2 once shortly. Output2 should by default switch on, but it does not. Four seconds later, output1 switches on because of the rule.
Press button3 shortly. Output1 should switch off by default, but it does not. Press button3 again shortly: output1 toggles only if the interval between the two presses is shorter than 4s, otherwise it does nothing. The same holds true for button2.

To summarize: the hold state is generated 4s after the button is released. if the button is pressed shortly later than 4s after the previous press, it does nothing.  

### EXPECTED BEHAVIOUR
That described in the documentation:
-short presses must always be taken into account, whatever the delay from the last action. 
-HOLD state must be generated 4 seconds after the button is pressed, provided it is held longer than this duration.
-The toggle action must occur when the button is pressed (this is currently correct).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXTURE
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2020-02-23:
As designed 

> HOLD state must be generated 4 seconds after the button is pressed, provided it is held longer than this duration.

Sorry, but no because if you want the full reset (40 seconds pressed button), you will have a Hold message and then the full reset (2 different actions with the same event from user).

closed 2020-02-23 by ascillato2

Phonog, 2020-02-25:
Why do you mention the full reset? This is not my concern. I do not press the button longer than the hold time. In addition, in my application, I configured the inputs as switches, not buttons, and switches do no produce a reset, whatever the duration of their actuation. 
All what I say is that unlike what the documentation says, with SWITCHMODE 6 the hold state is generated AFTER THE BUTTON IS RELEASED, whereas the documentation says WHILE THE BUTTON IS HELD FOR MORE THAN 4 SECONDS (if option 32 is left at its default value).
I have looked for the code that handles the switch in the file support_switch.ino.
I think the problem lies in the following block:

`        case PUSHBUTTONHOLD_INV:
          if ((NOT_PRESSED == button) && (PRESSED == Switch.last_state[i])) {
            Switch.hold_timer[i] = loops_per_second * Settings.param[P_HOLD_TIME] / 10;
          }
          if ((PRESSED == button) && (NOT_PRESSED == Switch.last_state[i]) && (Switch.hold_timer[i])) {
            Switch.hold_timer[i] = 0;
            switchflag = POWER_TOGGLE;   // Toggle with pushbutton to Gnd
          }
          break;
`We see that the first line detects the "down" to "up" transition of the switch to start the hold timer. Actually it should start on the "up" to "down" transition. This can be done by changing this first line to the following code:

`          if ((PRESSED == button) && (NOT_PRESSED == Switch.last_state[i])) {
            Switch.hold_timer[i] = loops_per_second * Settings.param[P_HOLD_TIME] / 10;
          }`

Thank you to have a look at this. I hope this will help.

PR #7831 (2020-03-01): Fix switchmode 6 according to issue 7778
## Description:

**Related issue (if applicable):** fixes #7778

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).