arendst/Tasmota #7779: Setting the color with RGB value reset setoption17

Ysbrand, 2020-02-23:
### PROBLEM DESCRIPTION
If got setoption17 set to 1

The moment I issue the following command:
Color EDA20C0000

Setoption17 is automatically set to off

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): H801
- [X ] Tasmota binary firmware version number used: 8.1.0
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used: Regular upgrade
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
14:16:51 CMD: Backlog Template; Module; GPIO 255
14:16:52 MQT: stat/Ledstrip1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
14:16:52 MQT: stat/Ledstrip1/RESULT = {"Module":{"20":"H801"}}
14:16:52 MQT: stat/Ledstrip1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"52":"Led1"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"41":"PWM5"},"GPIO5":{"57":"Led2i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"39":"PWM3"},"GPIO13":{"38":"PWM2"},"GPIO14":{"40":"PWM4"},"GPIO15":{"37":"PWM1"},"GPIO16":{"0":"None"}}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:

  STATUS 0 output here:
14:17:30 MQT: stat/Ledstrip1/STATUS11 = {"StatusSTS":{"Time":"2020-02-23T14:17:30","Uptime":"3T04:25:16","UptimeSec":275116,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"OFF","Dimmer":93,"Color":"237,162,12,0,0","HSBColor":"40,95,93","Channel":[93,64,5,0,0],"CT":153,"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"YBROUTER-2G","BSSId":"FC:EC:DA:81:90:CE","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

14:11:12 CMD: setoption17 on
14:11:12 MQT: stat/Ledstrip1/RESULT = {"SetOption17":"ON"}
14:11:22 CMD: Color EDA20C0000
14:11:22 MQT: tele/Ledstrip1/STATE = {"Time":"2020-02-23T14:11:22","Uptime":"3T04:19:08","UptimeSec":274748,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":3,"POWER":"OFF","Dimmer":93,"Color":"EDA20C0000","HSBColor":"40,95,93","Channel":[93,64,5,0,0],"CT":153,"Scheme":0,"Fade":"ON","Speed":5,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"YBROUTER-2G","BSSId":"FC:EC:DA:81:90:CE","Channel":11,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:06"}}
14:11:22 MQT: stat/Ledstrip1/RESULT = {"POWER":"OFF","Dimmer":93,"Color":"EDA20C0000","HSBColor":"40,95,93","Channel":[93,64,5,0,0],"CT":153}
14:11:31 CMD: setoption17
14:11:31 MQT: stat/Ledstrip1/RESULT = {"SetOption17":"OFF"}
14:12:00 CMD: setoption17 on
14:12:00 MQT: stat/Ledstrip1/RESULT = {"SetOption17":"ON"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
setoption17 to on
Color EDA20C0000

setoption17 is now off

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expect that setoption17 continues to be set to on

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-02-23:
As designed.

If you enter a hexadecimal color command it automagically switches SetOption17 to hexadecimal (0) too. If you want decimal output then enter colors also as decimal numbers.

Ysbrand, 2020-02-23:
Shouldn't this be documented? It took me several days before I realized that a rule was changing setoption17 (which did conflict with my Home Assistant setup).

closed 2020-02-23 by ascillato2