arendst/Tasmota #7784: Support for multiple HX711 or to read both channels

denespeti, 2020-02-24:
**Have you looked for this feature in other issues and in the docs?**  
yes
**Is your feature request related to a problem? Please describe.**  
Cannot connect to more than 4 load cells with one device. A huge loss of resources.

**Describe the solution you'd like**  
I need to read 8 load cells via hx711. Either to be able to read Channel B or to be able to connect 2 hx 711s to read two bridges.
After seeing this article: https://github.com/compugician/HX711-multi
Must be possible.

**Describe alternatives you've considered**  
I might install 4 dummy resistances and only connect the push sides of the load cells, but then there is a huge measurement error.

**Additional context**  
I dont have.

**(Please, remember to close the issue when the problem has been addressed)**  
I Will.

closed 2020-03-26 by stale[bot]

rt400, 2020-11-24:
@denespeti Did you succeed to connect multiple HX711 ?

Trrw, 2022-05-14:
Is this possible?

sfromis, 2022-05-14:
If someone does the work to support it. Or get two ESP boards :-)