arendst/Tasmota #7789: Smart Meter Interface cannot specify MODBUS unit id

TheChatty, 2020-02-24:
### PROBLEM DESCRIPTION
I compiled Tasmota like [this](https://github.com/arendst/Tasmota/wiki/smart-meter-interface). I want SMI to send MODBUS unit id 247. But it always sends unit id 1.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP-01s (black board, 1MB)
- [x] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [ ] IDE / Compiler used: gitpod
- [x] Flashing tools used: esptool.py
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
CMD: Backlog Template; Module; GPIO 255
RSL: stat/tasmota_0BBCBE/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
RSL: stat/tasmota_0BBCBE/RESULT = {"Module":{"0":"Generic"}}
RSL: stat/tasmota_0BBCBE/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
CMD: Status 0
RSL: stat/tasmota_0BBCBE/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota_0BBCBE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
RSL: stat/tasmota_0BBCBE/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Exception","Uptime":"0T00:39:35","StartupUTC":"2020-02-24T18:02:30","Sleep":50,"CfgHolder":4617,"BootCount":22,"BCResetTime":"2020-02-23T09:49:08","SaveCount":46,"SaveAddress":"F6000"}}
RSL: stat/tasmota_0BBCBE/STATUS2 = {"StatusFWR":{"Version":"8.1.0.9(tasmota)","BuildDateTime":"2020.02.22 23:01:29","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"361/699"}}
RSL: stat/tasmota_0BBCBE/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hinterhof",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
RSL: stat/tasmota_0BBCBE/STATUS4 = {"StatusMEM":{"ProgramSize":590,"Free":412,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":2,"Features":["00000407","87DAE397","043E83A1","22B617CD","01001BC0","00007889","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34,53"}}
RSL: stat/tasmota_0BBCBE/STATUS5 = {"StatusNET":{"Hostname":"tasmota_0BBCBE-7358","IPAddress":"192.168.178.42","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"5C:CF:7F:0B:BC:BE","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
RSL: stat/tasmota_0BBCBE/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0BBCBE","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
RSL: stat/tasmota_0BBCBE/STATUS7 = {"StatusTIM":{"UTC":"2020-02-24T18:42:05","Local":"2020-02-24T19:42:05","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:42","Sunset":"18:24"}}
RSL: stat/tasmota_0BBCBE/STATUS10 = {"StatusSNS":{"Time":"2020-02-24T19:42:05","MODBUS":{"Voltage_P1":0.00},"MODBUS":{"Current_P1":0.00}}}
RSL: stat/tasmota_0BBCBE/STATUS11 = {"StatusSTS":{"Time":"2020-02-24T19:42:05","Uptime":"0T00:39:35","UptimeSec":2375,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Hinterhof","BSSId":"2C:3A:FD:A2:59:11","Channel":11,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:06"}}}
RSL: stat/tasmota_0BBCBE/STATUS12 = {"StatusSTK":{"Exception":0,"Reason":"Exception","EPC":["4000dce5","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000","CallChain":["4022a580","4010070c","40100720","40100734","40100748","40285228","4024b9d9","40244c0a","40285228","4020630c","4024ba1c","401018f4","40285228","40285228","40101b7f","4024ba1c","402449f1","40244a10","4024baa0","40289abf","402064b4","40284b54","4022a73a","402315a0","4021023e","40281740","402814e0","402814e8","40210264","402814e8","40202b8e"]}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Compile Tasmota for SMI, configure a script with multiple MODBUS meters.

### EXPECTED BEHAVIOUR
Tasmota sending the modbus read register message with specified unit id.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Sent unid id is always 1.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-02-24:
As designed: 
> Declare a script >M section with the number of connected meters (n = 1..5)
See docu https://tasmota.github.io/docs/#/peripherals/Smart-Meter-Interface

TheChatty, 2020-02-24:
The problem is, I cannot configure which unit id is used in the modbus request. It's always '1'. But my smart meter listens to unit 247.

kugelkopf123, 2020-02-25:
> The problem is, I cannot configure which unit id is used in the modbus request. It's always '1'. But my smart meter listens to unit 247.

No.
My Modbus Meter has the ID 2 and I use it with this.
The ID from the Meter is configured in the requestregister and descriptor. 

If you are german: [HowTo](https://docs.google.com/document/d/1olSGZcaE_vkdNXzN_p0zGee61xKCcK3HJe3Al00xj2g/edit?usp=sharing)
Section Modbus RTU.
There is an Example of my ALE3.

TheChatty, 2020-02-25:
It works. The documentation of [<cmdTelegram>](https://github.com/arendst/Tasmota/wiki/smart-meter-interface) is, to say the least, incomplete.
Thanks for the hint. Could read you document easily ;-)

closed 2020-02-25 by TheChatty

meingraham, 2020-02-25:
@TheChatty @kugelkopf123 

The Tasmota community would be most appreciative, I am sure, if you could expand/enhance the Tasmota Docs on this topic. Would you guys be willing to add the information in that blog post? The docs are available and open for the community to contribute.

kugelkopf123, 2020-02-25:
@meingraham 
I've already started with that. However, my PR has been waiting to be merged for 5 days. Without any reason....

meingraham, 2020-02-25:
I must have missed it. Can you send that docs PR link so I can review?

kugelkopf123, 2020-02-25:
@meingraham https://github.com/tasmota/docs/pull/114 :)

meingraham, 2020-02-25:
Thx! I have asked Gerhard (@gemu2015 ) to review.