arendst/Tasmota #7793: Shutterbutton does not work for 1 button

stefanbode, 2020-02-25:
### PROBLEM DESCRIPTION
Defining: ShutterButton1 1 100 0
If doing the action the following code will always report a SIMULTANEOUS MULTI PRESS EVENT. 

related code segment in xdrv27_shutter
```
          uint32 min_shutterbutton_press_counter = -1;
          for (uint32_t i = 0; i < MAX_KEYS; i++) {
            AddLog_P2(LOG_LEVEL_DEBUG, PSTR("SHT: Settings.shutter_button[i] %ld, shutter_index %d, Button.press_counter[i] %d, min_shutterbutton_press_counter %d, i %d"), Settings.shutter_button[i], shutter_index, Button.press_counter[i] , min_shutterbutton_press_counter, i);
            if ((Settings.shutter_button[i] & (1<<31)) && ((Settings.shutter_button[i] & 0x03) == shutter_index) && (Button.press_counter[i] < min_shutterbutton_press_counter)) {
              min_shutterbutton_press_counter = Button.press_counter[i];
              AddLog_P2(LOG_LEVEL_DEBUG, PSTR("SHT: min_shutterbutton_press_counter %d"), min_shutterbutton_press_counter);
            }

          }
          if (min_shutterbutton_press_counter == Button.press_counter[button_index]) {
            // simultaneous shutter button press detected
            AddLog_P2(LOG_LEVEL_DEBUG, PSTR("SHT: simultanous presss deteced"));
```

Output from the Debug LOG:
```
13:22:52 WIF: Checking connection...
13:22:56 SHT: Settings.shutter_button[i] -2147483188, shutter_index 0, Button.press_counter[i] 1, min_shutterbutton_press_counter -1
13:22:56 SHT: min_shutterbutton_press_counter 1
13:22:56 SHT: Settings.shutter_button[i] 0, shutter_index 0, Button.press_counter[i] 0, min_shutterbutton_press_counter 1
13:22:56 SHT: Settings.shutter_button[i] 0, shutter_index 0, Button.press_counter[i] 0, min_shutterbutton_press_counter 1
13:22:56 SHT: Settings.shutter_button[i] 0, shutter_index 0, Button.press_counter[i] 0, min_shutterbutton_press_counter 1
13:22:56 SHT: simultanous presss deteced
13:22:56 MQT: hm/status/ESP_AC976F/RESULT = {"Shutter1":{"Button0":1}}
```

In my opinion, the if does not work when there is only one button. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____
- [x ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ x] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

to-scho, 2020-03-01:
You are right, I will take care for a fix

stefanbode, 2020-03-01:
Great. I am not able to get behind the math you want to call here. There is a similar subroutine but this is doing something on the hold event. Anyhow the time for mulitipress seems to be 0.5 seconds. This is quite fast for an old guy :-)

to-scho, 2020-03-02:
Code change is done but I need to find some time for testIng.

The 0.5s are inherited from tasmotas original button implementation. Extending it would result in slower reaction of the button since nothing is happening before that time after final button release. I can play with it to see how e.g. 1s feels.

closed 2020-03-02 by arendst

antonka2000, 2020-04-07:
in 8.2.0 it looks like the changes are inserted. But I still could not verify a proper funktion:
ShutterButton1 1 updown 0  - did not work for me.

Workaround:
rule1 on Power1#state==1 do ShutterButton1 1 down endon on Power2#state==1 do ShutterButton1 1 up endon

to-scho, 2020-04-08:
I tested single button in 8.2.0 for one of my shutters today and it works like expected. 

From your workaround I assume that you expect that the movement changes with every button press. Instead of that the implementation move the shutter up on a single press and down on a double press of the assigned button. 

I think your expectation is a more intuitive solution i.e. move the shutter down if its current position is >50 and up when current position <50. If you would like to have such behaviour please open a new issue as an enhancement request.

PR #7844 (2020-03-02): Shutterbutton does not work for 1 button, fixes #7793
## Description:

**Related issue (if applicable):** fixes #7793

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
commit 02ba705e (2020-03-02): Shutterbutton does not work for 1 button, fixes #7793
commit 4cb26902 (2020-03-02): Merge pull request #7844 from to-scho/hotfix/shutterbuttons_singlebutton

Shutterbutton does not work for 1 button, fixes #7793