arendst/Tasmota #7798: RGBCCT bulbs: White Temperature mapping with Alexa Whites temps

Bidulglloq, 2020-02-26:
### PROBLEM DESCRIPTION
Alexa's CCT commands (aka "Alexa, turn <lamp> to [Warm | Soft | Daylight | Cool] White") don't match the full range of CT values in Tasmota on a RGB+CCT lightbulb (Teckin SB50-2) 

### REQUESTED INFORMATION
How can I "remap" these Alexa CT names to match the full range of CT available on the bulb (or a customised range)?

### TO REPRODUCE
From the Tasmota Console on the bulb, I get these messages when asking **Alexa** for the following Color temperatures:
**"Warm White"**     => "Channel":[0,0,0,**34,66**]  and  **"CT":383**  
**"Soft White"**        => "Channel":[0,0,0,**43,57**]  and  **"CT":350**  
**"White"**                => "Channel":[0,0,0,**62,38**]  and  **"CT":284**  
**"Daylight White"** => "Channel":[0,0,0,**76,24**]  and  **"CT":234**  
**"Cool White"**       => "Channel":[0,0,0,**83,13**]  and  **"CT":199**  

### EXPECTED BEHAVIOUR
I would like to either:
**Option 1**: Have "Warm white" Matching [0,0,0,**0,100**] (**CT=500**, its maximum)
                Have "Cool white"   Matching [0,0,0,**100,0**] (**CT=153**, its minimum)

**Option 2**: Have the ability to define my own range of CT mapped to the Alexa's values
                 Example: Alexa's "Warm White" means Tasmota's CT = 450
                                  Alexa's "Cool White" means Tasmota's CT = 200
                                 So all other Temperatures are interpolated in between

Option 2 preferred, of course, to match the different types/brands of bulbs that may have different wW and cW leds

I had an extensive look at the Console options and didn't find anything

Jason2866, 2020-02-26:
Have you seen `SetOption82` ?

s-hadinger, 2020-02-26:
Correct, use `SetOption 82 1`

https://tasmota.github.io/docs/#/Commands?id=setoption82

Bidulglloq, 2020-02-26:
Didn't work: 
SetOption82 was "0/Off" (that's supposed to give access to the full 153-500 CT range)

Changed to SetOption82 1
Same as before: Warm white => CT:383  and Cool White => CT:199

s-hadinger, 2020-02-26:
CT is unchanged but the split between CW and WW channels is changed.

Bidulglloq, 2020-02-26:
> CT is unchanged but the split between CW and WW channels is changed.

Nope, values are still [0,0,0,34,66] for warm white and [0,0,0,87,13] for cool white

Bidulglloq, 2020-02-26:
```
12:51:32 CMD: SetOption82 0
12:51:32 RSL: RESULT = {"SetOption82":"OFF"}
```

**"Alexa, Kitchen Warm White"**
`12:51:45 RSL: RESULT = {"POWER":"ON","Dimmer":100,"Color":"00000056A9","HSBColor":"227,100,0","Channel":[0,0,0,34,66],"CT":383}
`

**"Alexa, Kitchen Cool White"**
`12:51:51 RSL: RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000DD22","HSBColor":"227,100,0","Channel":[0,0,0,87,13],"CT":199}
`

```
12:52:05 CMD: SetOption82 1
12:52:05 RSL: RESULT = {"SetOption82":"ON"}
```
**"Alexa, Kitchen Warm White"**
`12:52:11 RSL: RESULT = {"POWER":"ON","Dimmer":100,"Color":"00000056A9","HSBColor":"227,100,0","Channel":[0,0,0,34,66],"CT":383}
`
**"Alexa, Kitchen Cool White"**
`12:52:31 RSL: RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000DD22","HSBColor":"227,100,0","Channel":[0,0,0,87,13],"CT":199}`

Bidulglloq, 2020-02-26:
FYI, the bulbs were updated to the latest OTA firmware available
```
Program Version: 8.0.1(tasmota)
Build Date & Time: 2019-12-25T12:33:25
Core/SDK: 2_6_1/2.2.2-dev(38a443e)
```

ascillato, 2020-02-26:
Latest Tasmota is 8.1.0.9

Please, update to latest from dev branch at http://thehackbox.org/tasmota/tasmota.bin

Bidulglloq, 2020-02-26:
Super .... I Updated to the latest dev version

```
Program Version: 8.1.0.9(ad357ea-tasmota)
Build Date & Time: 2020-02-25T21:00:11
Core/SDK Version: 2_6_1/2.2.2-dev(38a443e)
```

Now I get:
```
13:47:05 CMD: setOption82
13:47:05 RSL: RESULT = {"SetOption82":"ON"}
```

"Alexa, Warm White"
```
13:47:14 RSL: RESULT = {"POWER":"ON","Dimmer":100,"Color":"00000000FF","HSBColor":"0,0,0","White":100,"CT":383,"Channel":[0,0,0,0,100]}
```

"Alexa, Cool White"
```
13:47:30 RSL: RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000FF00","HSBColor":"0,0,0","White":100,"CT":199,"Channel":[0,0,0,100,0]}
```

The CT values are still the same (199..383) but the channels for wW and cW are now ranging for 0 to 100%

Thanks to @s-hadinger and @ascillato for you awesome support !!!

closed 2020-02-26 by Bidulglloq