arendst/Tasmota #7799: 2 counters not showing in domoticz IDX

RdBakker, 2020-02-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I made a setup with wemos D1 with 2 reed-contacts for gas and watermonitoring. 

setting up D6 GPIO12 Relay1 as counter1
setting up D7 GPIO13 Led1i as counter2

counters work and count up,  they also write the mqtt message: 

11:27:42 MQT: tele/tasmota/watermeter/SENSOR = {"Time":"2020-02-26T11:27:42","COUNTER":{"C1":26,"C2":25}}

so the configuration works. 

as further, the settings in domoticz template:

Sensor idx 6 Count/PM1 is set to the right idx value and counts in domoticz (counter1)
But idx 7 isn't. (counter2) is also set to it's corresponding idx, but is not working

Sensor idx 7 Voltage/PM2.5 stays empty  I noticed that this is not a counter, but cannot be changed.

i'm working with the latest version 8.1.0
--

--

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-02-26:
Default only one sensor is supported. If you want to see the second counter you will need to use rules. 

See rule cookbook in the documentation how simple it can be ;-)

ascillato2, 2020-02-26:
Closing this issue as it has been answered. If you need further assistant, please address this to the Tasmota support chat. Thanks. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-26 by ascillato2