arendst/Tasmota #7801: Access to RtcReboot.fast_reboot_count from Rules

NiekertDev, 2020-02-26:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
I would like to be able to access the RtcReboot.fast_reboot_count variable as a Rule variable. That way you are able to make lights like the [Philips Sceneswitch](https://www.usa.lighting.philips.com/consumer/choose-a-bulb/sceneswitch) bulbs.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Writing to Mem every time you boot, but that is not very good for the flash / this variable already exists.

**Additional context**  
_Add any other context or screenshots about the feature request here._
RtcReboot.fast_reboot_count is a C++ variable containing the times a device has been restarted (turned on/off) between x seconds.

Correct me if I'm wrong, this is at least what I understand of the code.

PS. I'm Dutch sorry for this English

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-26:
I understand your request, but if you are cutting the power of which device, you are not able to control it when your switch is off.

Which device you have with Tasmota? If it is a sonoff basic or a shelly, it is better to re-wire them as the manufacturer recommends.

NiekertDev, 2020-02-26:
I have a [LSC Smart Connect GU10 bulb](https://templates.blakadder.com/lsc_smart_connect_MR16.html) (Tuya). The strange thing is that when turning the bulb on and off in a short time, I get a boot message in the console saying 'Some settings have been reset'. This message is only coded in Tasmota.ino line 259.

ascillato, 2020-02-26:
> Some settings have been reset

See in the docs `SetOption36`, `SetOption65` and https://tasmota.github.io/docs/#/Fast-power-cycle-device-recovery

If you leave them as default and you turn your bulb ON and OFF, 4 times in a row and the time that is ON is less than 10 seconds, your bulb will reset to default.

Jason2866, 2020-02-26:
@ascillato  i think reverting to defaults is not the issue. 
That is the example, there is a place to store a value that survives a power off.
What @NiekertDev  wants is to decide of the time how long was switched off to do a action via a rule

NiekertDev, 2020-02-26:
Yes, I know, but in issue [6695 andrethomas](https://github.com/arendst/Tasmota/issues/6695#issuecomment-544181238)  stated that the RTC memory is being reset after a power cycle, this is not happening (it was a example).

NiekertDev, 2020-02-26:
> What @NiekertDev wants is to decide based on the time how long was switched off to do a action via a rule the next time the device (bulb) is switched on

Not the time, more the amount of times the bulb is being switched on / off. This is being tracked in RtcReboot.fast_reboot_count. If I were able to access this variable in a rule, you can imitate a Philips Sceneswitch bulb.

NiekertDev, 2020-02-26:
Did some work this evening, my fork has this function implement now. Will test it and add functionallity if needed. See something wrong? Make a PR

ascillato2, 2020-03-03:
Great to know that now it is working for you.

If you want, please do a PR. 

Thanks.

closed 2020-03-03 by ascillato2