arendst/Tasmota #7808: Updates of Shutterposition in json format breaks compatibility with iobroker

he-so, 2020-02-27:
### PROBLEM DESCRIPTION

In 7.0.3 the shutter position was reported via MQTT as value Shutterposition1. This way it was possible to read and update the value via the sonoff mqtt adapter in iobroker.

Now this feature was broken, since the shutter position, direction and target values are json encoded. In iobroker this maps to an object like sonoff.0.DVES_917915.Shutter1_Position=0
When updating this objects value to open the shutter, the adapter send the folling mqtt command:
Send to "DVES_917915": cmnd/tasmota/Shutter1_Target = 100
But tasmota can not process this:
[DVES_917915] Received: stat/tasmota/RESULT = {"Command":"Unknown"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): Jinvoo Shutter Controller
- [ ] Tasmota binary firmware version number used: latest DEV
  - [ ] Pre-compiled
  - [x ] Self-compiled
    - [x ] IDE / Compiler used: Visual Studio Code / Platformio
- [x ] Flashing tools used: platformio
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
23:33:08 CMD: Backlog Template; Module; GPIO 255
23:33:08 MQT: stat/tasmota/RESULT = {"NAME":"TuyaTouchShutt","GPIO":[52,0,0,19,23,18,0,0,17,21,0,22,0],"FLAG":0,"BASE":18}
23:33:08 MQT: stat/tasmota/RESULT = {"Module":{"0":"TuyaTouchShutt"}}
23:33:08 MQT: stat/tasmota/RESULT = {"GPIO0":{"52":"Led1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"19":"Button3"},"GPIO4":{"23":"Relay3"},"GPIO5":{"18":"Button2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"17":"Button1"},"GPIO13":{"21":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"22":"Relay2"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```

23:33:40 MQT: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2","Tasmota3"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:33:40 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:40:14","StartupUTC":"2020-02-27T21:53:26","Sleep":50,"CfgHolder":4617,"BootCount":33,"BCResetTime":"2020-02-27T22:53:34","SaveCount":769,"SaveAddress":"F5000"}}
23:33:40 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(tasmota)","BuildDateTime":"2020-02-27T22:52:32","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"351/699"}}
23:33:40 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!Box 7490",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000E009","2805C8000100060000005A00000000000000","40000000","00000000"]}}
23:33:40 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":573,"Free":428,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35","Sensors":"1,2,3,4,5,6"}}
23:33:40 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6421","IPAddress":"192.168.4.43","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"BC:DD:C2:91:79:15","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
23:33:40 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.4.37","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_917915","MqttUser":"DVES_USER","MqttCount":4,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
23:33:40 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"2020-02-27T22:33:40","Local":"2020-02-27T23:33:40","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:37","Sunset":"18:29"}}
23:33:40 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2020-02-27T23:33:40","Shutter1":{"Position":0,"Direction":0,"Target":0}}}
23:33:40 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2020-02-27T23:33:40","Uptime":"0T00:40:14","UptimeSec":2414,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!Box 7490","BSSId":"34:81:C4:36:2D:79","Channel":11,"RSSI":80,"Signal":-60,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
23:34:48 MQT: Received Topic "cmnd/tasmota/Shutter1_Target", Data Size 3, Data "100"
23:34:48 SRC: MQTT
23:34:48 CMD: Group 0, Index 1, Command "SHUTTER1_TARGET", Data "100"
23:34:48 MQT: stat/tasmota/RESULT = {"Command":"Unknown"}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Change the object value

### EXPECTED BEHAVIOUR

The updated value should be processed, so the shutter can be moven via mqtt. The working command is "ShutterPosition1 100", but I see the naming problem. 
What about using the dot in between like  Shutter1.Position=100? The tasmota command interpreter seems to split these.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-02-27:
Hi,

That change should be addressed to the iobroker adapter repository. The json encoding for shutters gives more information than before. 

The iobroker adapter needs to update to support shutters. Please, address there and link here the issue. 

Thanks

stefanbode, 2020-02-28:
The command to update the position and also to get the value back is SHUTTERPOSITION. Because we do not want to break stuff we try to stay with the current approach. I do not see a big value if it is target or position. The JSON was a wish of many users to get more information about the current state.
The command: cmnd/tasmota/Shutter1_Target was never implemented and could not work with the default code.
The new splitting at the "." will break all other functionality because in TASMOTA the command ends at the first number. This number is the index. Everything behind this is the value. Therefore this idea might be great but impossible to implement.

Shutter1.Position 100  -->
17:06:42 CMD: Group 0, Index **1**, Command "**SHUTTER**", Data "**.Position 100**"

closed 2020-02-28 by ascillato2