arendst/Tasmota #7809: Disconecting from mqtt server

delta010, 2020-02-28:
### PROBLEM DESCRIPTION
my sonoff tasmota devices randomly disconnecting from the MQTT server with an error code Connect failed to 192.168.1.3:1883, rc -2. Retry in 10 sec i am using the Program Version 7.1.2(tasmota) and  Core/SDK Version 2_6_1/2.2.2-dev(38a443e) .But when I use the core version 2.3.0 there is no disconnect problem but there is no 2.3.0 core version For OTA for new tasmota version 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic):  Sonoff Basic
- [ X] Tasmota binary firmware version number used:  7.1.2(tasmota)
  - [X ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X ] Flashing tools used:  esptool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:

05:20:44 MQT: stat/sonofflamp/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"sonofflamp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
05:20:44 MQT: stat/sonofflamp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T15:12:23","StartupUTC":"2020-02-27T13:08:21","Sleep":50,"CfgHolder":4617,"BootCount":107,"SaveCount":681,"SaveAddress":"FB000"}}
05:20:44 MQT: stat/sonofflamp/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
05:20:44 MQT: stat/sonofflamp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Keralavision_WiFi73",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005A00000000000000","00000000","00000000"]}}
05:20:44 MQT: stat/sonofflamp/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
05:20:44 MQT: stat/sonofflamp/STATUS5 = {"StatusNET":{"Hostname":"sonofflamp-6546","IPAddress":"192.168.1.4","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:18:59:92","Webserver":2,"WifiConfig":4}}
05:20:44 MQT: stat/sonofflamp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.3","MqttPort":1883,"MqttClientMask":"sonofflamp","MqttClient":"sonofflamp","MqttUser":"hassio","MqttCount":4,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
05:20:44 MQT: stat/sonofflamp/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 28 04:20:44 2020","Local":"Fri Feb 28 05:20:44 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"07:35","Sunset":"18:30"}}
05:20:44 MQT: stat/sonofflamp/STATUS10 = {"StatusSNS":{"Time":"2020-02-28T05:20:44"}}
05:20:44 MQT: stat/sonofflamp/STATUS11 = {"StatusSTS":{"Time":"2020-02-28T05:20:44","Uptime":"0T15:12:23","UptimeSec":54743,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Keralavision_WiFi73","BSSId":"14:A7:2B:11:72:47","Channel":10,"RSSI":88,"LinkCount":1,"Downtime":"0T00:32:39"}}}
05:21:24 MQT: tele/sonofflamp/STATE = {"Time":"2020-02-28T05:21:24","Uptime":"0T15:13:03","UptimeSec":54783,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Keralavision_WiFi73","BSSId":"14:A7:2B:11:72:47","Channel":10,"RSSI":100,"LinkCount":1,"Downtime":"0T00:32:39"}}

```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-02-28:
There where updates between core 2.3.0 and the now used core 2.6.1 ;-) Tasmota mqtt does work. 

What mqtt broker do you use? Take a look in mqtt log.
`rc -2` : MQTT_CONNECT_FAILED - the network connection failed

Have you tried to compile with` MQTT_CLEAN_SESSION   0` 
https://github.com/arendst/Tasmota/blob/development/tasmota/my_user_config.h#L345

It is the broker setting you have set why Tasmota can not connect.
Tasmota does connect with all major used mqtt brokers configured with standard settings

delta010, 2020-02-28:
I am using mosquito MQTT broker, is the is any way tho change the keepalive value. the keep-alive value is k15 in my case

Jason2866, 2020-02-28:
I use Mosquitto too. Mosquitto does work fine, without changing anything in the standard config. 
There is no need to change the keepalive value using Tasmota.
You can try to change here https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_post.h#L710

It should work to set the define in `my_userconfig_override.h`

Jason2866, 2020-02-28:
What version of mosquito are you using? Maybe it is just to old.
See https://github.com/arendst/Tasmota/blob/development/lib/PubSubClient-EspEasy-2.7.12/src/PubSubClient.h

anrada73, 2020-02-28:
i have same.

1582899220: New client connected from 192.168.0.80 as ks (p2, c1, k30, u'ihome').
1582899281: Client ks has exceeded timeout, disconnecting.
1582899282: New connection from 192.168.0.80 on port 1883.
1582899282: New client connected from 192.168.0.80 as ks (p2, c1, k30, u'ihome').
1582899297: Socket error on client ks, disconnecting.
1582899309: New connection from 192.168.0.80 on port 1883.
1582899309: New client connected from 192.168.0.80 as ks (p2, c1, k30, u'ihome').
1582899354: Client ks has exceeded timeout, disconnecting.
1582899358: New connection from 192.168.0.80 on port 1883.
1582899358: New client connected from 192.168.0.80 as ks (p2, c1, k30, u'ihome').
1582899413: Client ks has exceeded timeout, disconnecting.
1582899421: New connection from 192.168.0.80 on port 1883.
1582899421: New client connected from 192.168.0.80 as ks (p2, c1, k30, u'ihome').
1582899548: Client ks has exceeded timeout, disconnecting.
1582899550: New connection from 192.168.0.80 on port 1883.
[INFO] found ihome on local database
1582899551: New client connected from 192.168.0.80 as ks (p2, c1, k30, u'ihome').
1582899621: New connection from 192.168.0.80 on port 1883.
1582899621: Client ks already connected, closing old connection.
1582899621: New client connected from 192.168.0.80 as ks (p2, c1, k30,

ascillato, 2020-02-28:
Please, update to latest Tasmota (v 8.1.0.9) from http://thehackbox.org/tasmota/tasmota.bin

Thanks

delta010, 2020-02-28:
I update sonoff basic from version 7.1.2 to 8.1.0 then it shows MINIMAL firmware
please upgrade

status 0 :
   stat/sonofflamp/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"sonofflamp","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
16:58:18 MQT: stat/sonofflamp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Blocked Loop","Uptime":"0T00:17:26","StartupUTC":"2020-02-28T15:40:52","Sleep":50,"CfgHolder":4617,"BootCount":109,"SaveCount":725,"SaveAddress":"FB000"}}
16:58:18 MQT: stat/sonofflamp/STATUS2 = {"StatusFWR":{"Version":"8.1.0(minimal)","BuildDateTime":"2019-12-25T12:34:47","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"322/699"}}
16:58:18 MQT: stat/sonofflamp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Keralavision_WiFi73",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005A00000000000000","00000000","00000000"]}}
16:58:19 MQT: stat/sonofflamp/STATUS4 = {"StatusMEM":{"ProgramSize":372,"Free":628,"Heap":29,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000809","80082181","040003A2","20021702","010000C0","00000000"],"Drivers":"1,2","Sensors":""}}
16:58:19 MQT: stat/sonofflamp/STATUS5 = {"StatusNET":{"Hostname":"sonofflamp-6546","IPAddress":"192.168.1.4","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"80:7D:3A:18:59:92","Webserver":2,"WifiConfig":4}}
16:58:19 MQT: stat/sonofflamp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.3","MqttPort":1883,"MqttClientMask":"sonofflamp","MqttClient":"sonofflamp","MqttUser":"hassio","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:58:19 MQT: stat/sonofflamp/STATUS7 = {"StatusTIM":{"UTC":"Fri Feb 28 15:58:19 2020","Local":"Fri Feb 28 16:58:19 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00"}}
16:58:19 MQT: stat/sonofflamp/STATUS10 = {"StatusSNS":{"Time":"2020-02-28T16:58:19"}}
16:58:19 MQT: stat/sonofflamp/STATUS11 = `{"StatusSTS":{"Time":"2020-02-28T16:58:19","Uptime":"0T00:17:27","UptimeSec":1047,"Vcc":3.463,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Keralavision_WiFi73","BSSId":"14:A7:2B:11:72:47","Channel":10,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:08"}}}`
          
[image]

ascillato, 2020-02-28:
Please update to latest with the provided link to 8.1.0.9

delta010, 2020-02-28:
I update using the  link  http://thehackbox.org/tasmota/tasmota.bin from version 7.1.2 via OTA.then that happends

ascillato, 2020-02-28:
That link is Tasmota 8.1.0.9 and full, your actual version is minimal 8.1.0

Please, download the version of the link and upload it using the upload from file version

Jason2866, 2020-02-28:
@anrada73  your problem is different. You have two devices with the same topic.
This is a configuration error. Every device needs a unique topic. 
Look in Tasmota wiki for more information.

anrada73, 2020-02-28:
> @anrada73 your problem is different. You have two devices with the same topic.
> This is a configuration error. Every device needs a unique topic.
> Look in Tasmota wiki for more information.

it's impossible.

ascillato, 2020-02-28:
@anrada73 please update your devices to latest Tasmota (8.1.0.9) and do `reset 5` on the console of those devices.

delta010, 2020-03-01:
how can I solve  Client sonofffan has exceeded timeout, disconnecting  problem

ascillato2, 2020-03-01:
That means that you have 2 or more devices with the same mqtt client name or you have wifi issues.

delta010, 2020-03-01:
No all my devices have different ClientID

TD-er, 2020-03-02:
> how can I solve Client sonofffan has exceeded timeout, disconnecting problem

What is the timeout configuration setting in your broker?
The default for Mosquitto is 10 seconds, so after that timeout has been exceeded by more than 50% (thus 15 sec), it will consider the client to be disconnected and post the LWT etc.

Does that node have issues with the WiFi connection? 
For example, is the connection to the MQTT broker much more stable when you keep sending ping packets to the ESP node?

delta010, 2020-03-02:
my sonoff have keepalive of  k30, there are no noticeable  issues while connecting to wifi, probably the  timeout and  disconnecting problem is caused by the router because there are about 20 wifi devices connected to the router

Jason2866, 2020-03-02:
@delta010  i think you nailed the problem. Many consumer router have problems if there
are more wifi devices connected.

closed 2020-03-05 by ascillato2