arendst/Tasmota #7810: Pulse timed relay stays on if polled within pulsetime

radionerd, 2020-02-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When a momentary relay is configured with pulsetime the relay stays on forever if polled with 'cm?cmnd=Power' while the relay is active.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [✓] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [✓] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [✓] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [✓] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [✓] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [✓] Device used (e.g., Sonoff Basic): nodemcu
- [✓] Tasmota binary firmware version number used:8.1.0.10
  - [✓] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [✓] Flashing tools used: OTA
- [✓] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

RESULT = {"Module":{"18":"Generic"}}
RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"4":"DS18x20"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"52":"Led1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"7":"WS2812"}}

```
- [n/a] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [✓] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
00:48:57 RSL: STATUS = {"Status":{"Module":18,"FriendlyName":["NodeMCU","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:48:57 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:33:08","StartupUTC":"2020-02-28T11:15:49","Sleep":50,"CfgHolder":4617,"BootCount":98,"BCResetTime":"2020-02-29T00:16:02","SaveCount":309,"SaveAddress":"F9000"}}
00:48:57 RSL: STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(9c8dd21-tasmota)","BuildDateTime":"2020-02-28T11:00:11","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"330/699"}}
00:48:57 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Wifee'sWifi",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00800008","00000000"]}}
00:48:57 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":572,"Free":428,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","000000CD","010013C0","C000F981","00000000"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35","Sensors":"1,2,3,4,5,6"}}
00:48:57 RSL: STATUS5 = {"StatusNET":{"Hostname":"nodemcu","IPAddress":"192.168.1.41","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"3C:71:BF:29:A7:A5","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
00:48:57 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-02-28T11:48:57","Local":"2020-02-29T00:48:57","StartDST":"2020-09-27T02:00:00","EndDST":"2020-04-05T02:00:00","Timezone":99,"Sunrise":"07:06","Sunset":"20:15"}}
00:48:57 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-02-29T00:48:57"}}
00:48:57 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-02-29T00:48:57","Uptime":"0T00:33:08","UptimeSec":1988,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","Dimmer":100,"Color":"FF0004","HSBColor":"359,100,100","Channel":[100,0,2],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Wifee'sWifi","BSSId":"06:02:6F:54:A8:0B","Channel":5,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:11"}}}
```
- [✓] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:51:05 SRC: WebCommand from 192.168.1.20
00:51:05 CMD: Group 0, Index 1, Command "POWER", Data "ON"
00:51:05 RSL: RESULT = {"POWER1":"ON"}
00:51:05 RSL: POWER1 = ON
00:51:05 RSL: STATE = {"Time":"2020-02-29T00:51:05","Uptime":"0T00:35:16","UptimeSec":2116,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","Dimmer":100,"Color":"FF0004","HSBColor":"359,100,100","Channel":[100,0,2],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Wifee'sWifi","BSSId":"06:02:6F:54:A8:0B","Channel":5,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:11"}}
00:51:06 HTP: Command
00:51:06 SRC: WebCommand from 192.168.1.20
00:51:06 CMD: Group 0, Index 1, Command "POWER", Data ""
00:51:06 RSL: RESULT = {"POWER1":"ON"}
00:51:06 RSL: POWER1 = ON
*** RELAY should have turned off after 2 seconds with pulsetime 20 but it stayed on indefinitely ***
00:51:21 WIF: Checking connection...
00:51:21 CMD: pulsetime
00:51:21 SRC: WebConsole from 192.168.1.20
00:51:21 CMD: Group 0, Index 1, Command "PULSETIME", Data ""
00:51:21 RSL: RESULT = {"PulseTime1":{"Set":20,"Remaining":0},"PulseTime2":{"Set":0,"Remaining":0},"PulseTime3":{"Set":0,"Remaining":0},"PulseTime4":{"Set":0,"Remaining":0},"PulseTime5":{"Set":0,"Remaining":0},"PulseTime6":{"Set":0,"Remaining":0},"PulseTime7":{"Set":0,"Remaining":0},"PulseTime8":{"Set":0,"Remaining":0}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
On the console set pulsetime 20 then run the script below using your own url in place of nodemcu.fritz.box :
```
# relay turns on forever with tasmota configured to pulsetime 20 if polled while the relay is active
curl http://nodemcu.fritz.box/cm?cmnd=POWER1%20ON
sleep 1 # change this to sleep 3 and the relay turns off as it should - go figure :-)
echo
curl http://nodemcu.fritz.box/cm?cmnd=POWER1
echo
```
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
With pulsetime set to 20 the relay should turn on for 2 seconds when activated and then release regardless of whether polled for its status by web command

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Also same result on nodemcu using 8.1.0 release (rather than dev build)
Also tested on Arlec PC189HA (similar to Sonoff S26) with same result on 8.1.0 release

Thanks for reading this far. Let me know if you need help.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-02-28:
As designed. A `Power` command resets `pulsetime`
To query the remaining time use `pulsetime` without a payload

radionerd, 2020-02-28:
Many thanks for your speedy response. Sounds more like a bug than an intended feature to me, as the wiki states that the power command _restarts_ the pulsetime<x>.
If the pulsetime is intentionally disabled with no power parameter then the wiki needs an additional line to specify the behaviour when no parameter is given. i.e. None - stop pulsetimer, 0-5 re-start pulse timer to pulsetime<x>.
Is there a way to poll the status of the relay without sometimes disabling pulsetimer depending on whether the relay happened to be active or not?

Jason2866, 2020-02-28:
You get the status with `status 11` without side effects

closed 2020-02-28 by ascillato2