arendst/Tasmota #7812: Sonoff Tasmota - losing IP address

BoroBoy2k5, 2020-02-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

BoroBoy2k5, 2020-02-28:
This is the first time I am using this forum, I was not able edit the other in order to provide more details, hence I am adding the details.

for our Home Heating system we are using Sonoff version 2 with DHT11 temperature sensors. We made 13 of these at beginning January. 2 of these have lost there IP Address, e.g. cannot be seen on the network, although the tasmota hotspot can be triggered.

The other issue we are having with some the Tasmota SonOff thermostats is that some of these start report null temperature and humidity, resetting the Tasmota via console has no effect, only way to sort this issue when it occurs is to either remove the 240 Volt power for about 1-2 minutes (Although 240 Volt on/off power occasionally sorts the problem out), the other way is to remove the 3.3 Volt power to the DHT11 temperature sensor.

Has anyone else encountered the same, if so how did they resolve it.

is it possible to reconfigure the IP address via the hotspot server, the only method we could find us via serial port interface to Laptop.

meingraham, 2020-02-28:
Not sure why you are not able to add the information to the issue template. I have not heard of anybody else not being able to do that... Not doing, yes. Not being able to. Never had that happen. We really can't help without that information. Your description only partially illuminates your situation. In any case, this is best handled on Discord.

Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-28 by meingraham