arendst/Tasmota #7821: MQTT user and pass on Initial configuration

nedimhx, 2020-02-29:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Hello,
Is it possible to Allow on initial configuration, on first setup, where we enter Wifi name and pass, also to have option to enter Mqtt user and pass, host, port, topics are hardcoded and in some case don't need to be changeable but username and password always need.
Can someone help with this ?
Regards

Jason2866, 2020-02-29:
You can got to webconsole and do all the settings you need.

ascillato2, 2020-02-29:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-02-29 by ascillato2

nedimhx, 2020-02-29:
@Jason2866 for end users its easyer to fill the form and click submit.
Have you any suggestion how to achieve that ?

meingraham, 2020-02-29:
@nedimhx 

"for end users its easyer to fill the form" - but still you didn't. The issue template we provided was left mostly incomplete by you.

Support for Tasmota is provided via the Discord chat. Posting an issue to the Tasmota GitHub is for _**verified**_ bugs (verified through the dialog on Discord) and for feature requests. Support for flashing, configuration, troubleshooting, etc. is only available via Discord.