arendst/Tasmota #7823: White command should not change RGB values

Simonmicro, 2020-02-29:
**Have you looked for this feature in other issues and in the docs?** 
Yes

**Is your feature request related to a problem? Please describe.**  
Yes - it is. I connected a RGBW strip to the controlling chip. When I use the `Color` command the white channel can be omitted by appending an `=` (see #6872 and someone should update the docs). So now I can control the RGB color AFTER setting the white value - because there is no command to only change the white cannel, without resetting the RGB part.

**Describe the solution you'd like**  
The white command should ONLY modify the fourth and / or fifth channel.

**Describe alternatives you've considered**  
* living with it
* Pulling the color, adding the new white and sending a full `Color` command (not possible with the apps I'm using)
* `SetOption37 128` ~ same behavior -> white resets the RGB part

**Additional context** 
-

Jason2866, 2020-02-29:
You are invited to change the doc 😀

Simonmicro, 2020-02-29:
@Jason2866 Docs update merge request created. But this does not fix or help me. Only other lost soles, digging in the documentation about this feature...

blakadder, 2020-03-01:
https://tasmota.github.io/docs/#/Lights?id=rgb-and-white-split

Simonmicro, 2020-03-01:
Yes. Here some test commands I used in that order to determine the results below...

```
Color #ffffffff
stat/tasmota/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100]}
```
All on
```
White 100
stat/tasmota/RESULT = {"White":"FFFFFFFF"}
```
RGB off, white on
```
White 0
stat/tasmota/RESULT = {"POWER":"OFF","Dimmer":0,"Color":"0,0,0,0","HSBColor":"0,0,0","Channel":[0,0,0,0]}
```
All off
```
White 100
stat/tasmota/RESULT = {"White":"0,0,0,0"}
```
RGB off, white on
```
SetOption37 128
stat/tasmota/RESULT = {"SetOption37":128}
```
Reboot (as expected)...
```
Color #ffffffff
stat/tasmota/RESULT = {"POWER1":"ON","Dimmer1":100,"POWER2":"ON","Dimmer2":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100]}
```
All on
```
White 0
stat/tasmota/RESULT = {"POWER1":"OFF","Dimmer1":0,"POWER2":"OFF","Dimmer2":0,"Color":"0,0,0,0","HSBColor":"0,0,0","Channel":[0,0,0,0]}
```
**ALL OFF**
```
White 255
stat/tasmota/RESULT = {"White":"0,0,0,0"}
```
RGB off, white on
```
Color #ffffffff
stat/tasmota/RESULT = {"POWER1":"ON","Dimmer1":100,"POWER2":"ON","Dimmer2":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100]}
```
All on
```
Color #000000=
stat/tasmota/RESULT = {"POWER1":"OFF","Dimmer1":0,"POWER2":"OFF","Dimmer2":0,"Color":"00000000","HSBColor":"0,0,0","Channel":[0,0,0,0]}
```
RGB off, white on (note that I only passed the RGB part, if I don't pass the `=` the white goes also off)
```
Color1 #ffffffff
stat/tasmota/RESULT = {"POWER1":"ON","Dimmer1":100,"POWER2":"ON","Dimmer2":100,"Color":"FFFFFFFF","HSBColor":"0,0,100","Channel":[100,100,100,100]} 
```
**ALL ON** (the white shouldn't change)
```
Color2 #000000ff
stat/tasmota/RESULT = {"POWER1":"OFF","Dimmer1":0,"POWER2":"ON","Dimmer2":100,"Color":"000000FF","HSBColor":"0,0,0","Channel":[0,0,0,100]}
```
**RGB OFF**, white on (the rgb should be its own light and not change)

closed 2020-04-28 by stale[bot]

sep-don-barrett, 2020-06-01:
Note that MQTT autodiscovered RGBW lights with Home Assistant use the COLOR command with three arguments to set the color palette, and the WHITE command to set the W component.  As a result of this long-standing issue, any adjustment to the light palette's white value slider in Home Assistant zeroes the RGB LED outputs.  Now one or the other or both projects could match the impedances where this communication takes place and fix this -- and I imagine a great many Tasmota lights are used in Home Assistant.  But it seems simpler to simply have WHITE not exercise side effects on the RGB PWMs -- side effects are almost always something that violates the principle of least surprise in software engineering.