arendst/Tasmota #7824: ESW15-US help

temocastro, 2020-02-29:
I load the tasmota, tried the esw1 template but not exactly controlling the relay, had anybody tried?

closed 2020-02-29 by close-issue-app[bot]

Shumphead, 2020-07-11:
@temocastro do u have a guide I can follow for flashing the ESW15?  I followed the information here https://tasmota.github.io/docs/Getting-Started/ and was able to get tasmota to flash to the device and it persist when I power cycle the 3.3v connection. However, when I plug it in to the wall it loads the stock firmware again... Do u have any idea what I am missing?

temocastro, 2020-07-12:
Did you make the jumper to D0 to gnd? I believe I had to do that in order
to let it pass the programming

Shumphead, 2020-07-12:
Thanks for the quick reply. Yeah I split the gnd wire to D0.  Tasmota is persisting on the plug. But it appears to be on a different partition that is booted only when it's in flash mode.  When I plug it into the wall it still has the stock firmware. Do you recall needing to do anything different from a standard flash? Tomorrow I will dig into the partitions and see if I need to flash it to one of the others.

temocastro, 2020-07-12:
I will program another tomorrow and will take pictures and notes

NeoMatrixJR, 2020-10-14:
Did this happen?  I'm also curious how to flash?  Can you use Tuya Convert and flash OTA or do you have to open it up?

Shumphead, 2020-10-14:
Unfortunately, you need to open the device up. It's pretty easy. my problems went away when I tried it again. I still need to flash another plug and can do a mini writeup.