arendst/Tasmota #7827: homebridge-hue support doesn't seem to work after updating to 8.1.0

suhajda3, 2020-02-29:
### PROBLEM DESCRIPTION
I have updated all my Sonoff-Tasmota devices (Sonoff Basic Module) from version 6.7.1 to 8.1.0 (first updating to 7.2.0 and then to 8.1.0). Before the update all my Sonoff devices worked great with [homebridge-hue](https://github.com/ebaauw/homebridge-hue) but after the update they stopped working.

Even when downgrading a device to 6.7.1 (and resetting the configuration) it still doesn't seem to work. It works with Alexa without a problem. You can find more information on [this ](https://github.com/ebaauw/homebridge-hue/issues/640)issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Enable Hue emulation on Tasmota 8.1.0 and try to retrieve paths with homebridge-hue ph command:
```
# ph -H 192.168.99.12 get /
[
  {
    "success": {
      "username": "cc6368"
    }
  }
]
```

### EXPECTED BEHAVIOUR
The paths should appear.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2020-03-01:
Hi, please complete the template in order to have more debug information so as to properly help you. Thanks

suhajda3, 2020-03-01:
I have updated the template. Thanks

Jason2866, 2020-03-01:
Tasmota does support only Alexa hardware. The fact that homebridge-hue worked before was luck.
It seems that the setup with 6.7.1. was not stable too because reverting back to this version
does not bring back the functionality.
Sorry no issue, since it was never supported.
You are welcome to submit a PR to support homebridge-hue since it seems it needs just
a few adaptions to get it running.
The main contributors will NOT do since no one uses homebridge-hue.

ebaauw, 2020-03-01:
> The fact that homebridge-hue worked before was luck.

Not quite, see https://github.com/ebaauw/homebridge-hue/issues/512.

The issue seems to be that with the newer firmware, the request to get the full state (GET /api/username) seems to be broken; the response looks like the response to a request to create a user (POST /api).  See https://github.com/ebaauw/homebridge-hue/issues/640.

Jason2866, 2020-03-01:
So if i get your statement correct, time to close this issue.

suhajda3, 2020-03-01:
> So if i get your statement correct, time to close this issue.

Could we wait until the issue is actually resolved? ;-)

(If the problem is on homebridge-hue side I will close this issue, but if it's on Tasmota side it would be great to excalate it.)

Jason2866, 2020-03-01:
Okay leave it open for reference

suhajda3, 2020-03-01:
@ebaauw confirmed that this problem is on the homebridge-hue side so I will close this issue now.

@Jason2866 could you please delete [this](https://github.com/arendst/Tasmota/issues/7826) issue. It is a duplicate of this one. Thanks

closed 2020-03-01 by suhajda3