arendst/Tasmota #7830: ShutterStop1 toggles Relay1 instead of stopping both relays

borisneubert, 2020-03-01:
### PROBLEM DESCRIPTION

I am trying to configure a [LoraTap Curtain Switch](https://www.loratap.com/tuya-smart-life-wifi-curtain-switch-for-electric-motorized-curtain-blind-roller-shutter-google-home-amazon-alexa-voice-control-p0014.html) (not yet available in the database).

Firmware is 8.1.0.10(lite), STAGE/2.2.2-dev(38a443e) (Tasmota development branch as of 2020-02-29), with SHUTTER turned on and DOMOTICZ, HOME_ASSISTANT, KNX turned off.

Configuration as follows:
```
# GPIO3  Button3 Pause
# GPIO5  Button1 Open
# GPIO12 Button2 Close
# GPIO13 Relay2 Close
# GPIO15 Relay1 Open
 
# ShutterMode 0  First relay: OFF/DOWN, Second relay: OFF/UP
SetOption80 1
ButtonTopic 0
Backlog PulseTime1 105; PulseTime2 105
Backlog Interlock 1,2; Interlock ON
Rule1 on Button3#state do ShutterStop1 endon
```

Button1 (Open) and Button2 (Close) work, they turn on Relay1 (Open) and Relay2 (Close) repectively with interlock for a few seconds.

In spite of the rule, Button3 toggles Relay1 (Open) instead of turning off both relays.

Executing `ShutterStop1` or `ShutterRelay1` in the console give error:

```
09:45:36 CMD: ShutterStop1
09:45:36 MQT: stat/tasmota_B9B00E/RESULT = {"Command":"Unknown"}
```
 

### REQUESTED INFORMATION

Configuration output here:

Firmware is 8.1.0.10(lite), STAGE/2.2.2-dev(38a443e) (Tasmota development branch as of 2020-02-29)

```
09:50:09 CMD: Backlog Template; Module; GPIO 255
09:50:09 MQT: stat/tasmota_B9B00E/RESULT = {"NAME":"Loratap","GPIO":[0,0,0,19,0,17,0,0,18,22,0,21,0],"FLAG":0,"BASE":18}
09:50:09 MQT: stat/tasmota_B9B00E/RESULT = {"Module":{"0":"Loratap"}}
09:50:10 MQT: stat/tasmota_B9B00E/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"19":"Button3"},"GPIO4":{"0":"None"},"GPIO5":{"17":"Button1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"18":"Button2"},"GPIO13":{"22":"Relay2"},"GPIO14":{"0":"None"},"GPIO15":{"21":"Relay1"},"GPIO16":{"0":"None"}}
```

```
09:50:40 MQT: stat/tasmota_B9B00E/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on Button3#state do ShutterStop1 endon"}
09:50:40 MQT: stat/tasmota_B9B00E/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
09:50:40 MQT: stat/tasmota_B9B00E/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```

```
09:51:36 CMD:   STATUS 0
09:51:36 MQT: stat/tasmota_B9B00E/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_B9B00E","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:17:53","StartupUTC":"2020-03-01T08:33:43","Sleep":50,"CfgHolder":4617,"BootCount":229,"BCResetTime":"2020-02-29T21:29:43","SaveCount":378,"SaveAddress":"FA000"}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS2 = {"StatusFWR":{"Version":"8.1.0.10(lite)","BuildDateTime":"2020.02.29 21:15:36","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"345/699"}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["RS65.58",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C00B","2805C8000100060000005A00000000000000","40000000","00000000"]}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS4 = {"StatusMEM":{"ProgramSize":466,"Free":536,"Heap":25,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000407","8F082383","041083A1","00000006","010001C0","00000000","00000000"],"Drivers":"1,2,3,4,9,10,16,20,21","Sensors":"3"}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS5 = {"StatusNET":{"Hostname":"tasmota_B9B00E-4110","IPAddress":"192.168.65.224","Gateway":"192.168.65.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.65.21","Mac":"98:F4:AB:B9:B0:0E","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS6 = {"StatusMQT":{"MqttHost":"pipad.rs65.neubert-volmar.de","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B9B00E","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS7 = {"StatusTIM":{"UTC":"2020-03-01T08:51:36","Local":"2020-03-01T09:51:36","StartDST":"2020-03-29T02:00:00","EndDST":"2020-10-25T03:00:00","Timezone":"+01:00","Sunrise":"07:31","Sunset":"18:33"}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS10 = {"StatusSNS":{"Time":"2020-03-01T09:51:36"}}
09:51:36 MQT: stat/tasmota_B9B00E/STATUS11 = {"StatusSTS":{"Time":"2020-03-01T09:51:36","Uptime":"0T00:17:53","UptimeSec":1073,"Vcc":3.410,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"RS65.58","BSSId":"44:4E:6D:B3:25:2F","Channel":1,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

### TO REPRODUCE
Configure as above and press middle pause Button3.

### EXPECTED BEHAVIOUR
Turns off both relays

### SCREENSHOTS
n/a

### ADDITIONAL CONTEXT
n/a

ascillato2, 2020-03-01:
Your rule is turned off

"Rule1":"OFF"

Please, enable it by command `Rule1 1`

borisneubert, 2020-03-01:
Thank you @ascillato2 for your reply.

I wonder why the rule was shown as turned off, since executing command

`Rule1 on Button3#state do ShutterStop1 endon`

gives

```
10:24:37 CMD:  Rule1 on Button3#state do ShutterStop1 endon
10:24:37 MQT: stat/tasmota_B9B00E/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":473,"Rules":"on Button3#state do ShutterStop1 endon"}
```

Well, 
with Rule1 disabled, Relay1 is toggled.
with Rule1 enabled, nothing happens if I touch the Pause Button.

Why does executing ShutterStop1 on the console gives

```
10:28:39 CMD: ShutterStop1
10:28:39 MQT: stat/tasmota_B9B00E/RESULT = {"Command":"Unknown"}
```

? 

To cross-check settings, I reassigned the pause button to Switch3. `Status 8` gives ON if my hands are off the pause button and OFF if I touch the pause button. As long as I touch the pause button with GPIO3 assigned to Switch3, Relay1 is on.

meingraham, 2020-03-01:
You need to self-compile to get shutter features. The firmware you are using does not have shutters enabled.

borisneubert, 2020-03-01:
@meingraham Sounds plausible though I did compile the firmware myself with shutters enabled.

I used VSCode with PlatformIO.

platformio_override.ini exists and contains, among others:

```
[platformio]
default_envs =
                tasmota-lite
[common]
platform                  = ${core_active.platform}
platform_packages         = ${core_active.platform_packages}
build_flags               = ${core_active.build_flags}
; *** Use settings from file user_config_override.h
                          -DUSE_CONFIG_OVERRIDE
```

user_config_override.h exists and contains, among others:
```
#define MY_LANGUAGE            de-DE           // German in Germany

#ifndef USE_SHUTTER
#define USE_SHUTTER           // Add Shutter support (+6k code)
#endif

#ifdef USE_DOMOTICZ
#undef USE_DOMOTICZ                              
#endif 

#ifdef USE_HOME_ASSISTANT
#undef USE_HOME_ASSISTANT                              
#endif 

#ifdef USE_KNX
#undef USE_KNX                              
#endif 

#ifdef USE_KNX
#undef USE_KNX                              
#endif 
```

Is there a way to query the currently running firmware about enabled and disabled features?

meingraham, 2020-03-01:
I believe selecting "lite" disables features in tasmota-post including shutters. Just compile tasmota.bin with your custom settings.

borisneubert, 2020-03-01:
Excellent! Thank you @meingraham 

Summary:
- tasmota-lite turns off SHUTTER feature even if you explicitely define USE_SHUTTER in user_config_override.h; thus use tasmota instead of tasmota-lite
- always check whether your rules are activated and use `Rule1 1` or so to activate
- Working settings for this [LaraTap Curtain Switch](https://www.loratap.com/tuya-smart-life-wifi-curtain-switch-for-electric-motorized-curtain-blind-roller-shutter-google-home-amazon-alexa-voice-control-p0014.html)
```

 # GPIO3  Button3 Pause
 # GPIO5  Button1 Open
 # GPIO12 Button2 Close
 # GPIO13 Relay2 Close
 # GPIO15 Relay1 Open
 
 # ShutterMode 0  First relay: OFF/DOWN, Second relay: OFF/UP
 SetOption80 1
 ShutterRelay1
 # interlock relays
 Backlog Interlock 1,2; Interlock ON
 # turn off PulseTimes (just in case they were previously enabled)
 Backlog PulseTime1 0; PulseTime2 0
 # assign UP and DOWN buttons
 Backlog ShutterButton1 1 up 1; ShutterButton1 2 down 1
 # durations: opening takes 5 seconds, closing takes 4 seconds
 Backlog ShutterOpenDuration1 5; ShutterCloseDuration 4
 # enable pause button 
 Rule1 on Button3#state do ShutterStop1 endon
 Rule1 1
```

closed 2020-03-01 by borisneubert

borisneubert, 2020-03-29:
Postscriptum:

Template {"NAME":"Loratap SC400W","GPIO":[0,0,0,19,0,17,0,0,18,22,0,21,0],"FLAG":0,"BASE":18}