arendst/Tasmota #7833: Some rule triggers fire continuously despite no-change

denisfrench, 2020-03-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Some rule triggers (eg Power1#State, and Analog#A0div10) fire strictly only upon state-change. Others (eg DS18B20#Temperature, and Counter#C1) fire continuously, even when there is no change to the trigger value.
The documentation for the above triggers states they should fire only when the value changes. Besides this, the current handling of rule triggers is inconsistent.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Wemos D1 R2
- [X] Tasmota binary firmware version number used: 8.1.0.9(tasmota)
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: Gitpod
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
21:15:37 RSL: RESULT = {"NAME":"proSmart BBoil","GPIO":[17,255,255,255,255,255,0,0,21,255,4,255,157],"FLAG":15,"BASE":18}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
21:18:51 RSL: RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":28,"Rules":"ON Power1#Boot DO Backlog Var1 0; Var2 0; Var3 0; Var4 0 ENDON ON Event#setPoint DO Mem1=%value% ENDON ON Event#setTime DO Backlog Mem2=%value%; Mem3=%value%-60; Mem4=%value%-120; Mem5=%value%-180; Mem6=%value%-240 ENDON ON DS18B20#Temperature>%Mem1% IF (Power#state=1) Power OFF; Event Report ENDIF ENDON ON Event#Report DO Publish losant/%topic%/state {"data":{"hwsTemp":%Var1%,"hwsPower":%Var3%,"hwsRelay":%Var2%,"hwsBoost":%Var4%,"hwsSetPoint":%Mem1%, "hwsSetTime":%Mem2%}} ENDON"}

21:18:51 RSL: RESULT = {"Rule2":"ON","Once":"OFF","StopOnError":"OFF","Free":34,"Rules":"ON Power1#Boot DO IF (Mem1==0) Event setPoint=75 ENDIF ENDON ON Power1#Boot DO IF (Mem2==0) Event setTime=360 ENDIF ENDON ON DS18B20#Temperature DO Var1 %value% ENDON ON Power1#State DO Backlog Var2 %value%; Var4=%Var3%*%value%; Event Report ENDON ON Analog#A0div10=0 DO Backlog Var3 0; Var4 0; Power OFF; Event Report ENDON ON Analog#A0div10=100 DO Backlog Var3 1; Var4=%Var2%*1; Event Report ENDON ON Counter#C1 DO IF (%value%==4200) Var5 0 ELSE Var5=3600/%value% ENDIF ENDON"}
  
21:18:51 RSL: RESULT = {"Rule3":"ON","Once":"OFF","StopOnError":"OFF","Free":61,"Rules":"ON Time#Minute=%Mem5% IF (Power1#state=0) AND DS18B20#Temperature<(%Mem1%-20) Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem4% IF (Power1#state=0) AND DS18B20#Temperature<(%Mem1%-10) Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem3% IF (Power1#state=0) AND DS18B20#Temperature<%Mem1% Power ON; Event Report ENDIF ENDON ON Time#Minute=%Mem2% IF (Power1#state=1) Power OFF; Event Report ENDIF ENDON ON Time#Minute|2 DO Event Report ENDON"}
```
- [x] Provide the output of this command: `Status 0`:
```
21:22:14 RSL: STATUS2 = {"StatusFWR":{"Version":"8.1.0.9(tasmota)","BuildDateTime":"2020-02-26T09:13:59","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"470/699"}}
21:22:14 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WitsEnd","WitsEndN"],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008000","2805C8000100060000005A00000000000000","00001000","00000000"]}}
21:22:14 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":529,"Free":2540,"Heap":30,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","8F8863C3","043603A1","200217C5","010008C0","00007881","00000000"],"Drivers":"1,2,3,4,6,8,9,10,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,34"}}
21:22:14 RSL: STATUS5 = {"StatusNET":{"Hostname":"59292a4a1ba24b0007ac7527-1424","IPAddress":"192.168.1.15","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"A0:20:A6:01:C5:90","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
21:22:14 RSL: STATUS7 = {"StatusTIM":{"UTC":"2020-03-01T10:22:14","Local":"2020-03-01T21:22:14","StartDST":"2020-10-04T02:00:00","EndDST":"2020-04-05T03:00:00","Timezone":99,"Sunrise":"06:49","Sunset":"19:39"}}
21:22:14 RSL: STATUS10 = {"StatusSNS":{"Time":"2020-03-01T21:22:14","COUNTER":{"C1":2365.820763},"ANALOG":{"A0":4},"DS18B20":{"Id":"000473554DFF","Temperature":26.1},"TempUnit":"C"}}
21:22:14 RSL: STATUS11 = {"StatusSTS":{"Time":"2020-03-01T21:22:14","Uptime":"0T02:40:50","UptimeSec":9650,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":81,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"WitsEnd","BSSId":"10:DA:43:1A:BC:94","Channel":6,"RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:28"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
21:25:02 RUL: COUNTER#C1 performs "IF (2365.820763==4200) Var5 0 ELSE Var5=3600/2365.820763 ENDIF"
21:25:02 SRC: Rule
21:25:02 CMD: Group 0, Index 1, Command "IF", Data "(2365.820763==4200) Var5 0 ELSE Var5=3600/2365.820763 ENDIF"
21:25:02 RSL: RESULT = {"If":"Done"}
21:25:02 SRC: Backlog
21:25:02 CMD: Group 0, Index 5, Command "VAR", Data "=3600/2365.820763"
21:25:02 RSL: RESULT = {"Var5":"1.522"}
21:25:02 APP: Button1 multi-press 1
21:25:02 RUL: DS18B20#TEMPERATURE performs "Var1 26.1"
21:25:02 SRC: Rule
21:25:02 CMD: Group 0, Index 1, Command "VAR", Data "26.1"
21:25:02 RSL: RESULT = {"Var1":"26.1"}
21:25:03 RUL: COUNTER#C1 performs "IF (2365.820763==4200) Var5 0 ELSE Var5=3600/2365.820763 ENDIF"
21:25:03 SRC: Rule
21:25:03 CMD: Group 0, Index 1, Command "IF", Data "(2365.820763==4200) Var5 0 ELSE Var5=3600/2365.820763 ENDIF"
21:25:03 RSL: RESULT = {"If":"Done"}
21:25:03 SRC: Button
21:25:03 RSL: RESULT = {"POWER":"OFF"}
21:25:03 RSL: POWER = OFF
21:25:03 SRC: Backlog
21:25:03 CMD: Group 0, Index 5, Command "VAR", Data "=3600/2365.820763"
21:25:03 RSL: RESULT = {"Var5":"1.522"}
21:25:03 RUL: POWER1#STATE performs "Backlog Var2 0; Var4=0*0; Event Report"
21:25:03 SRC: Rule
21:25:03 CMD: Group 0, Index 1, Command "BACKLOG", Data "Var2 0; Var4=0*0; Event Report"
21:25:03 RUL: DS18B20#TEMPERATURE performs "Var1 26.1"
[... 34 lines cut ...]

### TO REPRODUCE
_Steps to reproduce the behavior:_
Run a few rules with different triggers, such as Rule2 above.

The DS18B20#Temperature and Counter#C1 triggers are occuring spontaneously despite no change in sensor value.

The triggered Power1#State within the console output above was deliberately injected by toggling the relay (ie only occurred due to a change). The Analog#A0div10 trigger was not tested during the output, but is also only firing upon a change, as expected.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Rule triggers should only fire upon change of the underlying trigger value, as described within the documentation (https://tasmota.github.io/docs/#/Rules?id=examples-of-available-triggers).

Users should not have to handle different triggers inconsistently within their rules with bespoke filtering code for each. I'm hard-up against the rule char limit as it is!

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
#7119 may be related; as per @ascillato I have tried making Rule2 one-shot, but this results in the triggers firing once only ever, despite change in value.

The as-expected firing (on change) of Power1#State versus the spontaneous firing of Counter#C1 and DS18B20#Temperature I believe disproves that whether the trigger is conditional or not is relevant to the problem.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2020-03-01:
Confirmed. On 8.0.1.10, I was just now wondering what I was missing when playing with
`on BME280#Temperature do Status 10 endon`
and seeing the trigger fire every approx 5 seconds, with the same temperature value in the the JSON. Having set TempRes to 1, I was only expecting long intervals.

arendst, 2020-03-02:
To rule out certain features you might want to try rules WITHOUT USE_EXPRESSION and SUPPORT_IF_STATEMENT

It might fail too but it makes debugging a lot simpler as half of the code doesn't need to be checked. As I think the issue is in the base rule code I'm still working on it.

sfromis, 2020-03-02:
My confirm was based on precompiled
`"Version":"8.1.0.10(fb87a03-sensors)","BuildDateTime":"2020-02-27T19:01:39"`
having neither of those optional features.

arendst, 2020-03-02:
Looking at the code I think what you expect from ``on DS18B20#Temperature do Status 10 endon`` never has worked.

- With one-shot disabled every time the rule is checked it is valid because there is any temperature so you receive your status 10 output constantly.
- With one-shot enabled the first time you will receive a status 10 message but afterwards you won't because there is nothing to check the temperature to.

arendst, 2020-03-02:
Both ``Power1#State``, and ``Analog#A0div10`` are events which are handled differently than "normal" sensor events as they are being pre-processed by their respective drivers.

 ``Power1#State`` is only executed if a change in power state did take place. Same for ``Analog#A0div10`` which is only executed when the value has changed by more than 10.

Most other sensors do not push their data to the rules handler (which would become overflowed) but the rules handler pulls their data once per 100 mSec/per sensor and processes it. It has no knowledge of previous values and hence if no rule value is given it will always be true resulting in your experience.

ascillato, 2020-03-02:
A rule trigger without any comparison will be executed continuously. This behavior is used by several users.

A complete rule trigger requires the comparison part. 

For publishing a sensor value on every change, there is a rule example in the wiki where you set a threshold for that. That approach is really good because you can avoid the sensor noise or really small sensor changes, making your rules more robust.

So, seems that the docs must be corrected. A rule trigger without a comparison will ve executed continuously and NOT on sensor's change.

denisfrench, 2020-03-03:
So it's a feature not a bug then.

Solving the continuous temp readout was easy enough; change
```
ON DS18B20#Temperature DO Var1 %value% ENDON
```
to
```
ON DS18B20#Temperature!=%Var1% DO Var1 %value% ENDON
```
Similarly
```
ON Counter#C1!=%Var6% DO IF (%value%==4200) Var5 0; Var6 %value% ELSE Var5=3600/%value%; Var6 %value% ENDIF ENDON
```
does fix the Counter#C1 instance, but I don't have enough space left in the rules.

I'll have a play with MAX_RULE_SETS and/or MAX_RULE_SIZE and see what else that breaks :wink:

closed 2020-03-03 by denisfrench

denisfrench, 2020-03-04:
Unsurprisingly, messing with MAX_RULE_SETS or MAX_RULE_SIZE screws up the settings memory allocation, and the device reloads defaults at every boot.

Solving that is beyond me for now; I'll need to have a think about where-to from here :confused: