arendst/Tasmota #7834: domoticz and tasmota with blind support not comunicate properly

megamarco833, 2020-03-01:
### PROBLEM DESCRIPTION
startin from a setup like this:
n°2 blinds controlled by n°4 relay:
     - relay1 = up blind1
     - relay2 = down blind1
     - relay1 = up blind2
     - relay1 = down blind2

interlock 1 and 2   ;   3 and 4

the issue is that on tasmota web page relays and movement respond correctly.

but from domoticz integration not.
i mean that if fill inside tasmota web page dedicated to domoticz the idx of blind1 = 111 and idx of blind2 =154
the relays will not respond correctly.
it seams that if i move the percentage from domoticz it trigger the relay, but when finish, tasmota trigger again relays and it create a mismatch and wrong positioning.
second problem is that if i move in domoticz, only blind2, tasmota trigger the relay1 (that is dedicated to blind1) 

as i wrote, if i control from tasmota, without enable web page dedicated to domotiz without insert idxs all works fine.
the problem seams to be in integration of tasmota with domoticz

### REQUESTED INFORMATION

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  NodeMCU
- [x] Tasmota binary firmware version number used:  8.1 with anable blind, nothing else
  - [ ] Pre-compiled
  - [x] Self-compiled (using on line tasmota compiler)
    - [ ] IDE / Compiler used:  tasmota
- [ ] Flashing tools used: https://hassiohelp.eu/2020/01/25/tasmocompiler/
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
17:03:05 MQT: stat/tasmota_blind/RESULT = {"NAME":"tasmota_blind","GPIO":[0,12,9,11,30,29,0,0,32,10,31,0,0],"FLAG":0,"BASE":18}
17:03:05 MQT: stat/tasmota_blind/RESULT = {"Module":{"18":"Generic"}}
17:03:05 MQT: stat/tasmota_blind/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"12":"Switch4"},"GPIO2":{"9":"Switch1"},"GPIO3":{"11":"Switch3"},"GPIO4":{"30":"Relay2i"},"GPIO5":{"29":"Relay1i"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"32":"Relay4i"},"GPIO13":{"10":"Switch2"},"GPIO14":{"31":"Relay3i"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
  Configuration output here:

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
SetOption80 1
ShutterRelay1 1
ShutterRelay2 3
Backlog PulseTime1 0; PulseTime2 0; PulseTime3 0; PulseTime4 0
Backlog Interlock 1,2 3,4; Interlock ON
ShutterOpenDuration1 15
ShutterOpenDuration2 15
ShutterCloseDuration1 8
ShutterCloseDuration2 8

  Rules output here:
rules used only for switch, behaviour is:
     - switch1 => shorpress = activate relay1 (up blind1) for 1second
     - switch1 => longpress= activate relay1 (up blind1) till reach 100% up/open
     - switch2 => shorpress = activate relay2 (down blind1) for 1second
     - switch2 => longpress= activate relay2 (down blind1) till reach 100% down/close
     - switch3 => shorpress = activate relay3 (up blind2) for 1second
     - switch3 => longpress= activate relay3 (up blind2) till reach 100% up/open
     - switch4 => shorpress = activate relay4 (down blind2) for 1second
     - switch4 => longpress= activate relay4 (down blind2) till reach 100% down/close
  
so i use:
switchmode1 5    
switchmode2 5
switchmode3 5
switchmode4 5
setoption32 20   

[... 125 lines cut ...]
17:21:42 MQT: stat/tasmota_blind/RESULT = {"Shutter1":{"Position":41,"Direction":0},"Shutter2":{"Position":49,"Direction":0}}
17:21:43 SHT: Start in dir 0
17:21:43 MQT: stat/tasmota_blind/RESULT = {"POWER2":"ON"}
17:21:43 MQT: stat/tasmota_blind/POWER2 = ON
17:21:43 MQT: stat/tasmota_blind/RESULT = {"ShutterPosition1":0}
17:21:43 SHT: Shutter1 Real 10991, Start 12300, Stop 0, Dir -1, Delay 0, Rtc 0.35 [s], Freq 1875
17:21:43 MQT: stat/tasmota_blind/RESULT = {"Shutter1":{"Position":37,"Direction":-1},"Shutter2":{"Position":49,"Direction":0}}
17:21:44 SHT: Shutter1 Real 7625, Start 12300, Stop 0, Dir -1, Delay 0, Rtc 1.30 [s], Freq 1875
17:21:44 MQT: stat/tasmota_blind/RESULT = {"Shutter1":{"Position":25,"Direction":-1},"Shutter2":{"Position":49,"Direction":0}}
17:21:45 SHT: Shutter1 Real 3137, Start 12300, Stop 0, Dir -1, Delay 0, Rtc 2.45 [s], Freq 1875
17:21:45 MQT: stat/tasmota_blind/RESULT = {"Shutter1":{"Position":10,"Direction":-1},"Shutter2":{"Position":49,"Direction":0}}
17:21:46 MQT: domoticz/in = {"idx":111,"nvalue":0,"svalue":"","Battery":100,"RSSI":7}
17:21:46 MQT: stat/tasmota_blind/RESULT = {"POWER2":"OFF"}
17:21:46 MQT: stat/tasmota_blind/POWER2 = OFF
17:21:46 SHT: Shutter1 Real 0, Start 12300, Stop 0, Dir -1, Delay 0, Rtc 3.30 [s], Freq 1875
17:21:46 MQT: stat/tasmota_blind/SHUTTER1 = 0
17:21:46 MQT: stat/tasmota_blind/RESULT = {"Shutter1":{"Position":0,"Direction":0},"Shutter2":{"Position":49,"Direction":0}}
17:21:46 MQT: stat/tasmota_blind/RESULT = {"Shutter1":{"Position":0,"Direction":0},"Shutter2":{"Position":49,"Direction":0}}
17:21:46 MQT: stat/tasmota_blind/RESULT = {"ShutterPosition1":0}

again, this happened only if i full fille tha page of tasmota related to domoticz.
if i left all blank in this page, and i use only command and sliding inside tasmota all works perfectly.
the issue seams linked with communication between tasmota and domoticz, when in tasmota i activate it.

thanks to all for big effort putting in this.

stefanbode, 2020-03-01:
For me it looks like there is an additional command comming from domoticz switching relay 2. can you goto weblog 4 and I’m highly interested in the logs short before the sht: start in dir 0
In Weblog 4 we should be able to see what command is incoming and how this is send. The firmware just seem to execute what to do.

megamarco833, 2020-03-01:
hi Stefan, thanks for reply!

if i not wrong understand what you mean, i acrivate weblog 4 on cosolle.
then in domoticz i move the percentage of blind1, that was "close" and set at 30%.

i see that relay1 was activated (for me correct)
then when relay1 deactivated, the relay2 was activated (wrong!)
and relay2 stay active till reach close position
in domoticz, it was reported as blind1 = close (correct because the behaviur but i want blind1 at 30% not close :-) )

this is the output. it's what you need?
thanks

`18:32:56 MQT: Received Topic "domoticz/out", Data Size 303, Data "{
   "Battery" : 255,
   "RSSI" : 5,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82021",
   "idx" : 21,
   "name" : "Mansarda",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "22.60",
   "svalue2" : "39.7",
   "svalue3" : "0",
   "unit" : 1
}
"
18:32:56 DOM: idx 21, nvalue 0
18:32:56 WIF: Checking connection...
18:32:57 MQT: Received Topic "domoticz/out", Data Size 226, Data "{
   "Battery" : 255,
   "RSSI" : 4,
   "description" : "",
   "dtype" : "Temp",
   "id" : "82117",
   "idx" : 117,
   "name" : "Temp_ACS",
   "nvalue" : 0,
   "stype" : "LaCrosse TX3",
   "svalue1" : "44.90",
   "unit" : 1
}
"
18:32:57 DOM: idx 117, nvalue 0
18:32:57 MQT: Received Topic "domoticz/out", Data Size 264, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140C7",
   "idx" : 119,
   "name" : "rele_caldaia_ACS",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "On/Off",
   "unit" : 1
}
"
18:32:57 DOM: idx 119, nvalue 0
18:32:57 MQT: Received Topic "domoticz/out", Data Size 266, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140C2",
   "idx" : 114,
   "name" : "Valvola_TreVie_ACS",
[... 322 lines cut ...]
   "id" : "82014",
   "idx" : 14,
   "name" : "Corridoio",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "21.50",
   "svalue2" : "37.1",
   "svalue3" : "0",
   "unit" : 1
}
"
18:33:24 DOM: idx 14, nvalue 0
18:33:25 MQT: Received Topic "domoticz/out", Data Size 239, Data "{
   "Battery" : 255,
   "RSSI" : 8,
   "description" : "",
   "dtype" : "Temp",
   "id" : "82116",
   "idx" : 116,
   "name" : "Ritorno_Riscaldamento",
   "nvalue" : 0,
   "stype" : "LaCrosse TX3",
   "svalue1" : "19.20",
   "unit" : 1
}`

stefanbode, 2020-03-01:
I have only an iPad here and the code is in this block format very hard to read. Can you repost it. I am interested what Switches the relay 2, because until this everything is ok.

stefanbode, 2020-03-01:
I can see clearly that domoticz is sending in the /out a Shutterposition 0. why is it doing this. For sure in this case then with Idx168 the shutter closes down again. The command is directly send by domoticz. I would search the error there. The firmware just do what it got told.

18:33:11 MQT: Received Topic "domoticz/out", Data Size 260, Data "{ "Battery" : 100, "RSSI" : 7, "description" : "", "dtype" : "Light/Switch", "id" : "000140F8", "idx" : 168, "name" : "tap_tasmota_1", "nvalue" : 0, "stype" : "Switch", "switchType" : "Blinds Percentage Inverted", "unit" : 1 } " 
18:33:11 DOM: idx 168, nvalue 0 
18:33:11 DOM: Received Topic /ShutterPosition, Data 0 
18:33:11 SRC: MQTT 18:33:11 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "0" 
18:33:11 SHT: Pos. in: payload 0 (260), payload 0, idx 1, src 23 18:33:11 SHT: lastsource 23:, real 10500, target 0, payload 0

megamarco833, 2020-03-01:
> I am interested what Switches the relay 2, because until this everything is ok.

ok, sorry, hope that now is better.
i redo the test starting from blind1 at all close and moving to 38%

```
18:48:12 CMD: weblog 4
18:48:12 MQT: stat/tasmota_blind/RESULT = {"WebLog":4}
18:48:12 CFG: Saved to flash at F9, Count 307, Bytes 4096
18:48:16 WIF: Checking connection...
18:48:16 MQT: Received Topic "domoticz/out", Data Size 235, Data "{
   "Battery" : 255,
   "RSSI" : 6,
   "description" : "",
   "dtype" : "Temp",
   "id" : "82095",
   "idx" : 95,
   "name" : "Temperatura_Camino",
   "nvalue" : 0,
   "stype" : "LaCrosse TX3",
   "svalue1" : "64.30",
   "unit" : 1
}
"
18:48:16 DOM: idx 95, nvalue 0
18:48:17 MQT: Received Topic "domoticz/out", Data Size 305, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82014",
   "idx" : 14,
   "name" : "Corridoio",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "21.50",
   "svalue2" : "37.0",
   "svalue3" : "0",
   "unit" : 1
}
"
18:48:17 DOM: idx 14, nvalue 0
18:48:20 MQT: Received Topic "domoticz/out", Data Size 282, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 168,
   "name" : "tap_tasmota_1",
   "nvalue" : 2,
   "stype" : "Switch",
   "svalue1" : "38",
   "switchType" : "Blinds Percentage Inverted",
   "unit" : 1
}
"
18:48:20 DOM: idx 168, nvalue 2
18:48:20 DOM: Received Topic /ShutterPosition, Data 38
18:48:20 SRC: MQTT
18:48:20 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "38"
18:48:20 SHT: Pos. in: payload 38 (282), payload 38, idx 1, src 9
18:48:20 SHT: lastsource 9:, real 0, target 11400, payload 38
18:48:20 SHT: Start in dir 0
18:48:20 SRC: Shutter
18:48:20 SHT: Switched relay: 1 by Shutter
18:48:20 MQT: stat/tasmota_blind/RESULT = {"POWER1":"ON"}
18:48:20 MQT: stat/tasmota_blind/POWER1 = ON
18:48:20 MQT: stat/tasmota_blind/RESULT = {"ShutterPosition1":38}
18:48:20 CFG: Saved to flash at F8, Count 308, Bytes 4096
[... 224 lines cut ...]
   "svalue2" : "37.0",
   "svalue3" : "0",
   "unit" : 1
}
"
18:48:36 DOM: idx 14, nvalue 0
18:48:40 MQT: Received Topic "domoticz/out", Data Size 310, Data "{
   "Battery" : 255,
   "RSSI" : 9,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82153",
   "idx" : 153,
   "name" : "Cucina cantina",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "16.20",
   "svalue2" : "46.8",
   "svalue3" : "1",
   "unit" : 1
}
"
18:48:40 DOM: idx 153, nvalue 0
18:48:40 CMD: weblog 2
```

megamarco833, 2020-03-01:
here another test with blind2
i staerted with domoticz with blind2 at 30% and move the slide at 60%

it activated relay1 and relay2 (why not relay3 as i expected??)

```
18:54:43 CMD: weblog 4
18:54:43 MQT: stat/tasmota_blind/RESULT = {"WebLog":4}
18:54:44 CFG: Saved to flash at FA, Count 322, Bytes 4096
18:54:46 MQT: Received Topic "domoticz/out", Data Size 235, Data "{
   "Battery" : 255,
   "RSSI" : 6,
   "description" : "",
   "dtype" : "Temp",
   "id" : "82095",
   "idx" : 95,
   "name" : "Temperatura_Camino",
   "nvalue" : 0,
   "stype" : "LaCrosse TX3",
   "svalue1" : "64.10",
   "unit" : 1
}
"
18:54:46 DOM: idx 95, nvalue 0
18:54:47 MQT: Received Topic "domoticz/out", Data Size 240, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp",
   "id" : "63873",
   "idx" : 19,
   "name" : "Mandata Riscaldamento",
   "nvalue" : 0,
   "stype" : "LaCrosse TX3",
   "svalue1" : "19.1",
   "unit" : 129
}
"
18:54:47 DOM: idx 19, nvalue 0
18:54:48 MQT: Received Topic "domoticz/out", Data Size 305, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82014",
   "idx" : 14,
   "name" : "Corridoio",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "21.50",
   "svalue2" : "36.8",
   "svalue3" : "0",
   "unit" : 1
}
"
18:54:48 DOM: idx 14, nvalue 0
18:54:48 WIF: Checking connection...
18:54:49 MQT: Received Topic "domoticz/out", Data Size 282, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F9",
   "idx" : 169,
   "name" : "tap_tasmota_2",
   "nvalue" : 2,
   "stype" : "Switch",
   "svalue1" : "60",
   "switchType" : "Blinds Percentage Inverted",
   "unit" : 1
[... 265 lines cut ...]
   "svalue1" : "19.10",
   "unit" : 1
}
"
18:55:05 DOM: idx 116, nvalue 0
18:55:07 MQT: Received Topic "domoticz/out", Data Size 305, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82014",
   "idx" : 14,
   "name" : "Corridoio",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "21.50",
   "svalue2" : "36.8",
   "svalue3" : "0",
   "unit" : 1
}
"
18:55:07 DOM: idx 14, nvalue 0
18:55:09 WIF: Checking connection...
18:55:09 CMD: weblog 2
```

stefanbode, 2020-03-01:
Even if we accept this command the nvalue is false 0 is the old value. The. New position was send before and should be known to domoticz. Maybe someone having more experience with the domoticz can help here. From the firmware side of shutters everything seems ok. Maybe domoticz has problems to interpret the JSon just send before because the format changed from Tasmota 7 tasmota 8. anyhow this must be changed in the configuration of domoticz 

18:54:58 MQT: Received Topic "domoticz/out", Data Size 260, Data "{
   "Battery" : 100,
   "RSSI" : 7,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 168,
   "name" : "tap_tasmota_1",
   "nvalue" : 0,
   "stype" : "Switch",
   "switchType" : "Blinds Percentage Inverted",
   "unit" : 1
}

megamarco833, 2020-03-01:
another test now startin on move from tasmota web page.
i move blind2 that's around 50% and i slide it around 20%
now i very strange behavour:
i see trigger relay3 (=>blind2) and relay1(=>blind1) togheter
then stop relay3 and 1
and triger relay2
then stop relay2
and trigger rela1
at the end i see in domoticz blind1 = all close  ;   blind2= all close
and in tasmota web page i see blind1=all close   ;    blind2= at position where i slide (around 20%)

```
18:58:15 CMD: weblog 4
18:58:15 MQT: stat/tasmota_blind/RESULT = {"WebLog":4}
18:58:16 CFG: Saved to flash at F4, Count 328, Bytes 4096
18:58:16 MQT: Received Topic "domoticz/out", Data Size 235, Data "{
   "Battery" : 255,
   "RSSI" : 6,
   "description" : "",
   "dtype" : "Temp",
   "id" : "82095",
   "idx" : 95,
   "name" : "Temperatura_Camino",
   "nvalue" : 0,
   "stype" : "LaCrosse TX3",
   "svalue1" : "63.90",
   "unit" : 1
}
"
18:58:16 DOM: idx 95, nvalue 0
18:58:18 MQT: Received Topic "domoticz/out", Data Size 305, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82014",
   "idx" : 14,
   "name" : "Corridoio",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "21.50",
   "svalue2" : "36.8",
   "svalue3" : "0",
   "unit" : 1
}
"
18:58:18 DOM: idx 14, nvalue 0
18:58:18 SRC: WebGui from 192.168.0.4
18:58:18 CMD: Group 0, Index 2, Command "SHUTTERPOSITION", Data "17"
18:58:18 SHT: Pos. in: payload 17 (2), payload 17, idx 2, src 7
18:58:18 SHT: lastsource 7:, real 15300, target 5100, payload 17
18:58:18 SHT: Start in dir 0
18:58:18 SRC: Shutter
18:58:18 SHT: Switched relay: 8 by Shutter
18:58:18 MQT: domoticz/in = {"idx":169,"nvalue":1,"svalue":"","Battery":100,"RSSI":7}
18:58:18 MQT: stat/tasmota_blind/RESULT = {"POWER4":"ON"}
18:58:18 MQT: stat/tasmota_blind/POWER4 = ON
18:58:18 MQT: stat/tasmota_blind/RESULT = {"ShutterPosition2":17}
18:58:19 MQT: Received Topic "domoticz/out", Data Size 260, Data "{
   "Battery" : 100,
   "RSSI" : 7,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F9",
   "idx" : 169,
   "name" : "tap_tasmota_2",
   "nvalue" : 1,
   "stype" : "Switch",
   "switchType" : "Blinds Percentage Inverted",
   "unit" : 1
[... 450 lines cut ...]
   "stype" : "LaCrosse TX3",
   "svalue1" : "19.10",
   "unit" : 1
}
"
18:58:49 DOM: idx 116, nvalue 0
18:58:53 MQT: Received Topic "domoticz/out", Data Size 305, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82014",
   "idx" : 14,
   "name" : "Corridoio",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "21.50",
   "svalue2" : "36.7",
   "svalue3" : "0",
   "unit" : 1
}
"
18:58:53 DOM: idx 14, nvalue 0
18:58:55 CMD: weblog 2
```

stefanbode, 2020-03-01:
One important thing: domoticz is not sending the index of the shutter the Shutterposition Mut be Shutterposition1 or 2. there is something really wrong with your domoticz integration. This does not match at all. Maybe you get the position 0 from shutter2 back on Shutterposition because you missed the index.

megamarco833, 2020-03-01:
> Even if we accept this command the nvalue is false 0 is the old value. The. New position was send before and should be known to domoticz. Maybe someone having more experience with the domoticz can [...]

ok, maybe is here the problem, on configuration of tasmota web page related to domoticz.
because the test that i posted was made with this configuration:
[image]
[image]
as you can see i reported idx1 and idx2 = 168 (blind1)
and idx3 and idx4 = 169 (blind 2)

is i delete all value and put at zero, 
if i move percentage from domoticz nothing appened => no relays are triggered
if i move percentage from tasmota web page => correct behavier of relays, but nothin reported in domoticz

maybe the problem is inside tasmota and way/format of it report to domoticz?
or i need to change the configuration posted on picture above?
but i try many variants without succeed. i try for example to put
idx 1 = 168
idx 2 = 0
idx 3 = 169
idx 4 = 0
but no luky, same issue

megamarco833, 2020-03-01:
> One important thing: domoticz is not sending the index of the shutter the Shutterposition Mut be Shutterposition1 or 2. there is something really wrong with your domoticz integration. This does not [...]

ok, but inside domoticz, there are not parameters to change to send out shutterposition1 or shutter position2 if i use blind#2.

maybe Theo that use domoticz, can contribute better than me, but as far as i know in domoticz there is nothin to set.

megamarco833, 2020-03-01:
on proposed solution could be:
deactivate from tasmota web page all idx of domoticz.
then use a rule like this:

`Rule1 ON Shutter1#Position DO publish domoticz/in {"idx":111,"nvalue":2,"svalue":"%value%"} endon ON Shutter2#Position DO publish domoticz/in {"idx":154,"nvalue":2,"svalue":"%value%"} endon`

and so i'm able to send to domoticz the positioning if i use slide on tasmota main webpage, and if i use my wall switch.

now i need to implement in domoticz the istruction to give to tasmota if i start to move from domoticz.
maybe i can use lua, or dzvents (preferred for me) to make a sort of automation that, if i move slider of blind1, it recognize and send the mqtt messagge.

for example, please Stefan correct me if i'm wrong, should publish this:
cmnd/<topic_of_my_nodemcu_blind/shutterposition<number of blind to conrol> <percentage>
so for blind1 at 30%
cmnd/tasmota_blind/shutterposition1 30
correct?

my dubt is, if i use this command, using automation dzvents in domoticz, so sending directly the command, it could generate a loop?

i need also to implement something that in domoticz recognize something?

last question, now i'm testing with two blind, but i can use for example 4 blind and 4 wall switch.
there is a limitation on characters that i can use inside rule1?

because i would like to use rule1 to send data to domoticz
and rule2 to long/short press of switches up and down

thanks

stefanbode, 2020-03-01:
Hi yes, Shutterposition1 30 is the right command. Rules have 511 bytes but Take a look in the wiki. If you sending the same command several times, no problem.

megamarco833, 2020-03-01:
ok thanks, so tasmota should send to domoticz

publish domoticz/in {"idx":<IDX_number>,"nvalue":2,"svalue":"<percentage>"}
so for example:
publish domoticz/in {"idx":168,"nvalue":2,"svalue":"%value%"}
i guess i can use %value% to be updated on every position, right?

so for example i can use rules like:
ON Shutter1#Position DO publish domoticz/in {"idx":168,"nvalue":2,"svalue":"%value%"} endon

from domoticz side when i move the blind i need to publish this:
cmnd/tasmota_blind/shutterposition1 30

all correct till here?

what i'm afraid on, is:
if i use dzvents to recongnize when i move blind slide on domoticz, and publish
cmnd/tasmota_blind/shutterposition1 30
when tasmota finish to trigger relay1 at 30%, tasmota, thanks to rule1 (ON Shutter1#Position DO publish domoticz/in {"idx":168,"nvalue":2,"svalue":"%value%"} endon) will send:
publish domoticz/in {"idx":168,"nvalue":2,"svalue":"30"}

in this case we do not generate a loop? because domoticz receive:
publish domoticz/in {"idx":168,"nvalue":2,"svalue":"30"}
and send to tasmota:
cmnd/tasmota_blind/shutterposition1 30
and so on??

anyway, what i'm try to build is a workaround, i think that someone should have a look on tasmota side, in domoticz webpage, maybe creating special field for rollershutter, so tasmota can talk correctly with domoticz and viceversa.
thanks

stefanbode, 2020-03-02:
Can you just try to disable the second shutter in tasmota. If it then works we have with domoticz an issue with the numbering of shutters. I’m pretty sure we have several implementation running with at least one shutter

megamarco833, 2020-03-02:
hi, i remove on tasmota all blinds, and left only shutter 1
in domoticz i left all 4 blinds
in tasmota i added idx1= 168 and idx2=168

i tested  moving on domoticz blind1 start from 38  to 75%

and after i move blnd 4 start from 74 to 19%

this is the log

```

12:52:22 CMD: weblog 4
12:52:22 MQT: stat/tasmota_blind/RESULT = {"WebLog":4}
12:52:22 CFG: Saved to flash at F7, Count 177, Bytes 4096
12:52:23 MQT: Received Topic "domoticz/out", Data Size 303, Data "{
   "Battery" : 255,
   "RSSI" : 8,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82021",
   "idx" : 21,
   "name" : "Mansarda",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "22.10",
   "svalue2" : "40.1",
   "svalue3" : "1",
   "unit" : 1
}
"
12:52:23 DOM: idx 21, nvalue 0
12:52:24 MQT: Received Topic "domoticz/out", Data Size 305, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82014",
   "idx" : 14,
   "name" : "Corridoio",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "21.30",
   "svalue2" : "35.4",
   "svalue3" : "0",
   "unit" : 1
}
"
12:52:24 DOM: idx 14, nvalue 0
12:52:27 MQT: Received Topic "domoticz/out", Data Size 282, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "000140F8",
   "idx" : 168,
   "name" : "tap_tasmota_1",
   "nvalue" : 2,
   "stype" : "Switch",
   "svalue1" : "80",
   "switchType" : "Blinds Percentage Inverted",
   "unit" : 1
}
"
12:52:27 DOM: idx 168, nvalue 2
12:52:27 DOM: Received Topic /ShutterPosition, Data 80
12:52:27 SRC: MQTT
12:52:27 CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "80"
12:52:27 SHT: Pos. in: payload 80 (282), payload 80, idx 1, src 9
12:52:27 SHT: lastsource 9:, real 11400, target 24000, payload 80
[... 287 lines cut ...]
"
12:52:49 DOM: idx 153, nvalue 0
12:52:49 MQT: Received Topic "domoticz/out", Data Size 305, Data "{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Temp + Humidity",
   "id" : "82014",
   "idx" : 14,
   "name" : "Corridoio",
   "nvalue" : 0,
   "stype" : "THGN122/123/132, THGR122/228/238/268",
   "svalue1" : "21.30",
   "svalue2" : "35.4",
   "svalue3" : "0",
   "unit" : 1
}
"
12:52:49 DOM: idx 14, nvalue 0
12:52:50 CMD: weblog 2
```

the behaviuor is corrected, domoticz and tasmota now talk correctly.
but of course there is somenthing to menage to implment te usability of 4 blinds correct?
maybe on tasmota web page of domoticz should be introduced some fix for blind?

megamarco833, 2020-03-05:
> 
> 
> Can you just try to disable the second shutter in tasmota. If it then works we have with domoticz an issue with the numbering of shutters. I’m pretty sure we have several implementation running with at least one shutter

i'm try to maniging this issue with nodered
i need one help to avoid a ping-pong messages between domoticz and tasmota
when i try to controll with a master blind all blind togheter at same position.

how i can make a rule that send the position at domotiz, only when i use a switch (switchmode5, so long press and short press)

now i have:
`Rule1 ON Shutter1#Position DO publish domoticz/in {"idx":169,"nvalue":2,"svalue":"%value%"} endon ON Shutter2#Position DO publish domoticz/in {"idx":170,"nvalue":2,"svalue":"%value%"} endon ON Shutter3#Position DO publish domoticz/in {"idx":171,"nvalue":2,"svalue":"%value%"} endon ON Shutter4#Position DO publish domoticz/in {"idx":172,"nvalue":2,"svalue":"%value%"} endon`

and rule due to use wall switch long press and short press
```
rule2 on switch1#state=2 do backlog power1 on;delay 10;power1 off endon on switch1#state=3 do power1 on endon
on switch2#state=2 do backlog power2 on;delay 10;power2 off endon on switch2#state=3 do power2 on endon
on switch3#state=2 do backlog power3 on;delay 10;power3 off endon on switch3#state=3 do power3 on endon
on switch4#state=2 do backlog power4 on;delay 10;power4 off endon on switch4#state=3 do power4 on endon
```

but rule1 interfering when i send the comand from domoticz to all single blind together

i would like to remove rule1 and integrate in rule2
so when i use short press switch1=> activate relay1 for 1second and send to domoticz the position (svalue1)
publish domoticz/in {"idx":169,"nvalue":2,"svalue":"**HERE_I_NEED_THE_PERCENTAGE_VALUE**"}
when i use long press switch1=> acrivate relay1 for reach 100% and send to domoticz the position (svalue1)
publish domoticz/in {"idx":169,"nvalue":2,"svalue":"**HERE_I_NEED_THE_PERCENTAGE_VALUE**"}

last question, there is a command that can put all blinds on same ESP8266 at same position?

thanks

megamarco833, 2020-03-18:
i solve the issue related to send the value to domoticz without having a ping-pong messages between domoticz and tasmota not using `Shutter1#Position` but `Shutter1#Target`

```ON Shutter1#Target DO publish domoticz/in {"idx":169,"nvalue":2,"svalue":"%value%"} endon ON Shutter2#Target DO publish domoticz/in {"idx":170,"nvalue":2,"svalue":"%value%"} endon ON Shutter3#Target DO publish domoticz/in {"idx":171,"nvalue":2,"svalue":"%value%"} endon ON Shutter4#Target DO publish domoticz/in {"idx":172,"nvalue":2,"svalue":"%value%"} endon```

this is a workaround still an issue that the webpage in tasmota dedicated to idx of domoticz is not working for shutters.

stefanbode, 2020-04-10:
I will take a look on the domoticz special code in the shutter, If we can improve something there. From the point of the shutter firmware I cannot see an issue. It is domoticz that mixes the value of shutter2 with shutter1 and creates the inconsistent behavior there.

stefanbode, 2020-04-10:
I took a closer look at the domoticz implementation and I do think we have n issue in xdrv_07_domoticz.ino. The enhancement for shutters did not take into consideration that there is more than one shutter in my mind. Whoever made the changes and is familiar with domoticz should be able to make the required changes. I do not have insight how this software works.

megamarco833, 2020-04-10:
Yes, correct. Domoticz web page not consider in my opinion two things :
We have more than one shutter and we have relay1=up=shutter1
And relay2=down=shutter1
So every shutter has 2 relay but same idx in domoticz, because for donoticz shutter is like a dimmable light so only one idx but two outpur relay. 
I think that in tasmota shutter configuration for domoticz should considered a dedicated case. 
Adding to this, a very common configuration i guess is using pcf8575 for output (relay from 1 to 8)
And input for push button up and down for every shutters.
I think that tasmota domoticz web page should also consider that for idx assignment. 

If you need a tester, i can perform, because i have on my desk a 8ch relay a tasmota on nodemcu and 8 input pushbuttons to simulate longpress (activation contiusly relay to perform a complete up or down. That could be interrupted if I press again same push button or the opposite push button)
And short press to activate the relay up o down just for one second, to simulate the venetian blind orientation. 
We talked about it also in another request made by onother user. 
If you need some test i have all devices ready :)

megamarco833, 2020-04-10:
[VID_20200410_130609.zip](https://github.com/arendst/Tasmota/files/4461258/VID_20200410_130609.zip)
just look my video please.
inside the video you can see that:

i have 1 blind and 2 relay (up and down) and 2 push buttons (up and down)
if i shor press up button(time<2seconds) i tilt up => trigger relay1 for1second=> trigger motor up direction for 1 second
if i shor press down button (time<2seconds) i tilt down => trigger relay2 for 1second=> trigger motor for 1 second

if i longpress up button (time>=2seconds) i completly go up => trigger relay1 contiuosly=> trigger motor up direction contiusly to reach 100% open
if i longpress down button (time>=2seconds) i completly go down => trigger relay2 contiuosly=> trigger motor up direction contiusly to reach 0% close

if during a longress up or down i pres push button up or push button down i stop the continuos moviment of motor => stop activated relay1 or relay2 (dipend whitch one was actived at that time) and the blind stop.

megamarco833, 2020-05-18:
Hi, the issue on tasmota web page dedicated to domoticz, where there is the idx reserved to shutter still not working.
Thanks to all

closed 2020-06-17 by stale[bot]