arendst/Tasmota #7835: Zigbee - Retain flag

snakuzzo, 2020-03-01:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
No problem...but I need to have some sensors values retained or I lose states after hub restarts

**Describe the solution you'd like**  
A command to set retain for specific device
eg. `ZigbeeSensorRetain` with the list of short addresses (separated by a comma) enabled to retain 

**Describe alternatives you've considered**  
Enable retain for all Zigbee messages

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2020-03-01:
Have you checked to use rules for this use-case. Var1 to 4 are persistent and might help you with your problem during reboot. Or you create a rule and send the vaulte as a retained mqtt message. Also an idea. Where exactly you want the value to survive?

s-hadinger, 2020-03-01:
I can add a retain flag to the `ZbReceived` message, but since they are all on the same topic, I'm afraid the last one will override all previous. Would this require specific topics per device?

snakuzzo, 2020-03-01:
For now I found a solution using RULES (just posted few minutes ago on Discord>general) as also suggested by @stefanbode.
I think the retain flag should be enabled for specific device. If you add it to `ZbReceived` message all messages of every device will published with retain flag.
The best solution should be to have specific topic per device...but you answered my some weeks ago in another issue that unique topic was a community decision.

Shold be great to change SENSOR messages from...

```
tele/%topic%/SENSOR
```

...to something like this...

 ```
tele/%short_addr%/SENSOR
```

s-hadinger, 2020-03-01:
Unless it's easy to republish to a device specific topic via rules, we can also add a SetOption to insert the short_addr in the topic specifically for retained messages. The JSON format will remain the same though. That's the community decision on the JSON message format.

snakuzzo, 2020-03-01:
Can you post an example?
Now we are talking about 2 different things.
Retain and Topic per device...

Topic per device always helps to not implement checks on single entity.
About the retain, if there is a way to do a new publish retreiving same topic and same payload from trigger...I think this is the best solution and no require any development

PR #7854 (2020-03-04): Add Zigbee use distinct MQTT topics per device for SENSOR, allowing retained messages (#7835)
## Description:

Add `SetOption89` to have distinct MQTT topics Zigbee SENSOR messages. This allows to persist the last known state of a zigbee device via retained messages.

When `SetOption81 1`, the 4 hex digit id of the device is added.

Exemple:

```
SetOption89 0
tele/tasmota/Zigbee_home/SENSOR = {"ZbReceived":{"Bulb":{"Device":"0x5ADF","Dimmer":254,"Endpoint":1,"LinkQuality":72}}}
```

```
SetOption89 1
tele/tasmota/Zigbee_home/5ADF/SENSOR = {"ZbReceived":{"Bulb":{"Device":"0x5ADF","Dimmer":254, [... cut]
closed 2020-03-04 by arendst