arendst/Tasmota #7837: graphs not grouped properly in home assistant (missing units ?)

ptempier, 2020-03-01:
**Have you looked for this feature in other issues and in the docs?**
yes

**Is your feature request related to a problem? Please describe.**  
Graphs are mixed up in home assistant

**Describe the solution you'd like**  
These metrics are all appearing in the same graph, my understanding is there's no unit for theses metrics when the data is sent to HA (because HA groups graphs with the same unit together).

Apparent power (should be VA ?)
Factor (should be cos phi ?)
Reactive power (should be VAR ?)
Total start time (should be h ?)
Status (should be % ?)

**Describe the solution you'd like**
Have the graphs grouping properly in home assistant, grouping by unit as done by HA while not perfect, is a very good start that works in a very generic way and give graphs some visibility.

**Describe alternatives you've considered**
Make and group all the graphs manually, or use grafana.

**Additional context**  
It s rather annoying because the units are différent, the graphs arent scalled properly and a lot of graphs are grouped together.

(Please, remember to close the issue when the problem has been addressed)

 
**(Please, remember to close the issue when the problem has been addressed)**

effelle, 2020-03-02:
What Tasmota version are you using?

ptempier, 2020-03-02:
Program Version	8.1.0(tasmota)
Build Date & Time	2019-12-25T12:33:25
Core/SDK Version	2_6_1/2.2.2-dev(38a443e)
Uptime	0T17:13:06
Flash write Count	144 at 0xF5000
Boot Count	37
Restart Reason	Software/System restart

Emulation	None
mDNS Discovery	Disabled
 
ESP Chip Id	12253746
Flash Chip Id	0x144051
Flash Size	1024kB
Program Flash Size	1024kB
Program Size	566kB
Free Program Space	436kB
Free Memory	27kB

This runs on the blitzwolf SHP6 high tension one, with the corresponding template

I trust that there are issues with discovery , i run HA on my computer because i didnt decided on a server/nas yet, so i restart it everyday. When i do this , i need to restart tasmota devices aferward

But i saw there were tickets for this kind of stuff so i didnt open one.

effelle, 2020-03-02:
Dev version starting from 8.1.0.6 (IIRC) got a lot of new sensors with additional UM. 
8.1.0 master is missing a lot of new stuff for HA discovery. Some UM will be changed soon. 
RSSI has not a `%` yet because in truth is not a % value(but yes, perhaps I'll add it). 
Uptime can't have a unit of measure because is not created as sensor but just an informative field under  `HAssPublishStatus`.  
I suggest you to wait next week for the new discovery.
>I trust that there are issues with discovery , i run HA on my computer because i didnt decided on a server/nas yet, so i restart it everyday. When i do this , i need to restart tasmota devices aferward
But i saw there were tickets for this kind of stuff so i didnt open one.

I have no idea of what you're talking about. I don't recall issues opened about it here. Care to share a link and let me see?

ptempier, 2020-03-03:
Thanks for the quick response

1)
If there are a lot of changes, then i guess it's better to wait and see how the next version behaves.
I ll close this feature request awaiting for he next release.

From the last comment, the disconnecting issue i saw is apparently linked to the wifi router. https://github.com/arendst/Tasmota/issues/7809

2)
My rough guess is that my problem is different and that my server stays offline for too long.
I think i ll try to get some help over irc/discord to get some helpfull debug infos; then open another issue for the connection lost after server restart.

closed 2020-03-03 by ptempier